MeshInspector/MeshLib #2162: Negative offset in thickenMesh results in incorrect filling of mesh

JeffreyWardman, 2024-02-05:
[image]

```
    params = mrmeshpy.GeneralOffsetParameters()
    params.signDetectionMode = mrmeshpy.SignDetectionMode.WindingRule
    params.mode = mrmeshpy.GeneralOffsetParametersMode.Smooth

    faces = np.array(mesh.cells.copy(), np.int32)
    faces = np.ndarray(shape=faces.shape, dtype=np.int32, buffer=faces)

    verts = mesh.vertices.copy()
    verts = np.ndarray(shape=verts.shape, dtype=np.float32, buffer=verts)

    mesh = mrmeshnumpy.meshFromFacesVerts(faces, verts)
    mesh = mrmeshpy.thickenMesh(mesh, offset=-3, params=params)

    vertices = mrmeshnumpy.getNumpyVerts(mesh)
    cells = mrmeshnumpy.getNumpyFaces(mesh.topology)
```

Grantim, 2024-02-05:
Hello! 
Looks like you have open input mesh, thats why `params.signDetectionMode = mrmeshpy.SignDetectionMode.WindingRule` does not work good. We recomment to use 'params.signDetectionMode = mrmeshpy.SignDetectionMode.Unsigned' for open meshes thickening, please try this
```py
    params = mrmeshpy.GeneralOffsetParameters()
    params.signDetectionMode = mrmeshpy.SignDetectionMode.Unsigned
    params.mode = mrmeshpy.GeneralOffsetParametersMode.Smooth

    faces = np.array(mesh.cells.copy(), np.int32)
    faces = np.ndarray(shape=faces.shape, dtype=np.int32, buffer=faces)

    verts = mesh.vertices.copy()
    verts = np.ndarray(shape=verts.shape, dtype=np.float32, buffer=verts)

    mesh = mrmeshnumpy.meshFromFacesVerts(faces, verts)
    mesh = mrmeshpy.thickenMesh(mesh, offset=-3, params=params)

    vertices = mrmeshnumpy.getNumpyVerts(mesh)
    cells = mrmeshnumpy.getNumpyFaces(mesh.topology)

JeffreyWardman, 2024-02-05:
Unfortunately that does not enclose between the original mesh and the internal component.

[image]

Grantim, 2024-02-05:
Currently our thicken algorithm only creates inner/outer component but does not connect components. This [discussion](https://github.com/MeshInspector/MeshLib/discussions/2134) may be useful for you

Fedr, 2026-09-25:
This is now supported by `makeThickMesh` (available since v3.0.8.228). Unlike `thickenMesh`, which only produces separate offset components, it clones the mirrored surface, shifts the two parts in opposite directions, and stitches their boundaries, so an open input gives one closed watertight mesh:

```py
from meshlib import mrmeshpy, mrmeshnumpy

mesh = mrmeshnumpy.meshFromFacesVerts(faces, verts)
params = mrmeshpy.ThickenParams()
params.insideOffset = 3   # thickness against the normals (your offset=-3)
params.outsideOffset = 0  # keep the original surface in place
mesh = mrmeshpy.makeThickMesh(mesh, params)
```

It moves vertices along their pseudo-normals and doesn't use voxels, so it's fast and keeps the original triangulation. If the thickness is large compared with the surface's curvature and you see self-intersections, try reducing `params.normalsTrustFactor`. Closing this; feel free to reopen if it doesn't fit your case.

closed 2026-09-25 by Fedr