MeshInspector/MeshLib #2175: Readme installation steps are incorrect for Linux

JeffreyWardman, 2024-02-06:
Steps:

```
sudo apt-get install libglfw3
sudo apt-get install -y libglfw3-dev libxerces-c-dev
git clone https://github.com/MeshInspector/MeshLib.git
cd MeshLib
git submodule update --init --recursive
./scripts/install_thirdparty.sh
... etc
```

`./scripts/build_source.sh` fails due to various packages not being found as well.

Is there a CI/CD pipeline or docker image to help make the installation process smoother so I can test `uniteCloseVertices` in Python via the lastest version.

oitel, 2024-02-06:
Could you specify your Linux distro version and attach the log files of the building process? The missing dependencies might indicate that the automatic package installation is not supported on your platform, we'll investigate and resolve it if possible.

The build process description in both Readme files will be updated later.

JeffreyWardman, 2024-02-06:
I use Pop OS 22.04 which is based on Ubuntu. The issue is that the readme itself is out of date. For instance, it doesn't state to update the submodules recursively. `build_source.sh` is currently `build_sources.sh` in the readme as well.

I've also tried building both Ubuntu and the Fedora 39 docker images without success.

Have you been able to follow the steps to follow the docker images for them to build successfully locally?

Hdbgdz, 2024-02-13:
Following the feature/ubuntu_build I still cannot install MeshLib. Despite having ZLIB installed on my machine. When running  ./scripts/build_thirdparty.sh I get the error:
`CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR) (Required is
  at least version "1.1.2")
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.22/Modules/FindZLIB.cmake:120 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  libzip/CMakeLists.txt:179 (find_package)`

oitel, 2025-04-27:
Closing the issue for now as we couldn't reproduce the mentioned build errors with the latest release.

closed 2025-04-27 by oitel