MeshInspector/MeshLib #2176: Installation on Linux when sudo privileges are not available

balav94, 2024-02-06:
Hello, 

I just came across MeshLib, and I would like to install MeshLib on a Supercomputer center used by everyone in my University, where we do not have sudo privileges. Please note that it uses Redhat Linux distribution. How can I install MeshLib C++ code on the Supercomputer in this case? I know that I can install the python version using pip install, however I would like to build the C++ version to be able to use both C++ and Python interfaces of MeshLib.

I would really appreciate it if anyone could provide any suggestions in this regard, 

Thanks in advance

oitel, 2024-02-06:
We have plans for extending build and install customization, once the code and scripts will be updated we'll notify you. 

Right now you might specify a custom installation path using `CMAKE_INSTALL_PREFIX` and `MESHLIB_CUSTOM_INSTALL_PREFIX` variables and disable some features requiring external dependencies with CMake options (you can find them [there](https://github.com/MeshInspector/MeshLib/blob/4e4a96368052253a3e5c807dfb0204bbbaa49ee4/CMakeLists.txt#L69) and [there](https://github.com/MeshInspector/MeshLib/blob/4e4a96368052253a3e5c807dfb0204bbbaa49ee4/source/MRMesh/CMakeLists.txt#L7)).

Note that some dependencies are intended to be installed by your package manager, so you need either to contact your system administrator to install them or to build them by yourself. For RHEL the [Fedora package list](https://github.com/MeshInspector/MeshLib/blob/4e4a96368052253a3e5c807dfb0204bbbaa49ee4/requirements/fedora.txt) might be used as a reference probably.

oitel, 2025-04-27:
@balav94 We've started providing distro-agnostic Linux x64 builds (based on Rocky Linux 9 and vcpkg). You can use the binaries:
https://github.com/MeshInspector/MeshLib/releases/tag/v3.0.5.215
…or build MeshLib and its dependencies from sources using vcpkg. Note that the latter method still requires some system dependencies to be installed by an admin.

We'd be glad to receive your feedback on it.

closed 2025-08-15 by oitel