MeshInspector/MeshLib #2239: meshlib.mrmeshpy.uniteCloseVertices does not exist in Python

JeffreyWardman, 2024-02-15:
meshlib-2.3.8.148, Python 3.10

There is no uniteCloseVertices in the expected release.

`from meshlib.mrmeshpy import uniteCloseVertices`

Grantim, 2024-02-16:
Hello!

Please add information about your OS? I have just try `from meshlib.mrmeshpy import uniteCloseVertices` on windows `py -3.10` and it worked fine

JeffreyWardman, 2024-02-16:
I'm using a mac (arm). Can confirm, tested with Linux and it's there but not for the mac binary.

Grantim, 2024-02-16:
Output from Mac M1
```
Python 3.10.13 (main, Aug 24 2023, 12:59:26) [Clang 15.0.0 (clang-1500.1.0.2.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from meshlib.mrmeshpy import uniteCloseVertices
>>> 
```
not sure how to reproduce the problem, please provide more details

JeffreyWardman, 2024-02-22:
Closing the issue as reinstalling on mac led to it not occurring anymore. When I pip freezed in the same virtualenv it came up with the version outputted. Regardless, doesn't seem to be an issue and may have been myself using the previous version by accident.

closed 2024-02-22 by JeffreyWardman