MeshInspector/MeshLib #2270: Plane cut

pty0220, 2024-02-23:
Hi, I am a beginner at MeshLib.
 
I would like to cut mesh using the plane.
I found the Python example and followed like this 

`mesh = mrmeshpy.loadMesh(pathStl)

plane = mrmeshpy.Plane3f()
plane.n = mrmeshpy.Vector3f(vectorFoward[5,0], vectorFoward[5,1], vectorFoward[5,2])
plane.d = np.linalg.norm(pointsForward[5,:])

sections = mrmeshpy.extractPlaneSections(mesh, plane)
contours = mrmeshpy.planeSectionsToContours2f(mesh, sections, mrmeshpy.AffineXf3f())`

After that, how can I make 'sections' ('SurfacePaths') as mesh? 
I tried to find the solution, but I cannot find it.

Grantim, 2024-02-23:
Hello!

We have functions in c++ code for this purpose https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRMeshTrimWithPlane.h but unfortunately they are not exposed to python module yet, we will add them in next release, thanks!

Now you can workaround by making plane mesh and using `boolean` function:
```py
mesh = mrmeshpy.loadMesh(pathStl)

plane = mrmeshpy.Plane3f()
plane.n = mrmeshpy.Vector3f(vectorFoward[5,0], vectorFoward[5,1], vectorFoward[5,2])
plane.d = np.linalg.norm(pointsForward[5,:])

# create plane 0XY mesh
planeMesh = mrmeshpy.Mesh()
planeMesh.points.vec.resize( 4 )
planeMesh.points.vec[0] = mrmeshpy.Vector3f(-0.5,-0.5,0)
planeMesh.points.vec[1] = mrmeshpy.Vector3f(0.5,-0.5,0)
planeMesh.points.vec[2] = mrmeshpy.Vector3f(0.5,0.5,0)
planeMesh.points.vec[3] = mrmeshpy.Vector3f(-0.5,0.5,0)
tris = mrmeshpy.Triangulation()
tris.vec.resize(2)
tris.vec[0] = mrmeshpy.ThreeVertIds(mrmeshpy.VertId(0),mrmeshpy.VertId(1),mrmeshpy.VertId(2))
tris.vec[1] = mrmeshpy.ThreeVertIds(mrmeshpy.VertId(0),mrmeshpy.VertId(2),mrmeshpy.VertId(3))
planeMesh.topology = mrmeshpy.topologyFromTriangles(tris)

# find plane mesh position
box = mesh.computeBoundingBox()
planeXf = mrmeshpy.AffineXf3f.linear( mrmeshpy.Matrix3f.scale( box.diagonal() * 3 ) )
planeXf = mrmeshpy.AffineXf3f.linear( mrmeshpy.Matrix3f.rotation( mrmeshpy.Vector3f(0,0,1), plane.n ) ) * planeXf
centerDistance = mrmeshpy.dot( plane.n, box.center() ) - plane.d
centerProj = box.center() - centerDistance / plane.n.lengthSq() * plane.n
planeXf = mrmeshpy.AffineXf3f.translation( centerProj )  * planeXf

# move plane mesh to position
planeMesh.transform( planeXf )

# cut mesh
cutOnePart = mrmeshpy.boolean( mesh, planeMesh, mrmeshpy.BooleanOperation.InsideA  ).mesh
cutOtherPart = mrmeshpy.boolean( mesh, planeMesh, mrmeshpy.BooleanOperation.OutsideA  ).mesh

Fedr, 2024-02-24:
Actually we have something even now. Please try:
```
mrmeshpy.cutMeshWithPlane(mesh, plane);

closed 2024-03-05 by Grantim