MeshInspector/MeshLib #2375: Create a 3D Mesh from a list of vertices and faces.

marios-stam, 2024-03-10:
Hello,

I apologise if this is not the correct place to ask but am trying to test my algorithm with this library, and although it solves some of my problems compared to others, I am having some trouble creating a new mesh. Specifically, given a list of vertices and faces, I want to create a new Mesh object.

I am using Python and although I think I load the vertices properly, it is not clear how I should load the topology (faces).

```Python3
vertices = []
for i,v in enumerate(meshTri.vertices):
    vertices.append([mn.Vector3f(v[0],v[1],v[2]),i])

faces = meshTri.faces
mesh = mm.Mesh()
mesh.points = vertices
```

Could you please help me?

Fedr, 2024-03-10:
Hello,

We have an example how to create new mesh object from two arrays of vertices and faces in the documentation: https://meshinspector.com/MeshLib/doc/HowToBasic.html#PythonCodeExamples

```
import meshlib.mrmeshpy as mrmeshpy
import meshlib.mrmeshnumpy as mrmeshnumpy
import numpy as np
 
faces = np.ndarray(shape=(2, 3), dtype=np.int32, buffer=np.array([[0, 1, 2], [2, 3, 0]], dtype=np.int32))
 
# mrmesh uses float32 for vertex coordinates
# however, you could also use float64
verts = np.ndarray(shape=(4, 3), dtype=np.float32, buffer=np.array([[0.0, 0.0, 0.0], [1.0, 0.0, 0.0], [1.0, 1.0, 0.0], [0.0, 1.0, 0.0]], dtype=np.float32))
 
mesh = mrmeshnumpy.meshFromFacesVerts(faces, verts)
```

Does it work for you?

marios-stam, 2024-03-10:
Hello, and thank you for the quick response.

The solution provided partially addressed my issue, but I've observed an anomaly.

I'm encountering a situation where the mesh, generated from another library, is not entirely watertight. As you suggested, I'm loading it as a meshLib object and attempting to fill in the gaps using the recommended method, but without success.

Interestingly, saving the mesh as a .stl file and then loading it back using meshlib seems to resolve the issue, as it becomes watertight afterward.

It appears that there are additional functionalities available when loading a mesh from an .stl file compared to simply loading it from a list of vertices and faces.

Do you maybe have any idea how to address this?

Fedr, 2024-03-10:
In STL files, there are no vertex IDs stored, but only coordinates of vertices. So if you had two distinct vertices with equal coordinates before saving them in STL format, then after loading the mesh from this STL, such vertices will unite in one. This may explain your observations.

You can also try `mrmeshpy.uniteCloseVertices` function to solve your not-watertight meshes.

If nothing helps, please send an example of such mesh to us for better understanding of the problem.

closed 2024-05-15 by Fedr