MeshInspector/MeshLib #2413: Hole & Boundary edge length in Python

emil-peters, 2024-03-15:
Hi,

I know we can get all `HoleRepresentiveEdges` using  `mr_mesh.topology.findHoleRepresentiveEdges()` which returns  a list `vectorEdges[...]`.

There is also the `mr_mesh.topology.findBoundaryEdges()` to get all boundary edges.

I was now wondering if there is an option already exposed to:
- starting from a `vectorEdge` get the entire list of vertices/edges that make up the path on the mesh of the hole
- Getting the length of the hole

As I want to be able to use `fillHole` on all small holes in the mesh but not on the big hole as my mesh is not watertight but a shell.

If you could point me on how to expose these things, I'd be happy to help with the exposing :smile: 

Kind regards

ps. It would be nice to have a python release soon as well.

Grantim, 2024-03-15:
Hello!

You can find this example helpful
https://github.com/MeshInspector/MeshLib/blob/adf6cc3f88bf85116f5431f0cdf56ee2419559e4/python_scripts/healer.py#L3-L18

emil-peters, 2024-03-15:
Super thanks!

Any idea if there is also a method to get all indices or coords of all verts in such a hole?
Or would it be better for now to just manually loop the edge and store them?

Grantim, 2024-03-15:
Now there is no function in python to get `VertBitSet` from `EdgeLoop`. I think only way now is iterating and storing them, btw do you need `VertBitSet` or vector of `VertId`?

emil-peters, 2024-03-15:
Both are fine as I can do `np.nonzero(VertBitSet)` to get the `vertIds`

But I'm guessing VertBitSet is easier, as it can easily be transformed to a `List[int]` in python

Grantim, 2024-03-15:
Now you can store them like this
```py
from meshlib import mrmeshpy as mm

# load mesh
mesh = mm.Mesh()
# ...

vertsBS = mm.VertBitSet()
vertsBS.resize( mesh.topology.getValidVerts().size(), False )

holes = mm.findRightBoundary( mesh.topology )
for hole in holes:
    for e in hole:
        vertsBS.set( mesh.topology.org( e ) )