MeshInspector/MeshLib #2418: The Angle between two vectors is 0 to pi, why not 0-2pi

poodkcc, 2024-03-16:
I saw the code.
It's just my question, not a code bug.
```
/// angle in radians between two vectors
template <typename T> 
inline T angle( const Vector3<T> & a, const Vector3<T> & b )
{
    return std::atan2( cross( a, b ).length(), dot( a, b ) );
    // this version is slower and less precise
    //return std::acos( std::clamp( dot( a.normalized(), b.normalized() ), T(-1), T(1) ) );
}
```
std::atan2(sin,cos); sin is always greater than 0
So it's not going to be a negative Angle or a pi minus 2pi.
I think this Angle has lost its direction.

I hope you can point out that I'm wrong, or I'm right, and some other function implements it.
I've been thinking about this problem for two days, and I want someone to help me.

Sorry English is not my first language, the translator helped me.

Fedr, 2024-03-16:
Hello, and thank you for the question.

Indeed the angle returned by the function is always within [0,pi], and it is not "directed" in the sense that `angle(a,b) = angle(b,a)`.

So "angle" here is a minimal angle between two directions `a` and `b`.

[image]

Please let me know if more clarification is required.

poodkcc, 2024-03-18:
@Fedr 
Hello, and thank you for your answer.
If I have a faux pas, please point it out and I will try to correct it.

$\vec A \times \vec B = \vec n|\vec A||\vec B|\sin \theta$
$\vec B \times \vec A = -\vec n|\vec A||\vec B|\sin \theta$
$|\vec B \times \vec A| = |\vec A||\vec B|\sin \theta$
$|\vec A \times \vec B| = |\vec B \times \vec A|$

In my understanding, the right-handed coordinate system is x cross times y to produce z, and conversely y cross times x to produce -z. I think this is called the direction, and this direction is very useful for rotation.
When you take the modulus of the cross product, the vector gets rid of its direction, or the part that's less than zero gets rid of its direction.
In the two-dimensional Cartesian coordinate system, the Angle of [pi,2pi] is discarded.

I think the point of using the atan(b,a) function is that it returns [0,2pi] instead of the disoriented [0,pi], and you can get the same effect if you use acos(cos).
$arctan(\frac{sin(\theta)}{cos(\theta)}) =  \alpha ,\theta = \alpha,\theta \in [0,2\pi],cos(\theta)\neq0$
$arctan(\frac{|sin(\theta)|}{cos(\theta)}) =  \alpha ,\theta \neq \alpha,\alpha\in [0,\pi],cos(\theta)\neq0$
$arccos(cos\theta) = \alpha,\theta \neq \alpha,\theta\in [0,2\pi],\alpha\in[0,\pi]$
Maybe it's better because atan2 is faster, but the function itself returns the direction, and apparently it bothers me that the author dropped the direction attribute.

Fedr, 2024-03-18:
Hello, @poodkcc 

all your formulas are correct. And it was our goal to get the "undirected" angle in $[0,\pi]$.

So we use the code
```
std::atan2( cross( a, b ).length(), dot( a, b ) );
```
as a faster and more precise replacement of
```
std::acos( std::clamp( dot( a.normalized(), b.normalized() ), T(-1), T(1) ) );
```
Both of them return the angle in $[0,\pi]$, ignoring relative orientation of `a` and `b`, because such behavior is expected in our algorithms.

I agree with you that sometimes another angle-function is required, returning the full angle in $[0,2\pi]$. And we really use them, please find other calls to `std::atan2` in our codebase.

poodkcc, 2024-03-18:
Hello, @Fedr 
Thank you for your answer, you solved my problem, I wish you a happy life, distant friend.

closed 2024-03-18 by poodkcc