MeshInspector/MeshLib #2510: WSL Ubuntu 22 build issue: Target "MRCuda" requires the language dialect "CUDA20"

ziy-zhao, 2024-04-03:
Hi, I tried to run `./script/build_source.sh` in my WSL Ubuntu 22.04 and encountered following error. The cuda_std_20 cannot be found but cuda_std_03, cuda_std_11, cuda_std_14 and cuda_std_17 can be found.
[image]
- I have installed Nvidia driver 537.79 on the Windows and set up the CUDA tool kit 12.4 for Ubuntu. 
- cmake version 3.22.1
- g++ version 11.4.0
[buildLog.txt](https://github.com/MeshInspector/MeshLib/files/14850624/buildLog.txt)

MaxRayskiy, 2024-04-06:
Hi! Thank you for this issue

It seems that the problem is related to Ubuntu 22's default CMake version not supporting 'CUDA 20' dialect.
 
 CMake 3.25.2 [changelog](https://cmake.org/cmake/help/latest/release/3.25.html#id19):
`CUDA language level 20 (corresponding to C++20) is now supported with NVCC 12.0 and above.`
 
 I was managed to build this Dockerfile:
 
 ```
FROM ubuntu:22.04

RUN apt-get update -y && \
    apt-get install -y sudo git wget

# install CUDA tool kit 12.4
RUN wget https://developer.download.nvidia.com/compute/cuda/repos/wsl-ubuntu/x86_64/cuda-wsl-ubuntu.pin
RUN sudo mv cuda-wsl-ubuntu.pin /etc/apt/preferences.d/cuda-repository-pin-600
RUN wget https://developer.download.nvidia.com/compute/cuda/12.4.1/local_installers/cuda-repo-wsl-ubuntu-12-4-local_12.4.1-1_amd64.deb
RUN sudo dpkg -i cuda-repo-wsl-ubuntu-12-4-local_12.4.1-1_amd64.deb
RUN sudo cp /var/cuda-repo-wsl-ubuntu-12-4-local/cuda-*-keyring.gpg /usr/share/keyrings/
RUN sudo apt-get update -y
RUN sudo apt-get -y install cuda-toolkit-12-4

RUN git clone https://github.com/MeshInspector/MeshLib

WORKDIR MeshLib
RUN git submodule update --init --recursive

RUN ./scripts/install_apt_requirements.sh
ENV MR_STATE="DOCKER_BUILD"
RUN  ./scripts/build_thirdparty.sh

# update cmake to latest
RUN sudo apt remove --purge --auto-remove -y cmake
RUN sudo apt update -y && \
        sudo apt install -y software-properties-common lsb-release && \
        sudo apt clean all
RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | sudo tee /etc/apt/trusted.gpg.d/kitware.gpg >/dev/null
RUN sudo apt-add-repository "deb https://apt.kitware.com/ubuntu/ $(lsb_release -cs) main"
RUN sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 6AF7F09730B3F0A4
RUN sudo apt update -y
RUN sudo apt install -y cmake

RUN  ./scripts/build_source.sh
 ```

Sorry for such a mess in the Dockerfile, but it works

oitel, 2024-10-17:
The build of the CUDA component with older CMake versions has been fixed in the latest MeshLib version, please check it out: https://github.com/MeshInspector/MeshLib/releases/tag/v3.0.0.40

closed 2024-10-17 by oitel