MeshInspector/MeshLib #2535: Issue with `vectorEdges` in Python (Windows)

amin-kchaou, 2024-04-08:
Hi,

I have a small python script that loads an STL mesh with holes and attempts to repair those holes:

```python
holes = mesh.topology.findHoleRepresentiveEdges()
for hole in holes:
    mr.fillHole(mesh, hole)
```

This script silently crashes after the last iteration of the loop, or immediately after entering the loop context when the holes array is empty. I believe it has something to do with `vectorEdges`'s  iterator implementation.

When using `faulthandler`, I get this error pointing to the `for` loop line: `Windows fatal exception: access violation`

One workaround is to use a normal `for` loop like this:
```python
holes = mesh.topology.findHoleRepresentiveEdges()
for i in range(len(holes)):
    mr.fillHole(mesh, holes[i])
```

### System information:
- OS: Windows 11 Pro
- Python version: 3.11.8
- Installed MeshLib version: 2.3.8.21

Let me know if you need any more info.

Grantim, 2024-04-08:
Hello! 

Do you import other libraries? We have seen something similar here #1824

amin-kchaou, 2024-04-08:
Thank you for linking the other issue. I was indeed importing something else before the meshlib import. In the event viewer I found this:

[image]

Seems to be caused by `open3d`, judging from the dll path: `...\venv\Lib\site-packages\open3d\cpu\pybind.cp311-win_amd64.pyd`

However, shuffling the imports around doesn't fix this. Only when I remove my import that meshlib works again. `open3d` is needed in my application and it will be very difficult to remove. Is there a way around this?

Grantim, 2024-04-08:
Can you please attach minimal example so we can reproduce the issue on our end?

closed 2024-04-22 by amin-kchaou