MeshInspector/MeshLib #2587: Add overload to find closest intersections for computeSkyViewFactor

mariuszhermansdorfer, 2024-04-15:
Hi @Fedr & @Grantim,

could we please add an overload to the `computeSkyViewFactor` functions to return the closest intersections? I understand that it could slow things down that's why it's set to false by default, but in certain situations I care about finding the closest point. One example is a view analysis, where I want to visualize the field of view from a specific point. Here you can see that sometimes the backfaces are returned as hit points.

[image]

https://github.com/MeshInspector/MeshLib/blob/523fb4679100e22d5a474c378bd8dc99f6cd9189/source/MRCuda/MRCudaSolarRadiation.cpp#L84-L87

https://github.com/MeshInspector/MeshLib/blob/523fb4679100e22d5a474c378bd8dc99f6cd9189/source/MRMesh/MRSolarRadiation.cpp#L51

mariuszhermansdorfer, 2024-04-15:
Hmm, it might be even simpler. Each time we return std::vector<MR::MeshIntersectionResult>* outIntersections, these should be always the closest intersection. If we only care about occlusion tests, we wouldn’t return the intersection points.

closed 2024-04-17 by egrebenchenko

Grantim, 2024-04-17:
Hello! @egrebenchenko has update the functions, so in new version of meshlib it should return closest intersection if `outIntersections` is not nullptr

mariuszhermansdorfer, 2024-04-17:
Thanks @Grantim, I can see the commit addressed the CPU logic but not the Cuda version of this function. Could this be changed as well?

Grantim, 2024-04-17:
https://github.com/MeshInspector/MeshLib/commit/555e43c243b2b281a651e2dccea0c55d311f8cdc here it is

mariuszhermansdorfer, 2024-04-17:
Super! Thanks a million!