MeshInspector/MeshLib #2598: Anisotropic offsetting

tobias-scheepers, 2024-04-16:
Hi all, I'm very curious if anyone has made any attempt to make an anisotropic offset operation using meshlib. As I see it there is a set of two operations that would be very valuable but not seen very often in practice:

- Offsetting a shape in a single direction
- Offsetting a shape in all but a single direction (e.g. in-plane offsetting)

I'm not sure if this is something that could be made on-top any of the current functions or would require more low level changes to be made.

Grantim, 2024-04-16:
Hello!

Now I think we only have `fixUndercuts` function that is similar to  `Offsetting a shape in a single direction` 

<image width=300px src="https://github.com/MeshInspector/MeshLib/assets/10034350/f745947c-9ee8-4dc3-bf39-ec5ea2f3ed39"> <image width=300px src="https://github.com/MeshInspector/MeshLib/assets/10034350/caf2e475-fda3-40be-a53f-768efb03bfe7">

Can you please show some example of input and expected output?

tobias-scheepers, 2024-04-17:
Hi @Grantim the fix undercut feature is indeed help full in a lot of cases. The offsetting in a certain direction would however allow for some more fine grained operations in some cases:

Let's say the original mesh looks like so:
[image]

I believe we could create an offset function that if working symmetrically along the z-axis would look like so:
[image]

Once the offset is large enough to cover the gap it would avoid the self intersections as we are used to from your other offsetting operations:
[image]

Similarly one could make an operation to only work in a single direction:
[image]
[image]

And finally one could create an offset that would work in the plane (e.g. the XY plane). In this case I would expect something like so:
[image]
[image]

I'm not sure if this makes sense so far. I believe these operations might be a valuable operation for meshlib. But I'm not well enough informed to see how difficult it would be to implement. It seems the offsets in a single direction might be feasible with some smart segmentation and extrusion operations. I believe however it is the offsetting in plane that would be most valuable.

Grantim, 2024-04-17:
Thanks for the explanation! 

Unfortunately we don't have such functionality now, but I think we could implement it in future.

tobias-scheepers, 2024-04-18:
That sounds amaing @Grantim I'll keep a close eye on this.

tobias-scheepers, 2024-07-01:
Hi @Grantim any chance you have still been looking into this?

Grantim, 2024-07-01:
Hi @tobias-scheepers! Unfortunately, we had no time to look at this yet, thanks taking it up, we will discuss it with the team and come back here.

DmitryLambin, 2024-07-01:
Hi @tobias-scheepers.

As we discussed last time, we are ready to plan this feature's implementation as soon as you are ready to proceed with the project using MeshLib. Thanks for revisiting this topic. Don't hesitate to contact me to synchronize the schedule of this feature with your project needs.

Fedr, 2024-08-02:
In the last release of MeshInspector, one can find anisotropic Offset. Here is an example: a sphere is offset along Y twice more than along X, and along Z twice more than along Y, resulting in an ellipsoid:
[image]

tobias-scheepers, 2024-08-14:
@Fedr, @Grantim that sounds great! Has it also been exposed to python yet by any chance. I'll update the little tests we were running and report back!

@DmitryLambin will let you know once we pick this back up. It's moved a little to the back-burner for us. But will will keep an eye on meshlib as I'm sure we'll be crossing paths at some time in the future.

Fedr, 2024-08-14:
Thanks. At this moment the implementation is located within MeshInspector's tool only. Please give it a try. If everything is ok, we will work on moving it in MeshLib and python interface.

tobias-scheepers, 2024-08-15:
@Fedr thank you for the update. Inside MeshInspector it works like a charm. If this is available in meshlib/python I'll have a look if we can use it for my usecase. I had three questions that came up while playing with the new feature:

- Is it possible to have the anisotropic scaling factors be allowed to be below 1 (and even 0)? They now seem limited to [1, 10]
- Is it possible to have a custom aligned coordinate system for the anisotropic scaling factors? Or would you suggest to rotate the entire mesh and then use the current orthogonal directions?
- Is it possible to define 6 scaling factors for X+, X-, Y+, Y-, Z+ and Z-?

For all, I'm not sure if there is a technical limit in the implementation that guided the current choices. Anyway, great work on this!

Fedr, 2024-08-15:
Currently, the algorithm for anisotropic offsetting in MeshInspector works as follows:
1. Perform not-uniform scaling with different factors along X, Y, Z axes, where each factor is inversely proportional to desired offset in that direction.
2. Perform standard uniform offset.
3. Perform not-uniform scaling with reversed factors to step 1.

Although there is no single function in MeshLib to perform all steps at once, you can write python program that reaches the same result.

Answering your questions:
* Since on step 1. of the algorithm the factors are inversely proportional to desired final offset, too small desired values will result in huge scalings (and memory consumption during uniform offsetting). So we had to limit them in UI.
* Custom aligned coordinate system shall be easy, just rotate your body before step 1. to get desired axes as X,Y,Z; and after step 3. rotate it back.
* With the current implementation of anisotropy, independent scalings on e.g. +X and -X are not possible, we will need to devise another solution.

lslzl3000, 2025-09-20:
@Fedr  in 3D printing, especially for FDM or  SLA printers, they need apply Z direction offset called `Z-Compensation`. I found two industry software to apply something like independent offsets in +Z/-Z, and they not only add uniform scale in z axis, but also based on a angle factor for curved surface

one is Autodesk:
[image]
[image]

one is Magics, one of the 3D printing editer:
[image]
[image]
[image]

in our project, we need to do `Z-Compensation` on models. for some very simple cases, it's easy to add a offset to faces towards down direction:

[image]

but for real world models,  faces can be complicated, add offset to a face will lead to flip normal/self intersections/non-manifold/.... hard to fix then. It would be much better for `MeshLib` could add features to handle `Z-Compensation` in simple way.

DmitryLambin, 2025-09-20:
@lslzl3000 Thank you for sharing the implementation examples. This feature is important for 3D printing, but it fits more into our long-term roadmap than the short term. If you are planning any commercial use of our library with this feature, we can adjust priorities. Let’s discuss: https://meshlib.io/book-a-call/