MeshInspector/MeshLib #2624: mrmeshpy.decimateMesh question

Frq-F, 2024-04-19:
Hello , I have a question about mrmeshpy.decimateMesh 
I want to simplify the mesh，My code:

def prepare(mesh, num_faces_set):
    
    try:
        components_face = mrmeshpy.getAllComponents(mesh)
    except ValueError as e:
        my_log.error(e)
        return False

    for c in range(1, components_face.size()):
        mesh.topology.deleteFaces(components_face[c])

    # Check the number of faces after removal
    num_out_faces = mrmeshnumpy.getNumpyFaces(mesh.topology).size // 3
    if num_out_faces > num_faces_set:
        # Simplification settings
        settings = mrmeshpy.DecimateSettings()
        settings.maxDeletedFaces = num_out_faces - num_faces_set        
        settings.maxError  = 1

        # Simplify the mesh
        try:
            mrmeshpy.decimateMesh(mesh, settings)
            out_verts = mrmeshnumpy.getNumpyVerts(mesh)
            out_faces = mrmeshnumpy.getNumpyFaces(mesh.topology)
            print('new',out_verts.shape,out_faces.shape)
    
        except ValueError as e:
            my_log.error(e)
            return False
    return True

When I want to check the number of points and faces in the simplified mesh, but the output result is incorrect, it is the result of the mesh that has not been simplified。
code：
out_verts = mrmeshnumpy.getNumpyVerts(mesh)
out_faces = mrmeshnumpy.getNumpyFaces(mesh.topology)
print('new',out_verts.shape,out_faces.shape)

How can I solve this problem?Thanks

Fedr, 2024-04-19:
Hello,

I guess this is because by default the mesh after decimation is not packed, so many of faces and vertices in the arrays are not really used.

Please try to add
```
settings.packMesh = True
```

Does it solve your issue?

Frq-F, 2024-04-22:
Thanks, it solve my issue?
I have another question? How to downsample the pointcloud for python code?

Frq-F, 2024-04-22:
I try to add  settings.packMesh = True
but I check out_faces = mrmeshnumpy.getNumpyFaces(mesh.topology), The number of faces has decreased, but it has not decreased to the specified quantity.
I have another question : How to downsample pointcloud for python code?

Fedr, 2024-04-22:
> I try to add settings.packMesh = True
but I check out_faces = mrmeshnumpy.getNumpyFaces(mesh.topology), The number of faces has decreased, but it has not decreased to the specified quantity.

It may be because `settings.maxError  = 1`, which forces mesh decimation to stop earlier. Please try to increase `settings.maxError`.

> I have another question : How to downsample pointcloud for python code?

Please use python functions `pointGridSampling` or `pointUniformSampling` for that.

closed 2024-05-03 by Fedr