MeshInspector/MeshLib #2672: Installation via package manager using pip-24.0 & python 3.12.1 not successfull

mSamiolo, 2024-05-02:
## Error

The command: 

```sh
pip install meshlib
```
return 

```sh   
ERROR: Could not find a version that satisfies the requirement meshlib (from versions: none)
ERROR: No matching distribution found for meshlib
```

## Environment replication

```sh
spack install python@3.12 
spack load python@3.12
python -m venv new_env
source ./new_env/bin/activate
pip install meshlib 
```
Same error

```sh   
ERROR: Could not find a version that satisfies the requirement meshlib (from versions: none)
ERROR: No matching distribution found for meshlib
```

mSamiolo, 2024-05-02:
To enrich the above: the same steps but with ```python@3.9``` works without issues

Fedr, 2024-05-02:
Hello,

Please tell us what is your operating system?

mSamiolo, 2024-05-02:
It looks like agnostic from systems, that's why I used ```spack``` to isolate the problem, nonetheless the problem arise in:

```sh
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```

and 

```sh
Distributor ID:	Amazon
Description:	Amazon Linux release 2 (Karoo)
Release:	2
Codename:	Karoo
```

N.B. command used for the above: ```lsb_release -a```

Fedr, 2024-05-02:
I see. Indeed, till this moment we supported MeshLib installation via pip for python 3.12 only on Windows and macOS, but not on Linux. We will work on supporting it as well.

mSamiolo, 2024-05-02:
I realized now that the problem is more tricky, for example using ```python@3.9.18``` in Ubuntu  on x86_64, it works out of the box, while if I use the same configuration on the Amazon Linux I got the same error even if I use  ```python@3.12.xxx``` or ```python@3.7.xxx```:

```sh
pip install meshlib

ERROR: Could not find a version that satisfies the requirement meshlib (from versions: none)
ERROR: No matching distribution found for meshlib
```

Then I then though to use [your wheel](https://files.pythonhosted.org/packages/fe/78/2adf8b2b41eedfc26f02d9b8c3e6f1a6cab81752bf8be6bba4c49e232bf4/meshlib-2.4.0.158-cp38-none-manylinux_2_31_aarch64.whl) , but the returned error on Amazon Linux (RHEL based on x86_64) 

```sh
pip3 install meshlib-2.4.0.158-cp38-none-manylinux_2_31_x86_64.whl 

ERROR: meshlib-2.4.0.158-cp38-none-manylinux_2_31_x86_64.whl is not a supported wheel on this platform.
```

If you think the solution is that you need only additional to support it, and nothing is wrong I am happy to close the issue.

Thanks in advance!!!

Fedr, 2024-05-02:
At this moment we build python wheels as follows for manylinux_2_31_x86_64:
* Python 3.8
* Python 3.9
* Python 3.10
* Python 3.11

We plan to support Python 3.12 as well, but not Python 3.7.

mSamiolo, 2024-05-02:
I do get trouble to install your wheel manually

```sh
uname -m       ## Return x86_64
spack load python@3.9.18
python3 -m venv geometry_kernel-ENV-3.9
source geometry_kernel-ENV-3.9/bin/activate
python --version  ## Return      Python 3.9.18
which python       ## Return    ~/geometry_kernel-ENV-3.9/bin/python
which pip             ## Return    ~/geometry_kernel-ENV-3.9/bin/pip
pip install ./meshlib-2.4.0.158-cp38-none-manylinux_2_31_x86_64.whl 
```

return:

```sh
ERROR: meshlib-2.4.0.158-cp38-none-manylinux_2_31_x86_64.whl is not a supported wheel on this platform.
```

Are you seeing error that I cannot see?  
The wheel does not work on both Ubuntu and AmazonLinux. (I am wondering is this has to be a separate issue)

MaxRayskiy, 2024-05-02:
> ```shell
> meshlib-2.4.0.158-cp38-none-manylinux_2_31_x86_64.whl
> ```

You are trying to install wheel for python 3.8 - `cp38`. Try this one for python3.9 please: [
meshlib-2.4.0.158-cp39-none-manylinux_2_31_x86_64.whl ](https://files.pythonhosted.org/packages/de/62/cfe2375d4c30375181b135545378b136d1c5a1cf0fbc4ae3e6214dc39714/meshlib-2.4.0.158-cp39-none-manylinux_2_31_x86_64.whl)

mSamiolo, 2024-05-02:
I do have to say that is not really working @MaxRayskiy but many thanks for the correction. Let me post the replication of the error:

```sh
(geometry_kernel-ENV-3.9) [ec2-user@ip-10-0-0-221 ~]$ which pip
~/geometry_kernel-ENV-3.9/bin/pip
(geometry_kernel-ENV-3.9) [ec2-user@ip-10-0-0-221 ~]$ which python
~/geometry_kernel-ENV-3.9/bin/python
(geometry_kernel-ENV-3.9) [ec2-user@ip-10-0-0-221 ~]$ python --version
Python 3.9.18
(geometry_kernel-ENV-3.9) [ec2-user@ip-10-0-0-221 ~]$ pip install ./meshlib-2.4.0.158-cp39-none-manylinux_2_31_x86_64.whl 
ERROR: meshlib-2.4.0.158-cp39-none-manylinux_2_31_x86_64.whl is not a supported wheel on this platform.
```

The system is x86_64, and I tried to load  a newer version of glibc

```sh
spack load glibc@2.36 
```

but it fails the same, and not knowing the details of your build system I do not have any oher idea to have support on this machine

MaxRayskiy, 2024-05-02:
I have checked glibc version for Amazon Linux 2:
```
$ Idd --version
Idd (GNU libc) 2.26
```
and for Amazon Linux 2023:
```
$ ldd --version
ldd (GNU libc) 2.34
```

Our wheels are for `manylinux_2_31+` (they support only glibc2.31+ due to C++ compiler limitations). Therefore, they are only compatible with Amazon Linux 2023 and later versions. If you want to use meshlib wheel, you will need to update your OS to Amazon Linux 2023, sorry

mSamiolo, 2024-05-02:
Many thanks @Max I will work around with a docker instance.  Really helpfull

closed 2024-05-02 by mSamiolo

Fedr, 2024-05-03:
We have just released MeshLib 2.4.1.114, which adds support for python 3.12 on many modern Linux'es including Ubuntu 20+ and Fedora 32+.