MeshInspector/MeshLib #2780: Compilation Error in MRPch Project with Eigen3

feifeizuo, 2024-05-29:
Operating System: Windows 11 23H2
CUDA Version: 12.0
Visual Studio Version: 2022 17.9.6
Issue Description:
When attempting to compile the solution project MRPch, I encounter the following error:

[image]

Additional Information:
Eigen3 Version: 3.4.0
I have tested Eigen3 in other projects, and it works without any issues.
Could you please provide guidance on how to resolve this compilation error?
Thank you!

Grantim, 2024-05-29:
Hello!

MeshLib solution is configured to treat warnings as errors, so we manually suppress warnings from thirdparty libraries. Looks like version of Eigen is different from the we that we use, so there is warning that is not suppressed here 
https://github.com/MeshInspector/MeshLib/blob/18f2645600f0ae9cb15230e0ca5765361dc08657/source/MRPch/MREigen.h#L5-L32

Is you vcpkg version same as we have in instructions? https://github.com/MeshInspector/MeshLib/blob/18f2645600f0ae9cb15230e0ca5765361dc08657/readme.md?plain=1#L160-L169

feifeizuo, 2024-05-30:
Thanks for you reply.
I recently reinstalled vcpkg (version 2024.04.26) and ran thirdparty/install.bat (all library download and compile again). The version of eigen3 installed is 3.4.0. However, I am still encountering compilation errors.

[image]

[image]

what version of eigen3 is correct ?

Any assistance in resolving this issue would be greatly appreciated. Thank you!

Fedr, 2024-05-30:
Could you please try [this](https://stackoverflow.com/a/37871934/7325599) solution in Visual Studio:
```
Control Panel -> Clock,Language,and Region -> Region and Language -> 
Administrative -> Language for non-Unicode programs -> Change system locale -> English

feifeizuo, 2024-05-30:
Fantastic! The system locale and region settings were the crucial factors.