MeshInspector/MeshLib #2865: depth test problem

feifeizuo, 2024-06-20:
Hello, 
I attempted to implement distance measurement and found that the depth test option was ineffective regardless of whether it was enabled or not. Additionally, I also tried using the viewport.drawpoints() function, but the depth test was still ineffective. How can this be resolved?

some code：
` distanceMeasurementObject_ = std::make_shared<MR::DistanceMeasurementObject>();
 distanceMeasurementObject_->setLocalPoint( firstPos_ );
 distanceMeasurementObject_->setName(tr("Distance measurement") + std::to_string(++distanceIdx_ ));
 distanceMeasurementObject_->setVisualizeProperty(true, VisualizeMaskType::DepthTest, ViewportMask::all));
 obj->addChild( distanceMeasurementObject_ );`

` PointPairs sourcePairPoints = icp_->getFlt2RefPairs();

 std::vector<Vector3f> srcPoints;

 for ( auto& pair : sourcePairPoints.vec )
 {
     srcPoints.push_back( pair.srcPoint );
 }

 if ( srcPoints.size() > 0 )
 {
     viewport.drawPoints( srcPoints, std::vector<Vector4f>{srcPoints.size(), Vector4f( 1, 0, 0, 1 )}, 5.0f, true);
 }`

Grantim, 2024-06-20:
Hello! 

To render `DistanceMeasurementObject` we use UI library Dear ImGui, and it is always drawn with depth test disabled (VisualizeMaskType::DepthTest this property does not affect it in any way). But `viewport.drawPoints` should respect the it, so it may be a bug. Could you please provide some data, so we could reproduce it.

>Please note that `DistanceMeasurementObject` draws lines and `viewport.drawPoints` only draws points. If you want to draw lines with depth test you should either use `ObjectLines` or `viewport.drawLines` . Also, please note that you should call `viewport.draw...` function in each frame.

feifeizuo, 2024-06-20:
The following is the complete sample code. The attachment contains a video.  

```
#include "MRViewer/MRStatePlugin.h"
#include "MRViewer/MRRibbonMenu.h"
#include "MRViewer/MRViewport.h"

https://github.com/MeshInspector/MeshLib/assets/6001243/91ce574c-671e-4596-ace7-8252b1eaaf8c

#include "MRViewer/MRViewer.h"
#include "MRMesh/MRVector.h"

class HelloWorldPlugin : public MR::StatePlugin
{
public:
    HelloWorldPlugin():
        MR::StatePlugin( "Hello World" )
    {
        // random add 10 points
        for ( int i = 0; i < 10; ++i )
            srcPoints.push_back( MR::Vector3f( ( float )rand() / RAND_MAX * 10.0f, ( float )rand() / RAND_MAX, ( float )rand() / RAND_MAX * 10.0f) );
    }
    virtual void drawDialog( float menuScaling, ImGuiContext* ) override
    {
        if ( !ImGuiBeginWindow_( { .collapsed = &dialogIsCollapsed_,.width = 200.0f * menuScaling,.menuScaling = menuScaling } ) )
            return;
        ImGui::TextWrapped( "Hello World!" );
        ImGui::EndCustomStatePlugin();

        drawPoints();
    }

    private:

        std::vector<MR::Vector3f> srcPoints;

        void drawPoints()
        {
            auto& viewport = viewer->viewport();

            if ( srcPoints.size() > 0 )
            {
                viewport.drawPoints( srcPoints, std::vector<MR::Vector4f>{srcPoints.size(), MR::Vector4f( 1, 0, 0, 1 )}, 5.0f, true );
            }
        }
};

MR_REGISTER_RIBBON_ITEM( HelloWorldPlugin )
```

Grantim, 2024-06-20:
Thanks! Now I see what is the problem,

We use separate texture to render scene objects, while UI is rendered directly on the screen. `drawDialog` function is used to draw UI, so all opengl calls that are done here goes to screen directly ignoring scene texture and its depth buffer.

This way it will points will be rendered in the scene texture:
```c++
#include "MRViewer/MRStatePlugin.h"
#include "MRViewer/MRRibbonMenu.h"
#include "MRViewer/MRViewport.h"
#include "MRViewer/MRViewer.h"
#include "MRMesh/MRVector.h"

class HelloWorldPlugin : public MR::StateListenerPlugin<DrawListener>
{
public:
    HelloWorldPlugin():
        MR::StatePlugin( "Hello World" )
    {
        // random add 10 points
        for ( int i = 0; i < 10; ++i )
            srcPoints.push_back( MR::Vector3f( ( float )rand() / RAND_MAX * 10.0f, ( float )rand() / RAND_MAX, ( float )rand() / RAND_MAX * 10.0f) );
    }
    virtual void drawDialog( float menuScaling, ImGuiContext* ) override
    {
        if ( !ImGuiBeginWindow_( { .collapsed = &dialogIsCollapsed_,.width = 200.0f * menuScaling,.menuScaling = menuScaling } ) )
            return;
        ImGui::TextWrapped( "Hello World!" );
        ImGui::EndCustomStatePlugin();

        //drawPoints();
    }

    private:

        std::vector<MR::Vector3f> srcPoints;
        void drawPoints()
        {
            auto& viewport = viewer->viewport();

            if ( srcPoints.size() > 0 )
            {
                viewport.drawPoints( srcPoints, std::vector<MR::Vector4f>{srcPoints.size(), MR::Vector4f( 1, 0, 0, 1 )}, 5.0f, true );
            }
        }

        
        virtual void draw_() override
        {
            drawPoints();
            // note that if points changed you need to call:
            // viewer->setSceneDirty();
        }
};

MR_REGISTER_RIBBON_ITEM( HelloWorldPlugin )
```

closed 2024-06-20 by feifeizuo