MeshInspector/MeshLib #2899: Upload a closed triangular mesh to MeshInspector but it displays mesh with holes

citystrawman, 2024-06-27:
Hi, I am just learning MeshLib and I want to upload a model and try to remesh it. This model is a closed model, as can be seen in meshlab:

[image]

However, when I upload the model to MeshInspector, it displays mesh with holes:

[image]

I am not sure if there's anything wrong in my model? 

I attached my off file here:
[slope_slide_clip_merge.zip](https://github.com/user-attachments/files/16010930/slope_slide_clip_merge.zip)

Fedr, 2024-06-27:
Thanks for the question.

In your mesh there a lot of non-manifold vertices:
[image]

MeshLib's topology is designed to represent only manifold meshes, so some triangles violating manifoldness are skipped during file loading.

Basically, there is some error in the software that produced such wrong cut in the mesh.

Still, you can fix that mesh using MeshHealer/Rebuild tool in MeshInspector:
[slope_slide_fixed.ply.zip](https://github.com/user-attachments/files/16011546/slope_slide_fixed.ply.zip)

citystrawman, 2024-06-27:
> Thanks for the question.
>
> In your mesh there a lot of non-manifold vertices: [...]

Thank you! I made this model non-manifold because I want to simulate layers of the rock, and the fixed model seems to merge the two layers into one. Is there any way that can simulate layers of rock./soil and still maintaining manifold?

Fedr, 2024-06-27:
Could you please elaborate your task and provide input data (rock before modification etc.)? Then we probably could suggest something.

citystrawman, 2024-06-27:
> Could you please elaborate your task and provide input data (rock before modification etc.)? Then we probably could suggest something.

Sure. I am trying to intersect two surface meshes , named slope and slide as follows:
[image]
[image]
You can see that parts of slide lies inside slope, and what I did was intersecting the two meshes (by using CGAL:: clip) and clip the slide part that is outside slope, and then combine the two into one:
[image]

Here we can see non-manifold vertices because I merged the intersect points of slope and slide into one(In finite element method this is required)

and what I want to do is to remesh this model so that its topology remains but the triangles near the intersect points has a "better" shape,which means the triangles should be more close to equilateral.

I have done this by using Rhino Griddle plug-in shown as follows, but I want to make it by using opensource code.
[image]

Here I attached all models: 
[models.zip](https://github.com/user-attachments/files/16015633/models.zip)
In this Zip file you can find the original slope.off and slide.off, the clipped slope and slide, and the merged model.

Fedr, 2024-06-27:
I see, very interesting task.

The first part - clipping - is already available in MeshLib. In MeshInspector interface, it is named Mesh Boolean tool:
* `slide_clip` is what called "Inside" there,
* `slope_clip` is "Intersection" of `slide` and `slope`.

And as to simultaneous remeshing of two objects (preserving common cut line and same location of vertices there), it is something that needs to be developed in MeshLib. We will need to evaluate the efforts for this.

citystrawman, 2024-06-28:
> I see, very interesting task.
>
> The first part - clipping - is already available in MeshLib. In MeshInspector interface, it is named Mesh Boolean tool:
>
> * `slide_clip` is what called "Inside" there, [...]

Thank you!

Fedr, 2024-06-30:
Just to better understand the task. Here are your objects right after clipping (there are tons of small and degenerate triangles):
[image]

And here both objects are re-meshed, but Slope and Slide still have coinciding boundaries:
[image]

Re-meshed objects to download: [Remesh.zip](https://github.com/user-attachments/files/16046228/Remesh.zip)

Some minor artifacts near the boundaries are still present, but are we on the right way?

citystrawman, 2024-07-01:
> Just to better understand the task. Here are your objects right after clipping (there are tons of small and degenerate triangles): [...]

Yes. They are very near to waht I want. But just as you said, there're some minor problems:
[image]
For example, in the picture above the slide's triangle goes above slope, and some triangles are still "bad shape":
[image]
but it's already a great leap. 
BTW may I know what is your strategy for optimize the surface mesh?

Fedr, 2024-07-01:
> BTW may I know what is your strategy for optimize the surface mesh?

Currently I am trying to use our mesh decimation algorithm to achieve your task:
1. I already implemented a new mode where contour's edge collapse in one object will force corresponding edge collapse in the other object.
2. And then remaining small edges and almost degenerate triangles near already fixed contours must be eliminated. This is in progress.

Fedr, 2024-07-01:
Here are some updated results from today:
* much less small edges and degenerate triangles remain,
* only close triangles to the intersection contour are touched, not the ones far from it.

[image]

Meshes: [Remesh2.zip](https://github.com/user-attachments/files/16057252/Remesh2.zip)

citystrawman, 2024-07-04:
> Here are the final results: [Remesh3.zip](https://github.com/user-attachments/files/16068500/Remesh3.zip)
> 
> Just for comparison. Your input mesh: ![image](https://private-user-images.githubusercontent.com/3136125/345034029-6ed25569-1714-4094-813d-410d7f2c6f4d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxMDQzMDgsIm5iZiI6MTcyMDEwNDAwOCwicGF0aCI6Ii8zMTM2MTI1LzM0NTAzNDAyOS02ZWQyNTU2OS0xNzE0LTQwOTQtODEzZC00MTBkN2YyYzZmNGQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDRUMTQ0MDA4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWM1MWMwODQzOTM5ZTY0N2IzM2I0NGM3YWEwMTc1NzcwMzE0Y2M5ZTA5NTU1MWVjMTZkZTIzMTE5YTUzMTQ5NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.xFd2zCZmkEU8qgAotULVzenuTYfz6b3OpG7OHPHF-_Y)
> 
> The same place after simplification: ![image](https://private-user-images.githubusercontent.com/3136125/345034158-3004cd42-f236-4a4c-8c3a-6849689e7ffe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxMDQzMDgsIm5iZiI6MTcyMDEwNDAwOCwicGF0aCI6Ii8zMTM2MTI1LzM0NTAzNDE1OC0zMDA0Y2Q0Mi1mMjM2LTRhNGMtOGMzYS02ODQ5Njg5ZTdmZmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDRUMTQ0MDA4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2YzNGVhODk4NDNhZDI0NGE0ODIxNWMzMmZmMjlhMDAyYjM3ZjQ2NzRmMzEwNmIxMzY3MDQxODdiZGY0OTU4NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.T7ED1NReQKUmhFdF68_9A4dVKkgCvkPNBbx2V0l_brs)
> 
> And this is C++ code for performing joint Slope/Slide simplification:
> 
> ```c++
>     using namespace MR;
>     // open mesh files
>     auto slide = MeshLoad::fromAnySupportedFormat( "slide_clip.off" );
>     auto slope = MeshLoad::fromAnySupportedFormat( "slope_clip.off" );
>     if ( !slide || !slope )
>         return 1;
> 
>     // put input meshes in one as separate connected components
>     Mesh mesh = *slope;
>     mesh.addPart( *slide );
> 
>     // find cut contour as the edges having same coordinates both in Slide and Slope
>     const auto twinPairs = findTwinEdgePairs( mesh, 0 );
>     UndirectedEdgeBitSet twinEdges = findTwinUndirectedEdges( twinPairs );
>     UndirectedEdgeHashMap twinMap = findTwinUndirectedEdgeHashMap( twinPairs );
> 
>     // eliminate edges longer than 10
>     DecimateSettings settings;
>     settings.strategy = DecimateStrategy::ShortestEdgeFirst;
>     settings.maxError = 10;
> 
>     // collapse contour edges together in Slide and Slope
>     settings.twinMap = &twinMap;
>     // and never flip them
>     settings.notFlippable = &twinEdges;
>     // or move due to collapse of nearby edges
>     VertBitSet bdVerts = getIncidentVerts( mesh.topology, twinEdges );
>     settings.bdVerts = &bdVerts;
>     settings.touchBdVerts = false;
>     settings.collapseNearNotFlippable = true;
>     // allow flipping of other edges
>     settings.maxAngleChange = PI_F / 3;
> 
>     // process edges not further than 3 hops away from cut contour
>     auto regionEdges = getIncidentEdges( mesh.topology,
>         getIncidentEdges( mesh.topology, getIncidentEdges( mesh.topology, twinEdges ) ) );
>     settings.edgesToCollapse = &regionEdges;
> 
>     // run re-meshing
>     decimateMesh( mesh, settings );
> 
>     // separate result on Slide and Slope
>     auto slopeFaces = MeshComponents::getLargestComponent( mesh );
>     Mesh slopeRes;
>     slopeRes.addPartByMask( mesh, slopeFaces );
>     Mesh slideRes;
>     slideRes.addPartByMask( mesh, mesh.topology.getValidFaces() - slopeFaces );
> 
>     // save in files
>     MeshSave::toAnySupportedFormat( slopeRes, "slopeRes.off" );
>     MeshSave::toAnySupportedFormat( slideRes, "slideRes.off" );
> ```

Thank you! BTW where can I get the infomation about `UndirectedEdgeBitSet ` ? I dont quite understand this,  I searched it in documentation but no further infomation is documented.

closed 2024-07-04 by citystrawman

Fedr, 2024-07-04:
>BTW where can I get the infomation about UndirectedEdgeBitSet  ? I dont quite understand this, I searched it in documentation but no further infomation is documented.

We will work on documentation later. In short, `UndirectedEdgeBitSet` can be seen as compact map from undirected edge Id to `bool`. If the value is `true` then corresponding undirected edge Id is considered included in the set, otherwise it is excluded.

For example, `UndirectedEdgeBitSet twinEdges` contains `true` values only for the edges common between Slide and Slope.

citystrawman, 2024-07-05:
> We will work on documentation later. In short, `UndirectedEdgeBitSet` can be seen as compact map from undirected edge Id to `bool`. If the value is `true` then corresponding undirected edge Id is [...]

So `UndirectedEdgeBitSet twinEdges  ` should have the same size as `twinPairs` ? In my example, twinPairs has 1340 and twinEdges   has 1270?

Fedr, 2024-07-05:
The size of `std::vector<EdgePair> twinPairs` is equal to the number of edges in the intersection contour (or common edges between `slide_clip.off` and `slope_clip.off`) times 2, because the edges are *directed* (both orientations). In your example meshes, the size if `1324`.

And the  number of set bits in `UndirectedEdgeBitSet twinEdges` is again `1324`, because:
* here the edges are undirected (two times less),
* each `EdgePair` is actually about two edges, so we get the same number:
```
UndirectedEdgeBitSet findTwinUndirectedEdges( const std::vector<EdgePair> & pairs )
{
    MR_TIMER
    UndirectedEdgeBitSet res;
    for ( const auto & [e1, e2] : pairs )
    {
        res.autoResizeSet( e1.undirected() );
        res.autoResizeSet( e2.undirected() );
    }

    return res;
}
```

And during decimation the number of such edges decreases.

citystrawman, 2024-07-05:
> The size of `std::vector<EdgePair> twinPairs` is equal to the number of edges in the intersection contour (or common edges between `slide_clip.off` and `slope_clip.off`) times 2, because the edges [...]

Here's the screenshot:
[image]

after executing `findTwinUndirectedEdges`,  `twinEdges` m_bits size is 1270. I guess that the actual size is 1324 as you said, but because the info is stored in bits, the bits size does not necessarily equals to 1324? Do I understand it correct?

Fedr, 2024-07-05:
`UndirectedEdgeBitSet` (as all our `BitSet`'s) has `boost::dynamic_bitset` under the hood.

`m_bits` there has type `std::vector<Block, Allocator>`, where `Block=std::uint64_t`.

So `m_bits.size() = 1270` means that 8*1270=10160 bytes are allocated for the set, but it does not tell us how many bits are `1` or `0` among them. Please use `BitSet::count()` function to count the number of set bits:
[image]

citystrawman, 2024-07-05:
> `UndirectedEdgeBitSet` (as all our `BitSet`'s) has `boost::dynamic_bitset` under the hood.
>
> `m_bits` there has type `std::vector<Block, Allocator>`, where `Block=std::uint64_t`.
>
> So `m_bits.size() = [...]

Thank you!

citystrawman, 2024-07-08:
> `UndirectedEdgeBitSet` (as all our `BitSet`'s) has `boost::dynamic_bitset` under the hood.
>
> `m_bits` there has type `std::vector<Block, Allocator>`, where `Block=std::uint64_t`.
>
> So `m_bits.size() = [...]

Hi Sir, I just downloaded a [slide](https://graphics.stanford.edu/courses/cs468-10-fall/LectureSlides/08_Simplification.pdf) on mesh decimation and on page 38 to 47 it discussed Half-Edge Collapse method.

Does your code:
```
// collapse contour edges together in Slide and Slope
settings.twinMap = &twinMap;
// and never flip them
settings.notFlippable = &twinEdges;
```
do the same thing as Half-Edge Collapse method?  

I read `findTwinUndirectedEdgeHashMap` and it seems that `twinMap` is just a map that has `twinPairs` first element as its key and second element as its value, and then double the map by interchanging key and value, and by setting `twinEdges` as notFlippable, only the correct collapse direction is executed.  Do I understand it correct?

Fedr, 2024-07-08:
In general, `decimateMesh` function does what is described in the slides.

The purpose of `twinEdges` is to perform *simultanous* collapse of two edges (one on cut contour in Slope and the other on cut contour in Slide). Otherwise there will be no common edge loop on two independently simplified meshes.

citystrawman, 2024-07-09:
> In general, `decimateMesh` function does what is described in the slides.
>
> The purpose of `twinEdges` is to perform _simultanous_ collapse of two edges (one on cut contour in Slope and the other on [...]

Hi, I just found that there're two triangles in the slideRes.off got stuck out:
[image]
[image]

and I optimized this by flip the connected edges:
[image]

May I know if this little error can be avoided by code?

Fedr, 2024-07-09:
Please try increasing this setting:
```
settings.maxAngleChange = PI_F / 3;
```
It will allow flipping of more edges (even if surface changes significantly).

citystrawman, 2024-07-09:
> Please try increasing this setting:
> 
> ```
> settings.maxAngleChange = PI_F / 3;
> ```
> 
> It will allow flipping of more edges (even if surface changes significantly).
I increased maxAngleChange to PI_F/2 but it still does not flip

Fedr, 2024-07-09:
If we are talking about this edge:
[image]

Then according to our criteria, which minimize maximal radius of circumscribed circle for each mesh triangle, this edge is optimal. Flipping it will result in appearance of larger circles. See `checkDeloneQuadrangle` function.