MeshInspector/MeshLib #3121: build WASM/Emscripten error

SivHMA, 2024-08-04:
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_utf-8.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_err_str.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_source_file_stdio_named.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_random_unix.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_crypto_mbedtls.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_winzip_aes.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_source_winzip_aes_decode.c.o
[ 39%] Building C object libzip/lib/CMakeFiles/zip.dir/zip_source_winzip_aes_encode.c.o
[ 40%] Linking C static library ../../libzip.a
[ 40%] Built target zip
[ 40%] Linking CXX static library ../../../clang_18.0_cxx20_32_release/libtbb.a
[ 40%] Built target tbb
[ 40%] Linking CXX executable ../tl-expected-tests.js
[ 40%] Built target tl-expected-tests
[ 40%] Linking CXX executable ../loader_example.js
cache:INFO: generating system asset: symbol_lists/4ee8f995f0a454a8d6666038952951a64ee74834.json... (this will be cached in "/home/czc/emsdk/upstream/emscripten/cache/symbol_lists/4ee8f995f0a454a8d6666038952951a64ee74834.json" for subsequent builds)
cache:INFO:  - ok
[ 40%] Built target loader_example
make: *** [Makefile:156: all] Error 2
emmake: error: 'make -j 16' failed (returned 2)

The above error occurs.
How to fix it?

Fedr, 2024-08-04:
Could you please explain what command was run? And please attach full log file.

SivHMA, 2024-08-04:
> Could you please explain what command was run? And please attach full log file.

[image]
I followed the steps in the readme.
The error occurs when building a third-party library.(./scripts/build_thirdparty.sh, I choose 'Y')

Fedr, 2024-08-04:
I see. And could you please attach full log file with the error (last few lines as above are not enough)?

SivHMA, 2024-08-05:
> I see. And could you please attach full log file with the error (last few lines as above are not enough)?

I have successfully built third-party libraries. (./scripts/build_thirdparty.sh)
But I encountered a new mistake in './scripts/build_source.sh'.

[build_source.log](https://github.com/user-attachments/files/16492004/build_source.log)

adalisk-emikhaylov, 2024-08-05:
Can you please build in verbose mode and attach the log from that as well?

To do that, edit `build_source.sh` and replace `emmake make` with `emmake make VERBOSE=1`, then re-run it.

SivHMA, 2024-08-05:
> I see. And could you please attach full log file with the error (last few lines as above are not enough)?

I tried singlethreaded and it success.
What is the difference between the 'Y' and 's' options?Like performance differences?

SivHMA, 2024-08-05:
> Can you please build in verbose mode and attach the log from that as well?
> 
> To do that, edit `build_source.sh` and replace `emmake make` with `emmake make VERBOSE=1`, then re-run it.

[Uploading build_source_05-08-2024_165258.log…]()

SivHMA, 2024-08-05:
I guess it's a multi-threaded problem.

Fedr, 2024-08-05:
> What is the difference between the 'Y' and 's' options? Like performance differences?

Single-threaded WASM build can run in absolutely all browsers, including iOS and Facebook. And multi-threaded WASM build can run much faster in modern OSes and browsers.

SivHMA, 2024-08-05:
> > What is the difference between the 'Y' and 's' options? Like performance differences?
> 
> Single-threaded WASM build can run in absolutely all browsers, including iOS and Facebook. And multi-threaded WASM build can run much faster in modern OSes and browsers.

I'm still having errors compiling the multithreaded version.
The log file is shown above. Is there a way to solve the above problems?

adalisk-emikhaylov, 2024-08-05:
Hmm, can you also attach `CMakeFiles/MRTestC.dir/linklibs.rsp` and `CMakeFiles/MRTestC.dir/objects1.rsp`? Those should be somewhere in the build directory.

SivHMA, 2024-08-05:
> Hmm, can you also c `CMakeFiles/MRTestC.dir/linklibs.rsp` and `CMakeFiles/MRTestC.dir/objects1.rsp`? Those should be somewhere in the build directory.
> 
> Also, what's your `cmake --version`?

cmake verison is 3.19.3.
[linklibs.txt](https://github.com/user-attachments/files/16495252/linklibs.txt)
[objects1.txt](https://github.com/user-attachments/files/16495253/objects1.txt)

adalisk-emikhaylov, 2024-08-05:
Are you on Ubuntu 22.04? The latest CMake there should be 3.22.1, can you try updating to it? And if that doesn't work, try installing the latest CMake [from their PPA](https://apt.kitware.com/).

adalisk-emikhaylov, 2024-08-05:
Another thing to check is `emsdk` version. Are you on `3.1.48`? Can you attach the output from `emsdk list`?

SivHMA, 2024-08-06:
> Are you on Ubuntu 22.04? The latest CMake there should be 3.22.1, can you try updating to it? And if that doesn't work, try installing the latest CMake [from their PPA](https://apt.kitware.com/).

I switched the cmake version to 3.22.1, and it work.
Thank you!

adalisk-emikhaylov, 2024-08-06:
I didn't expect it to work, but happy that it worked. Oh well. :slightly_smiling_face:

closed 2024-08-06 by adalisk-emikhaylov