MeshInspector/MeshLib #3208: Merge multiple meshes with overlapping areas

HiJasonZou, 2024-08-21:
Author, thank you for your excellent work. To scan complete data of a large object, due to insufficient computer performance, it is necessary to scan different areas multiple times to obtain a complete object model. Each scan results in a mesh model, which requires merging multiple meshes with overlapping areas.Can Meshlib do this?

Grantim, 2024-08-21:
Hello!

You can try these steps:
1. Convert all meshes into point clouds (https://github.com/MeshInspector/MeshLib/blob/0227635ae2963876db7028ed34ad4fc765b87358/source/MRMesh/MRMeshToPointCloud.h#L1-L11)
2. Merge all point clouds (please see this https://github.com/MeshInspector/MeshLib/blob/0227635ae2963876db7028ed34ad4fc765b87358/source/MRMesh/MRObjectPoints.cpp#L136-L195)
3. Fuse merged point cloud (https://github.com/MeshInspector/MeshLib/blob/0227635ae2963876db7028ed34ad4fc765b87358/source/MRMesh/MRPointsToMeshFusion.h#L37-L39)

HiJasonZou, 2024-08-22:
thank you ! I want to ask one questions. I use poisson  reconstruction to construct meshes to ensure high quality meshes."Fuse merge point cloud" step: extracting meshes from general TSDF does not guarantee mesh quality in sharp and thin-walled areas.

Fedr, 2024-08-23:
Poisson reconstruction is a very good method, but typically it works mush slower than fusion from MeshLib.

Could you please share some examples with input point clouds where Poisson reconstruction outperforms us by quality?

HiJasonZou, 2024-09-02:
> Poisson reconstruction is a very good method, but typically it works mush slower than fusion from MeshLib.
> 
> Could you please share some examples with input point clouds where Poisson reconstruction outperforms us by quality?

I scan an object using a speckle sensor and the point cloud is very noisy, I reasoned that poisson reconstruction would be better. I will compare the effects later.
I formed a point cloud from the vertices and normals of the two meshes, imported it into the MeshInspector software, and used "points to mesh" (Create mesh objects by taking vertices from active point clouds and adding triangles) to solve multi-resolution mesh fusion.  How to implement this function using meshlib.

Grantim, 2024-09-02:
Hello!

"Points to Mesh" is `triangulatePointCloud` function from https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRPointCloudTriangulation.h this header.

You can find more information here https://doc.meshinspector.com/group__PointCloudTriangulationGroup.html

Fedr, 2024-09-02:
If your points are very noisy, it is a good idea to sample them before running "Points to Mesh".

HiJasonZou, 2024-09-03:
> Hello!
> 
> "Points to Mesh" is `triangulatePointCloud` function from https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRPointCloudTriangulation.h this header.
> 
> You can find more information here https://doc.meshinspector.com/group__PointCloudTriangulationGroup.html

my code:
`#include <MRMesh/MRMesh.h>
#include <MRMesh/MRMeshLoad.h>
#include <MRMesh/MRMeshSave.h>
#include"MRMesh/MRPointCloudTriangulation.h"
#include <MRMesh/MRPointsLoad.h>
#include <MRMesh/MRPointCloud.h>
#include <iostream>

int main()
{
    // Load mesh
    MR::PointCloud in_points = *MR::PointsLoad::fromAnySupportedFormat("D:\\git_clone\\test_meshlib\\all_points.ply");
    std::cout << "points num: " <<in_points.points.size()<< "normals num: " << in_points.normals.size() << std::endl;

    MR::TriangulationParameters params;
    params.numNeighbours = 12;
    params.critHoleLength = 0;

    std::cout << "start triangulate " << std::endl;
    MR::Mesh out_mesh = *MR::triangulatePointCloud(in_points, params);

    //// Save result
    auto saveRes = MR::MeshSave::toAnySupportedFormat(out_mesh, "pc2tri_Mesh.ply");
    std::cout << "success " << std::endl;
    getchar();
    return 0;
}`
The code runs in the function "MR::triangulate Point Cloud" and cannot be exited. Is this why? I can mesh this point cloud ply data in the software meshinspector.

HiJasonZou, 2024-09-03:
> If your points are very noisy, it is a good idea to sample them before running "Points to Mesh".

Thank you for your good advice.I will try.

Grantim, 2024-09-03:
>The code runs in the function "MR::triangulate Point Cloud" and cannot be exited. Is this why? I can mesh this point cloud ply data in the software meshinspector.

Hello! It seams strange, could you please provide the mesh, so we could reproduce the problem on our end?

HiJasonZou, 2024-09-04:
> > The code runs in the function "MR::triangulate Point Cloud" and cannot be exited. Is this why? I can mesh this point cloud ply data in the software meshinspector.
> 
> Hello! It seams strange, could you please provide the point cloud, so we could reproduce the problem on our end?

ok.I am use tag v2.3.8-182-g3ae18517. Maybe i need to update.
[mypoints_3.zip](https://github.com/user-attachments/files/16874004/mypoints_3.zip)

Grantim, 2024-09-05:
Hello!
Thanks for the file, I have just had a look at it,. It looks like there are ~200 duplicated points that don't let to get good result, so sampling might really work for you, I've managed to get good result with code like this:

```c++
#include <MRMesh/MRMesh.h>
#include <MRMesh/MRMeshLoad.h>
#include <MRMesh/MRMeshSave.h>
#include "MRMesh/MRUniformSampling.h"
#include"MRMesh/MRPointCloudTriangulation.h"
#include <MRMesh/MRPointsLoad.h>
#include <MRMesh/MRPointCloud.h>
#include <iostream>

int main()
{
    // Load mesh
    MR::PointCloud in_points = *MR::PointsLoad::fromAnySupportedFormat("D:\\git_clone\\test_meshlib\\all_points.ply");
    std::cout << "points num: " <<in_points.points.size()<< "normals num: " << in_points.normals.size() << std::endl;

    auto sampled_points = MR::makeUniformSampledCloud( in_points, { .distance = 1e-6f } );

    if ( !sampled_points )
        return 1;

    std::cout << "points num: " << sampled_points->points.size() << "normals num: " << sampled_points->normals.size() << std::endl;

    MR::TriangulationParameters params;
    params.numNeighbours = 16;
    //params.critHoleLength = 0; // do not change this to fill possible small holes

    std::cout << "start triangulate " << std::endl;
    MR::Mesh out_mesh = *MR::triangulatePointCloud( *sampled_points, params );
    //// Save result
    auto saveRes = MR::MeshSave::toAnySupportedFormat(out_mesh, "pc2tri_Mesh.ply");
    std::cout << "success " << std::endl;
    getchar();
    return 0;
}

HiJasonZou, 2024-09-10:
> Hello! Thanks for the file, it looks like there are ~200 duplicated points that don't let to get good result, so sampling might really work for you, I've managed to get good result with code like this:
> 
> ```c++
> #include <MRMesh/MRMesh.h>
> #include <MRMesh/MRMeshLoad.h>
> #include <MRMesh/MRMeshSave.h>
> #include "MRMesh/MRUniformSampling.h"
> #include"MRMesh/MRPointCloudTriangulation.h"
> #include <MRMesh/MRPointsLoad.h>
> #include <MRMesh/MRPointCloud.h>
> #include <iostream>
> 
> int main()
> {
>     // Load mesh
>     MR::PointCloud in_points = *MR::PointsLoad::fromAnySupportedFormat("D:\\git_clone\\test_meshlib\\all_points.ply");
>     std::cout << "points num: " <<in_points.points.size()<< "normals num: " << in_points.normals.size() << std::endl;
> 
>     // filter duplicated points
>     auto sampled_points = MR::makeUniformSampledCloud( in_points, { .distance = 1e-6f } );
> 
>     if ( !sampled_points )
>         return 1;
> 
>     std::cout << "points num: " << sampled_points->points.size() << "normals num: " << sampled_points->normals.size() << std::endl;
> 
>     MR::TriangulationParameters params;
>     params.numNeighbours = 16;
>     //params.critHoleLength = 0; // do not change this to fill possible small holes
> 
>     std::cout << "start triangulate " << std::endl;
>     MR::Mesh out_mesh = *MR::triangulatePointCloud( *sampled_points, params );
>     //// Save result
>     auto saveRes = MR::MeshSave::toAnySupportedFormat(out_mesh, "pc2tri_Mesh.ply");
>     std::cout << "success " << std::endl;
>     getchar();
>     return 0;
> }
> ```

thank you ! it works. Which papers is the “MR::triangulatePointCloud” algorithm based on?

Grantim, 2024-09-10:
This algorithm is designed by our team, I am not sure that there are papers

HiJasonZou, 2024-09-11:
This point cloud meshing algorithm is amazing. What is the core idea of ​​the algorithm?

HiJasonZou, 2024-09-11:
I tested the "pointsTo MeshFusion" algorithm and the mesh quality was great, you guys did a great job.

Grantim, 2024-09-11:
> This point cloud meshing algorithm is amazing. What is the core idea of ​​the algorithm?

Main idea is quite simple:
1. Build and optimize local triangulation around each vertex (with given number of neighbors or fixed radius)
2. Add faces from local triangulations to final result prioritized by number if votes (if all 3 vertices have same face in their local triangulation face has 3 votes, if only two have this face then it has only 2 votes)
3. Fill small holes

closed 2024-10-30 by Fedr