MeshInspector/MeshLib #3229: Improve active voxels support.

andrewerf, 2024-08-23:
Instead of storing the active box manually (separate from the active voxels representation in OpenVDB), get it from the underlying OpenVdb-volume and cache it when possible.

andrewerf, 2024-08-23:
#3228

andrewerf, 2024-08-23:
This is only an internal change, it should not have an impact on user experience.

andrewerf, 2024-08-23:
Testing tips:
- Check backward compatibility with `.mru` files (containing voxels data) saved with previous version of MeshLib
- Check that active bounds preserved when saving to `.mru` file.

ADniill, 2024-08-27:
Active voxels constantly change when history changes, which breaks the work of other tools
Steps playbcak:
1. Launche tool Creat simple objects
2. Create sphere (default settings)
3. Launche tool mesh to voxels
4. Click button Apply
5. Launche tool Set Active voxels Box
6. Trim some voxels
7.Use Undo Redo
Expected resuslt: 
Active voxels are displayed in the correct number, before and after trimming, without changes
Actual result: 

When history changes, the number of active voxels is different each time
[MeshInspector (Debug_ 2.4.4.73) lildonya@mail.ru normals.mru_ 2024-08-27 14-51-44.zip](https://github.com/user-attachments/files/16761214/MeshInspector.Debug_.2.4.4.73.lildonya%40mail.ru.normals.mru_.2024-08-27.14-51-44.zip)

Grantim, 2024-08-28:
should be fixed

closed 2024-08-28 by Grantim