MeshInspector/MeshLib #3264: Fast Signed Distance Calculation for a list of points

emil-peters, 2024-08-29:
To calculate the signed distance from a list of points to the surface of the mesh in python I'm currently doing the following:

Looping over each point and calculating the distance. While this works it's generally slower then using the [PySDF library](https://github.com/sxyu/sdf). But less error prone, as the PySDF library tends to return weird high values.

```python
    proj_results = []
    for point in points:
        proj_result = mr.findProjection(mr.Vector3f(*point), mr_mesh_part)
        proj_results.append(list(proj_result.proj.point))
    
    proj_results = np.array(proj_results)
    distances = np.linalg.norm(points - proj_results, axis=1)
```

I was wondering if there was a way to have a quick SDF calculation in meshlib for a set of points, or to have this `mr.findProjection` accept a `vectorVector3f` for quick parallel computation of the signed distance

Grantim, 2024-08-29:
Hello!

Thanks for the suggestion, we will add function for parallel computation, for now you can try using `projectAllMeshVertices`, if initial `points` is part of some mesh

https://github.com/MeshInspector/MeshLib/blob/master/test_python/test_meshProjections.py

emil-peters, 2024-08-29:
Thanks! Seems to be indeed way quicker!

closed 2024-10-30 by Fedr