MeshInspector/MeshLib #3300: Question about reconstruction

Gooda11, 2024-09-03:
Hello! I have a question about reconstruction from CT pictures. I know we can do this by using iso-surface extraction. But, is it possible to do this not in straight direction but in a Circular direction? For example, a tube not scanned from bottom to top, but from 0 to 360 degree around circular direction?
Thanks.

Grantim, 2024-09-04:
Hello!
>I know we can do this by using iso-surface extraction.

Really it works on already reconstructed volume, I think you are looking for the algorithm of initial reconstruction, like in this video: 
[image]

We have this algorithm in MeshInspector, can you please verify that it works for you?

Gooda11, 2024-09-04:
> Hello!
> 
> > I know we can do this by using iso-surface extraction.
> 
> Really it works on already reconstructed volume, I think you are looking for the algorithm of initial reconstruction, like in this video: [image]
> 
> We have this algorithm in MeshInspector, can you please verify that it works for you?

Hello! It seems to be what I want, but can I use the function via python (including all the parameters, like rotation)? Or it is only in the software?

Grantim, 2024-09-05:
Hello! 
Sorry for long response, now this algorithm is available only in MeshInspector application.

Gooda11, 2024-09-05:
Hello, thank you for your reply. Is it possible that you will plan to implement this in Python in future?

DmitryLambin, 2024-09-05:
@Gooda11 Hi, My name is Dmitry. I'm a product marketer at MeshInspector MeshLib. MeshLib may be more than helpful in CT reconstruction and other parts of the voxel and mesh data processing workflow. Would you be open to having an exploration session and learning more about MeshLib opportunities? If yes, please email me at dmitry.lambin@meshinspector.com

closed 2024-10-30 by Fedr