MeshInspector/MeshLib #3351: Expose `findAlphaShapeAllTriangles()` and `findAlphaShapeNeiTriangles()` to Python

LucaLumetti, 2024-09-13:
The functions `findAlphaShapeAllTriangles()` and `findAlphaShapeNeiTriangles()` are currently unavailable in the Python module.

Is there a reason why these functions have not been exported?

LucaLumetti, 2024-09-13:
Looks like other function not exported are:
- `findPointsInBall`
- `circumballCenters`

Fedr, 2024-09-13:
Thanks for your question.

Alpha-shape functions are not-finished and not recommended for use yet.

And as to others, we are working on a general solution to expose all C++ functions in python. It should be ready by October.

Fedr, 2026-09-23:
Hello! An update on this issue: the alpha-shape code was rewritten this summer, and as of MeshLib v3.1.4.297 it is available in Python together with the other functions you mentioned:

- `mrmeshpy.findAlphaShape( cloud, radius )` builds a mesh directly from the point cloud
- `mrmeshpy.findAlphaShapeAllTriangles( cloud, radius )` returns only the triangles
- `mrmeshpy.getAlphaShapeData` + `mrmeshpy.findAlphaShapeNeiTriangles` find the triangles around individual points
- `mrmeshpy.findPointsInBall` and `mrmeshpy.circumballCenters` are exposed as well

```python
from meshlib import mrmeshpy as mr

cloud = mr.meshToPointCloud( mr.makeSphere( mr.SphereParams() ) )
mesh = mr.findAlphaShape( cloud, 2.0 )  # alpha = -1/radius
print( mesh.topology.numValidFaces() )
```

The ball-emptiness tests use exact predicates on integer coordinates, so duplicate and degenerate (cospherical) points are handled robustly.

If you have MeshLib 3.1.3 or older installed, please uninstall it before installing the new version (upgrading in place leaves a broken package, see [Upgrading from 3.1.3 or older](https://meshlib.io/documentation/MeshLibPythonSetupGuide.html#PythonSetupUpgrade)):
```sh
pip uninstall -y meshlib
pip install meshlib
```
Let us know if it works for you.