MeshInspector/MeshLib #3429: centroid for fillHoleTrivially()

DMBabich, 2024-09-30:
Hello!
In documentation about [fillHoleTrivially()](https://doc.meshinspector.com/group__FillHoleGroup.html#ga0241a36cc2b1016fd0cf66e9ae71f722)
You write:
**"... by creating one new vertex in the centroid of boundary vertices  and connecting new vertex with all boundary vertices."**

---
Q1: How I can change centroid coordinate? (For example: Z coord -= 5)
Q2: How I can use smoothed for Q1?
Q3: How I can force use fillHoleTrivially() if this method include in fillHoles() and used if cannot fill hole without multiple edges?
Lang: Python

---
I think it's look like how:
```
import meshlib.mrmeshpy as mrmeshpy
# Load mesh
mesh = mrmeshpy.loadMesh("mesh.stl")

---smoothed---
avgEdgeLength = 0.0
numEdges = 0
for i in range(mesh.topology.undirectedEdgeSize()):
	dirEdge = mrmeshpy.EdgeId(i*2)
	org = mesh.topology.org(dirEdge)
	dest = mesh.topology.dest(dirEdge)
	avgEdgeLength += (mesh.points.vec[dest.get()] - mesh.points.vec[org.get()]).length()
	numEdges = numEdges + 1
avgEdgeLength = avgEdgeLength/numEdges
---

# Find single edge for each hole in mesh
hole_edges = mesh.topology.findHoleRepresentiveEdges()
for e in hole_edges:
    #  Setup filling parameters
    params = mrmeshpy.FillHoleParams()
    params.metric = mrmeshpy.getUniversalMetric(mesh)

    ---Change centroid coords---
    e.centroid[2] -=5  # get coordinates and change Z-coordinate
    ---

    #  Fill hole represented by `e`
    mrmeshpy.fillHole(mesh, e, params)

    ---use_smoothed---
    newVerts = mrmeshpy.VertBitSet()
    subdivSettings = mrmeshpy.SubdivideSettings()
    subdivSettings.maxEdgeLen = avgEdgeLength
    subdivSettings.maxEdgeSplits = 20000
    subdivSettings.region = newFaces
    subdivSettings.newVerts = newVerts
    mrmeshpy.subdivideMesh(mesh,subdivSettings)
    mrmeshpy.positionVertsSmoothly(mesh,newVerts)
    ---
# Save result
mrmeshpy.saveMesh(mesh, "filledMesh.stl")
```

---
Please, help ♥

Grantim, 2024-09-30:
Hello!

Unfortunately, `fillHoleTrivially` is not yet exposed to python, so one cannot use it directly there (we will release new modules soon, they will contain all c++ functions, so it will not be an issue).
So:
**Q3**: One cannot use `fillHoleTrivially` directly in python

**Q1**: You can detect that `fillHoleTrivially` was used inside `fillHole` by checking amount of vertices (`fillHole` does not produce new vertices, while `fillHoleTrivially` produces one), then you can change coordinates of this new vertex:
```python
from meshlib import mrmeshpy as mm

meshWithHole = mm.loadMesh("path/to/mesh_with_hole.stl")
numValidVerts = meshWithHole.topology.getValidVerts().count()
hole_edges = mesh.topology.findHoleRepresentiveEdges():
for e in hole_edges:
    params = mrmeshpy.FillHoleParams()
    params.metric = mrmeshpy.getUniversalMetric(mesh)
    mm.fillHole(meshWithHole, e, params)
    if ( meshWithHole.topology.getValidVerts().count() > meshWithHole ): # fillHoleTrivialy was used
        meshWithHole.points.vec[ len(meshWithHole.points.vec) - 1 ] = centroid # set new position to new vert
        meshWithHole.invalidateCaches() # important to call if you manually change coordinates
```

**Q2**: please have a look here https://github.com/MeshInspector/MeshLib/issues/2606 and also note that as far as you are going to reposition all new vertices after subdivision specifying centroid will not have any effect (because this vertex will also be affected by `positionVertsSmoothly`)

DMBabich, 2024-09-30:
> Unfortunately, `fillHoleTrivially` is not yet exposed to python, so one cannot use it directly there (we will release new modules soon, they will contain all c++ functions, so it will not be an issue). So: **Q3**: One cannot use `fillHoleTrivially` directly in python
> 
Cool!
Can you give an approximate release date?

Grantim, 2024-09-30:
We hope to release this week

closed 2024-09-30 by DMBabich

Grantim, 2024-10-21:
Hello!
New release can be installed like this
```
pip install meshlib==3.0.0.40
```
> note that version should be set explicitly because new python modules have some signature differences from older version

---

Backlink to following discussion: https://github.com/MeshInspector/MeshLib/discussions/3532