MeshInspector/MeshLib #3468: Request for Example Code on Mesh Deformation in Python

ome-13, 2024-10-07:
Hi MeshLib Developers,

First, thank you for the amazing work on MeshLib! I've been exploring the library, especially for mesh deformation tasks like Laplacian and freeform deformation. However, I'm having trouble finding Python examples or documentation related to these deformation functionalities.

Could you please provide a Python example or guidance on how to apply mesh deformation (e.g., Laplacian deformation, smoothing) using the Python bindings? I would love to see a simple example of loading a mesh, applying deformation, and saving the result.

Additionally, if any specific parameters need attention or there's a recommended workflow for deformation tasks, that information would be incredibly helpful!

Thank you for your time and support!

Grantim, 2024-10-07:
Hello!

Now we only have couple of Laplacian functions available in python:
```py
positionVertsSmoothly
positionVertsSmoothlySharpBd
inflate
```

They have very similar interface:
```py
from meshlib import mrmeshpy as  mm

# load mesh
mesh = mm.loadMesh("path/to/mesh.stl")

# setup vertices to reposition
verticesToReposition = mm.VertBitSet()
verticesToReposition.resize( mesh.topology.getValidVerts().size(), False )
# for example set vertex with id == 4
# and its neighbors
verticesToReposition.set( mm.VertId(4) )
# 3 layers of neighbors
mm.expand( mesh.topology, verticesToReposition, 3 )

# you can play with different weight types
mm.positionVertsSmoothly( mesh, verticesToReposition, mm.LaplacianEdgeWeightsParam.CotanWithAreaEqWeight )
# mm.positionVertsSmoothlySharpBd( mesh, verticesToReposition )
# mm.inflate( mesh, verticesToReposition )
```

You can also find useful `help(mm.inflate)` command that you can apply to any class or function in module to see information about.

Also we are going to release new version of python modules soon, they will include all c++ functions so both FreeForm deformation and Laplacian will be available. It will have same signatures as c++ library so you can find https://doc.meshinspector.com/ this page useful too.

ome-13, 2024-10-08:
Thank you so much for the detailed response and for providing the example code! I appreciate the guidance on using the available Laplacian functions (positionVertsSmoothly, positionVertsSmoothlySharpBd, and inflate). I will definitely experiment with these functions and the weight types you've mentioned.

It's also great to hear that a new version of the Python modules will be released soon, including more of the C++ functionalities, especially FreeForm deformation and full Laplacian support. I’ll keep an eye on the MeshInspector documentation as well.

Grantim, 2024-10-15:
@ome-13 
We have just released new version python modules where you can try 
https://doc.meshinspector.com/ExamplePythonFreeForm.html
https://doc.meshinspector.com/ExamplePythonLaplacian.html

To install new version please use
```sh
python3 -m pip install meshlib==3.0.0.40
```
(we require to set version explicitly because new python modules are now in beta testing)