MeshInspector/MeshLib #3476: Ubuntu 22 Debian Install

Doomerdinger, 2024-10-08:
I am attempting to use the installable debian provided on the releases page.
I have installed `meshlib_v2.4.4.153_ubuntu22-dev` and I am running Ubuntu 22.04.4 LTS.

When I attempt to tun the cmake integration example I encounter an erro, `**Process finished with exit code 139 (interrupted by signal 11:SIGSEGV)**.

I also tried the latest pre-release version, as well as the previous stable version, all resulting in the same error.

Is this a known issue?

oitel, 2024-10-09:
Hello, we couldn't reproduce your issue. Please provide detailed steps to reproduce it (the commands you use to build and run the example) and a crash dump (usually it can be found in /var/crash/ directory).

oitel, 2024-10-23:
Not sure which test you are referring to `cmake-example`, the `examples/cpp-examples` directory contains multiple ones. Notice that some tests assume there are input mesh files in the working directory and no checks for their absence (for the example code's simplicity). This could lead to a segfault too.

Doomerdinger, 2024-10-25:
I was attempting to run the code found [here](https://github.com/MeshInspector/MeshLib/tree/v2.4.2.198/examples/cmake-example) initially, as there was only one I did not specify.

I have gone back and tested with `MeshBoolean.dox.cpp` and `MeshModification.dox.cpp` (the latter seemingly being analogous to the code I was running previously) targeting the correct version of the codebase.

The boolean operation seems to work fine, but the mesh modification has the same flaky segfault behavior as before.

Fedr, 2024-10-27:
Unfortunately, we are unable to reproduce the segfault in `MeshModification.dox.cpp`.

Could we ask you, please, to use the debugger, and let us know the place where segfault happens and the call stack to it?

Doomerdinger, 2024-10-28:
When using the latest release (3.0) I am unable to replicate the issue.
Additionally, I tried building meshlib myself and testing using that (for additional debugging purposes), and was not able to replicate the issue, regardless of if I targeted the latest or if I targeted 2.4.x.

That said, installing an older version from a release debian still causes problems.

I personally am not concerned with it at the moment, as I should be able to use the latest release to avoid problems, we can close this if you do not care about investigating the older version.

In the v2.4.5.32 version just in case. The segfault occurs when calling `MR::relax( mesh, {{5}} );`
No call stack beyond that, as I was not able to replicate the issue when building on my local machine, so I could not get any further debug symbols.

adalisk-emikhaylov, 2024-10-30:
This looks like it's caused by a [GCC bug that was fixed in GCC 12](https://github.com/gcc-mirror/gcc/commit/a37e8ce3b66325f0c6de55c80d50ac1664c3d0eb), that caused some struct layouts to be different in GCC 12+ compared to older versions. 

Meaning binaries compiled with GCC 12+ can be incompatible with those compiled with 11 and older, and on Ubuntu we compile the .deb using GCC 12, while the default GCC version is 11, which is probably what you ended up using. I can repro on GCC 11, but not when I switch to GCC 12.

In any case, we [worked around this bug in v3](https://github.com/MeshInspector/MeshLib/commit/d6c44ff46125495d84bdd38d47ae3492e0460dc8#diff-bf9a42f76bd86ce90999fc9f29a6e47bdc31f9e983e04e4b09007f659a10d91a), so it now shouldn't matter what GCC version you're using.

closed 2024-10-30 by adalisk-emikhaylov