MeshInspector/MeshLib #3507: ValueError: vector too long meshlib

Gooda11, 2024-10-16:
Hello, I try to load a series of tiff files. But I got 'ValueError: vector too long meshlib'
Is it possible to deal with this?
Thanks!

Fedr, 2024-10-16:
Hello, please clarify the steps you are doing. It would be nice to see your sample program and have to data to reproduce the issue.

Gooda11, 2024-10-16:
> Hello, please clarify the steps you are doing. It would be nice to see your sample program and have to data to reproduce the issue.

Hello, I was uploading my tiff files using 'settings.voxelSize = mr.Vector3f(380/5760, 120/1818, 130/260)' and 'volume = mr.loadTiffDir(settings)'. Then I got this error. I tried to to make stl file from tiff files.

Grantim, 2024-10-16:
Hello @Gooda11 ! Could you please send us data so we could reproduce the issue? If you don't want to share the data publicly you can send it to our email `support@meshinspector.com`

Gooda11, 2024-10-16:
> Hello [@Gooda11](https://github.com/Gooda11) ! Could you please send us data so we could reproduce the issue? If you don't want to share the data publicly you can send it to our email `support@meshinspector.com`

Ok, I will send the data later (in this evening)! Maybe it is because of the total size of all the data? Looking forward to the next release!

Fedr, 2024-10-30:
The issue should be fixed in MeshLib master. And the fix will be available in next MeshLib release.

closed 2024-10-30 by Fedr