MeshInspector/MeshLib #3522: Introduce I18n Library for Multi-Language Support

feifeizuo, 2024-10-18:
Issue Description：  Currently, the MeshLib library has limited support for multiple languages, which may affect the user experience for non-English speakers. To make MeshLib more accessible to a global audience, it is proposed to integrate an internationalization (I18n) library.

Proposed Enhancement ： It is suggested to incorporate an internationalization (I18n) library, such as gettext or ICU, to enable easy support for multiple languages. This would allow developers and users to select their preferred language when using the library, thereby enhancing user experience and the usability of the library.

An assessment of the current codebase for readiness to integrate I18n is needed to determine specific requirements and steps for introducing the I18n library.
Considering the specificity of different languages, additional support for fonts or encoding might be required.

I hope this suggestion will be considered to make MeshLib more user-friendly and accessible. I can provide translation support for Simplified Chinese.

 Thank you!

Fedr, 2024-10-18:
Thanks for suggestion.

Just to clarify. Do you write your own application based on MeshLib, or would you like to get internationalization support in MeshInspector?

feifeizuo, 2024-10-18:
support in MeshInspector, and also support for this repo(MRViewer and other library)

Fedr, 2024-10-18:
I see, thanks.

It is a big effort for us to introduce internationalization in both MeshLib and MeshInspector. We will need to think more about it.

alpinebuster, 2025-06-19:
> I see, thanks.
> 
> It is a big effort for us to introduce internationalization in both MeshLib and MeshInspector. We will need to think more about it.

The repository maintainers can set up the i18n framework, and the community can help share the workload for the translation tasks.

Fedr, 2026-09-23:
Internationalization support has landed: MeshLib got a gettext-compatible localization system (Boost.Locale backend) in #5768, followed by a series of i18n PRs. MeshInspector now ships with de, es, fr, it, ja, ko, pt, ru and zh (Simplified Chinese). The translations live in the public repo https://github.com/MeshInspector/MeshInspector-translations, including `zh/MeshLib.po` for MRViewer strings, and corrections are welcome there. To localize your own MeshLib-based application, see the [Internationalization Guide](https://meshlib.io/documentation/I18nGuide.html). Closing as completed.

closed 2026-09-23 by Fedr