MeshInspector/MeshLib #3564: how to convert vectorEdges to numpy array?

nanjjines, 2024-10-25:
I extracted the hole edge from the mesh. 
I want to get the extracted hole edge to the numpy array. 
How to do?

`
mesh_data = mrmeshpy.loadMesh(stl_path)
hole_edges = mesh_data.topology.findHoleRepresentiveEdges()

# It's not working
vertices = mrmeshnumpy.getNumpyVerts(hole_edges)
faces = mrmeshnumpy.getNumpyFaces(hole_edges.topology)
`

Grantim, 2024-10-25:
Hello!

Could you please specify what exact result do you want to achieve?

nanjjines, 2024-10-28:
Hello~!
What I want is to find the boundaries in a mesh that only consists of surfaces, like the shell-shaped mesh in the attached image. I want to extract the internal holes of the surface that are defects within the surface itself (as highlighted in the image), not at the boundary, and convert them into a point cloud array, then save that as an .npy file.
[image]

Grantim, 2024-10-28:
Thanks for the description.

In this scenario `findHoleRepresentiveEdges` function will return 3 `EdgeId`s, one for each hole in mesh (upper boundary, lower boundary and small hole in the surface). You will need to distinguish which holes are good and which is not (by length for example).
So you code might look like this:
```py
from meshlib import mrmeshpy as mm
from meshlib import mrmeshnumpy as mm

def holeLength( mesh : mm.Mesh, hole : mm.vectorEdges )->float:
	sumLength = 0.0
	for e in hole:
		org = mesh.topology.org( e )
		dest = mesh.topology.dest( e )
		sumLength += (mesh.points.vec[dest.get()] - mesh.points.vec[org.get()]).length()
	return sumLength

def exportHole( mesh : mm.Mesh, hole : mm.vectorEdges ):
    holePoints = np.ndarray(shape=(len(hole),3),dtype=np.float32)
    for i in range(len(hole)):
        pt = mesh.points.vec[mesh.topology.org(hole[i]).get()]
        holePoints[i][0] = pt.x
        holePoints[i][1] = pt.y
        holePoints[i][2] = pt.z
    return holePoints

mesh = mm.loadMesh("path/to/mesh.stl")
holes = mm.findRightBoundary( mesh.topology ) # find all holes (not only one edge per hole)
for hole in holes:
    if ( holeLength( mesh, hole ) < 1.0 ): # only for holes with small length
        print( exportHole( mesh, hole ) )