MeshInspector/MeshLib #3592: [Request] C++ Viewer Example

Doomerdinger, 2024-10-29:
Is there an example anywhere for the c++ viewer like there is for the python viewer? I see the basic "MainFunc" run the viewer example, but I actually want to interact with the viewer.

I want to use the viewer to display meshes I am manipulating in code for faster feedback than saving to disk and opening in some other software.
The python guide allows for this quite nicely, but I see no similar example for c++.

I am working my way through on my own, using the python example + the backing source code for the python bindings, but I want to make sure I am not overlooking some example somewhere in the codebase.

On a related note, is there any example for Laplacian deformation in c++?

Doomerdinger, 2024-10-29:
I think I more or less have figured out what I need for the c++ viewer by working backwards from the python example.

I am still curious if an example for Laplacian deformation exists somewhere.

Grantim, 2024-10-30:
Hello! Thanks for the request we will improve our examples in future.

As for viewer - it shows `SceneRoot` with its children, so to add object one need to add child to `SceneRoot`
https://github.com/MeshInspector/MeshLib/blob/1e993bb18823e2938b280266bd6589f6e1f5ec08/source/MRViewer/MRViewer.cpp#L1220

If it is for example `std::shared_ptr<ObejctMesh>` you can access `Mesh` that is stored like this:
https://github.com/MeshInspector/MeshLib/blob/1e993bb18823e2938b280266bd6589f6e1f5ec08/source/MRMesh/MRObjectMesh.h#L21-L22

> Note that if you change `mesh.topology` or `mesh.points` directly you need to set corresponding dirty flags into `ObjectMesh` so renderer could see that it needs to update

Some useful examples can be found here https://github.com/MeshInspector/MeshLib/tree/master/source/MRCommonPlugins

---

As for Laplacian deformation, it is much like python (as far as we keep classes and functions interfaces among c++ and python): 
```c++
int main()
{
auto mesh = *MR::MeshLoad::fromAnySupportedFormat( "mesh.stl" ); // on should check that it is valid before dereferencing, here we do it for simplification

auto ancV0 = mesh.topology.getValidVerts().find_first();
auto ancV1 = mesh.topology.getValidVerts().find_last();
MR::VertBitSet freeVerts( mesh.topology.getValidVerts().size() );
freeVerts.set( ancV0 );
freeVerts.set( ancV1 );
MR::expand( mesh.topology, freeVerts, 5 );

MR::Laplacian lDeformer( mesh );
lDeformer.init( freeVerts, MR::EdgeWeights.CotanWithAreaEqWeight );

auto shiftAmount = mesh.computeBoundingBox().diagonal() * 0.01f;
lDeformer.fixVertex( ancV0, mesh.points[ancV0] + mesh.normal(ancV0) * shiftAmount );
lDeformer.fixVertex( ancV1, mesh.points[ancV1] + mesh.normal(ancV1) * shiftAmount );
 
lDeformer.apply();
 
// Invalidate mesh because of external vertices changes
mesh.invalidateCaches(); // also need to set dirty to holder ObjectMesh if this mesh is attached to it

MR::MeshSave::toAnySupportedFormat( mesh, "lDeformedMesh.stl" );

return 0;
}
```

Doomerdinger, 2024-10-31:
Excellent! Thank you very much.

closed 2024-10-31 by Doomerdinger