MeshInspector/MeshLib #3593: SphereParams doesn't accept parameters

VoRodin, 2024-10-30:
I am trying to create mesh of the Sphere in Python API, but when I do it like this:
```python
params=mm.SphereParams(100.0,200)
sphere=mm.makeSphere(params)
```

the next error appears:

```python
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. meshlib.mrmeshpy.SphereParams()
```

I see that by default it takes 1.0, and 100, and in the end I adjusted it via scaling,
but it would be nice to have a straightforward approach.

Grantim, 2024-10-30:
Hello!

Looks like you try to construct `mm.SphereParams` with constructor that does not exist
```py
class SphereParams(pybind11_builtins.pybind11_object)
 |  Generated from:  MR::SphereParams
 |
 |  Method resolution order:
 |      SphereParams
 |      pybind11_builtins.pybind11_object
 |      builtins.object
 |
 |  Methods defined here:
 |
 |  __init__(...)
 |      __init__(*args, **kwargs)
 |      Overloaded function.
 |
 |      1. __init__(self: meshlib.mrmeshpy.SphereParams) -> None
 |
 |      Implicit default constructor.
 |
 |      2. __init__(self: meshlib.mrmeshpy.SphereParams, arg0: meshlib.mrmeshpy.SphereParams) -> None
 |
 |      Implicit copy constructor.
 |
 |  ----------------------------------------------------------------------
 |  Data descriptors defined here:
 |
 |  numMeshVertices
 |
 |  radius
 |
 |  ----------------------------------------------------------------------
 |  Static methods inherited from pybind11_builtins.pybind11_object:
 |
 |  __new__(*args, **kwargs) class method of pybind11_builtins.pybind11_object
 |      Create and return a new object.  See help(type) for accurate signature.
```

This code should work
```py
params=mm.SphereParams()
params.numMeshVertices = 200
params.radius = 100.0
sphere=mm.makeSphere(params)
```

VoRodin, 2024-10-30:
Thank you, it's much easier this way!

closed 2024-10-30 by VoRodin