MeshInspector/MeshLib #3608: Coordinates change after volume rendering

Gooda11, 2024-11-05:
Hello, I was uploading my tiff files using 'settings.voxelSize = mr.Vector3f( )' and 'volume = mr.loadTiffDir(settings)'.  Then, I got a 3D volume. But the coordinates have changed. How can I avoid this, i.e., I want to keep the (x,y) coordinates from my tiff file in the created volume.
Thanks a lot

Grantim, 2024-11-05:
Hello! 

`settings.voxelSize = mr.Vector3f( )` means `mr.Vector3f( 0, 0, 0 )` which is not valid values in this case. You need to specify positive voxel size

Gooda11, 2024-11-05:
Hello, I have specified the number(I didn’t show here). And I can get the results I want . It is just the (x,y) has changed

Grantim, 2024-11-05:
Could you please attach an example how to reproduce the issue?

Grantim, 2026-09-25:
This was solved over email with the provided data. Here is a summary for anyone who runs into the same problem.

TIFF images store only pixel values, so `loadTiffDir` puts the volume in its own voxel coordinates: X is the pixel column, Y is the pixel row (row 0 is the **top** row of the image), and Z is the index of the file (files are sorted by name). After `gridToMesh`, the voxel from column `i`, row `j` of the `k`-th file is located at `(i * voxelSize.x, j * voxelSize.y, k * voxelSize.z)`.

So compared to how the images look in a viewer, the result is shifted (a corner of the images is at the origin) and mirrored in Y (image rows go down, while the Y axis goes up). To get the coordinates you need, apply an affine transformation to the mesh. For example, to put the center of the images at the origin with Y pointing up:

```python
import meshlib.mrmeshpy as mr

settings = mr.VoxelsLoad.LoadingTiffSettings()
settings.dir = r"path/to/tiff/folder"
settings.voxelSize = mr.Vector3f(0.1, 0.1, 0.5) # pixel size in X and Y, distance between slices in Z
volume = mr.VoxelsLoad.loadTiffDir(settings)

gridSettings = mr.GridToMeshSettings()
gridSettings.voxelSize = volume.voxelSize
gridSettings.isoValue = 0.5 * (volume.min + volume.max)
mesh = mr.gridToMesh(volume.data, gridSettings)

# move the center of the images to the origin
size = mr.Vector3f((volume.dims.x - 1) * volume.voxelSize.x, (volume.dims.y - 1) * volume.voxelSize.y, 0)
translation = mr.AffineXf3f.translation(-size * 0.5)
# mirror Y, so the top row of the images gets the largest Y
flipY = mr.AffineXf3f.linear(mr.Matrix3f.scale(1, -1, 1))
mesh.transform(flipY * translation) # translation is applied first
# mirroring turns the mesh inside out, restore the orientation of its faces
mesh.topology.flipOrientation()

mr.saveMesh(mesh, "mesh.stl")
```

Any other placement (a known scan origin, a rotation, flipped Z) works the same way: build the `AffineXf3f` you need and pass it to `mesh.transform`. Only transformations that mirror the mesh (negative determinant) need `flipOrientation()`.

Closing the issue, feel free to reopen it if something is still unclear.

closed 2026-09-25 by Grantim