MeshInspector/MeshLib #3623: Sorry,this is  no issue .   How to get the wasm file of MeshLib, I want to use it in the browser

LoongZP, 2024-11-07:


Grantim, 2024-11-07:
Hello!

Now we don't distribute wasm binaries, only way is to build meshlib locally by this guide:
https://github.com/MeshInspector/MeshLib?tab=readme-ov-file#wasmemscripten

LoongZP, 2024-11-07:
Hello!

Can MeshLib handle non-manifold models?

Grantim, 2024-11-07:
MeshLib internal mesh representation forbid non-manifold topologies, so any mesh converted into MeshLib format will be manifold (some vertices might be duplicated and some faces might be skipped to achieve this).

LoongZP, 2024-11-08:
When I try to build a docker image via emscriptenDockerfile, I get an error. I can't figure it out.

[image]

[image]

Grantim, 2024-11-08:
Hello!
It looks like you have not cloned submodules if MeshLib
please call
```
git submodule update --init --recursive
```

lslzl3000, 2025-03-22:
> Hello!
> 
> Now we don't distribute wasm binaries, only way is to build meshlib locally by this guide: https://github.com/MeshInspector/MeshLib?tab=readme-ov-file#wasmemscripten

cannot find the guide now, and any tutorials for wasm version? how to use it in browser env?

Grantim, 2025-03-24:
We moved our documentation here https://meshlib.io/documentation/MeshLibCppSetupGuide.html#CppSetupWASM

Fedr, 2026-09-23:
MeshLib is now published prebuilt for WebAssembly, so building it locally is no longer necessary:

- npm: [`@meshinspector/meshlib`](https://www.npmjs.com/package/@meshinspector/meshlib) (single-threaded) and [`@meshinspector/meshlib-mt`](https://www.npmjs.com/package/@meshinspector/meshlib-mt) (multi-threaded, requires a cross-origin isolated page)
- In the browser without npm:
  ```js
  import createMeshLib from 'https://js.meshlib.io/meshlib/meshlib.mjs';
  const ml = await createMeshLib();
  ```
- Setup guide: https://meshlib.io/documentation/MeshLibJsSetupGuide.html

Closing this issue; feel free to reopen if something is missing.

closed 2026-09-23 by Fedr