MeshInspector/MeshLib #3729: Adapting a mesh to a 3D Curve

Doomerdinger, 2024-11-21:
I am attempting to deform a mesh such that it connects with a line in 3D space, with some constraints.
For context if anyone is familiar with the dental domain, I am trying to attach a tooth to a margin line.

Currently I am selecting/and or creating vertices on the mesh and dragging them to points on the line, applying a laplacian deformation.
I won't get in to how I select the points on the mesh or how I determine exactly where I am dragging them to, but I can if that information seems like it may be relevant.

[image]
My current modifications. You can see a large amount of concavity in this region.
[image]
The input mesh I am modifying.
[image]
An example of what may be closer to ideal. The mesh maintains a more convex shape where possible, before ultimately becoming concave because the line in space I am fitting to forces it.

Before I continue down the path of something more custom, any chance there are existing tips/tricks/examples to push my results more to what I desire?

Grantim, 2024-11-22:
Hello!

One thing you can try is to use `FreeFormBestFit` 
https://github.com/MeshInspector/MeshLib/blob/deb3d3221ef84a7414f31795ad6c05baf3c428e8/source/MRMesh/MRFreeFormDeformer.h#L47-L58

To construct this class you need to path "tooth" bounding box and resolution of free form transformer
[image][image]

Then you need to add pairs of points to fit, I suggest you to make Polyline3 from "margin" line 
https://github.com/MeshInspector/MeshLib/blob/deb3d3221ef84a7414f31795ad6c05baf3c428e8/source/MRMesh/MRPolyline.h#L14-L29

And project vertices of the polyline on mesh with
https://github.com/MeshInspector/MeshLib/blob/deb3d3221ef84a7414f31795ad6c05baf3c428e8/source/MRMesh/MRMeshProject.h#L27-L40

So for each point of "margin" you will have corresponding point on "tooth".

Add these pairs to `FreeFormBestFit` and find best transformation:
https://github.com/MeshInspector/MeshLib/blob/deb3d3221ef84a7414f31795ad6c05baf3c428e8/source/MRMesh/MRFreeFormDeformer.h#L68-L69

Then you need to apply it to "tooth" like [this](https://meshlib.io/documentation/ExampleFreeForm.html) (it is example in python, but c++ code should be similar)
https://github.com/MeshInspector/MeshLib/blob/deb3d3221ef84a7414f31795ad6c05baf3c428e8/source/MRMesh/MRFreeFormDeformer.h#L10-L25
https://github.com/MeshInspector/MeshLib/blob/deb3d3221ef84a7414f31795ad6c05baf3c428e8/source/MRMesh/MRFreeFormDeformer.h#L34

What do you think?

Doomerdinger, 2024-12-03:
I'll give this a shot here in the near future and report back!
I got things generally working how I would like based on my existing methods, but I'd like to see how this alternative would work.