MeshInspector/MeshLib #3739: C# examples?

seghier, 2024-11-24:
Hi
Is there any examples how to create mesh from pointcloud, and how to use remesh and read openvdb file?
And in this example i don't find ` Mesh.FromAnySupportedFormat`

```
using MR.DotNet;
using System;
using System.Reflection;
 
class Program
{
    static void Main(string[] args)
    {
        if (args.Length != 1 && args.Length != 2)
        {           
            Console.WriteLine("Usage: {0} INPUT [OUTPUT]", Assembly.GetExecutingAssembly().GetName().Name);
            return;
        }
 
        try
        {
            string input = args[0];
            string output = args.Length == 2 ? args[1] : args[0];
 
            var mesh = Mesh.FromAnySupportedFormat( input );
 
            DecimateParameters dp = new DecimateParameters();
            dp.strategy = DecimateStrategy.MinimizeError;
            dp.maxError = 1e-5f * mesh.BoundingBox.Diagonal();
            dp.tinyEdgeLength = 1e-3f;
            dp.packMesh = true;
 
            var result = [MeshDecimate](https://meshlib.io/documentation/namespaceMeshDecimate.html).Decimate(mesh, dp);
            Mesh.ToAnySupportedFormat(mesh, output);
        }
        catch (Exception e)
        {
            Console.WriteLine("Error: {0}", e.Message);
        }
    }
}
```

seghier, 2024-11-24:
I try this to open **vdb file or fbx file** but i get an error

`1. An error occurred: unsupported file extension`

Rhino.Geometry.Mesh MeshLidToRhinoMesh(MR.DotNet.Mesh mesh)
{
    // Create Rhino mesh
    var rhinoMesh = new Rhino.Geometry.Mesh();

    // Add vertices
    foreach (var point in mesh.Points)
    {
        rhinoMesh.Vertices.Add(point.X, point.Y, point.Z);
    }

    // Add faces
    foreach (var triangle in mesh.Triangulation)
    {
        rhinoMesh.Faces.AddFace(triangle.v0.Id, triangle.v1.Id, triangle.v2.Id);
    }
    rhinoMesh.Compact();
    return rhinoMesh;
}

Rhino.Geometry.Mesh ReadFile(string file)
{
    MR.DotNet.Mesh mesh = MR.DotNet.MeshLoad.FromAnySupportedFormat(file);
    var rhinoMesh = MeshLidToRhinoMesh(mesh);
    return rhinoMesh;
}
```

astrowander, 2024-11-25:
Hi

Could you please clarify a couple of questions:

1. What do you mean by creating mesh from point cloud? Just triangulation or something else? Triangulation is supported in MeshLib but it is not exported to C# version yet,  as well as Remesh.

2. FBX file format is not now supported by MeshLib. VDB contains volumetric data, not meshes. That's why it's impossible to read it with MeshLoad class. So what exactly would you like to do? Read a volume from VDB and convert it to a mesh? This is possible with MeshLib, but no voxel-related are exported to C# yet.

We can plan to add these features to C# version of MeshLib, but we need more info about your request.

yuvashrikarunakaran, 2024-12-02:
1.Creating a mesh from a point cloud typically involves generating a connected geometric structure (usually a triangulated mesh) that represents the surface of an object defined by a set of unstructured points in 3D space. This process can range from simple triangulation to more complex methods like surface reconstruction and remeshing depending on the desired outcome.
example:
using System;
using System.Collections.Generic;
using System.Linq;

namespace PointCloudToMesh
{
    public class Point
    {
        public double X { get; set; }
        public double Y { get; set; }

        public Point(double x, double y)
        {
            X = x;
            Y = y;
        }

        public override string ToString() => $"({X}, {Y})";
    }

    public class Triangle
    {
        public Point A { get; }
        public Point B { get; }
        public Point C { get; }

        public Triangle(Point a, Point b, Point c)
        {
            A = a;
            B = b;
            C = c;
        }

        public override string ToString() => $"Triangle: {A}, {B}, {C}";
    }

    public class DelaunayTriangulation
    {
        private List<Triangle> triangles = new List<Triangle>();

        public List<Triangle> Triangulate(List<Point> points)
        {
            if (points == null || points.Count < 3)
                throw new ArgumentException("At least three points are required to create a triangulation.");

           
            Point pMin = new Point(points.Min(p => p.X), points.Min(p => p.Y));
            Point pMax = new Point(points.Max(p => p.X), points.Max(p => p.Y));
            double dx = pMax.X - pMin.X;
            double dy = pMax.Y - pMin.Y;
            double deltaMax = Math.Max(dx, dy) * 2;

            Point p1 = new Point(pMin.X - deltaMax, pMin.Y - deltaMax);
            Point p2 = new Point(pMax.X + deltaMax, pMin.Y - deltaMax);
            Point p3 = new Point((pMin.X + pMax.X) / 2, pMax.Y + deltaMax);

            Triangle superTriangle = new Triangle(p1, p2, p3);
            triangles.Add(superTriangle);

            
            foreach (var point in points)
            {
                List<Triangle> badTriangles = new List<Triangle>();
                foreach (var triangle in triangles)
                {
                    if (IsPointInCircumcircle(point, triangle))
                        badTriangles.Add(triangle);
                }

                List<(Point, Point)> polygon = new List<(Point, Point)>();
                foreach (var badTriangle in badTriangles)
                {
                    triangles.Remove(badTriangle);
                    polygon.Add((badTriangle.A, badTriangle.B));
                    polygon.Add((badTriangle.B, badTriangle.C));
                    polygon.Add((badTriangle.C, badTriangle.A));
                }

                polygon = RemoveDuplicateEdges(polygon);

                foreach (var edge in polygon)
                {
                    triangles.Add(new Triangle(edge.Item1, edge.Item2, point));
                }
            }

          
            triangles.RemoveAll(t => t.A == p1 || t.B == p1 || t.C == p1 ||
                                      t.A == p2 || t.B == p2 || t.C == p2 ||
                                      t.A == p3 || t.B == p3 || t.C == p3);

            return triangles;
        }

        private bool IsPointInCircumcircle(Point p, Triangle t)
        {
            double ax = t.A.X - p.X, ay = t.A.Y - p.Y;
            double bx = t.B.X - p.X, by = t.B.Y - p.Y;
            double cx = t.C.X - p.X, cy = t.C.Y - p.Y;

            double det = (ax * (by * cy - by * cy) - ay * (bx * cx - cx * bx) + ax * (bx * cy - by * cx));
            return det > 0;
        }

        private List<(Point, Point)> RemoveDuplicateEdges(List<(Point, Point)> edges)
        {
            var uniqueEdges = new List<(Point, Point)>();
            foreach (var edge in edges)
            {
                var reversedEdge = (edge.Item2, edge.Item1);
                if (!uniqueEdges.Contains(edge) && !uniqueEdges.Contains(reversedEdge))
                    uniqueEdges.Add(edge);
            }
            return uniqueEdges;
        }
    }

    class Program
    {
        static void Main(string[] args)
        {
            List<Point> pointCloud = new List<Point>
            {
                new Point(0, 0),
                new Point(1, 0),
                new Point(0, 1),
                new Point(1, 1),
                new Point(0.5, 0.5)
            };

            var delaunay = new DelaunayTriangulation();
            List<Triangle> mesh = delaunay.Triangulate(pointCloud);

            Console.WriteLine("Generated Mesh:");
            foreach (var triangle in mesh)
            {
                Console.WriteLine(triangle);
            }
        }
    }
}
2.The scenario you're describing involves converting volumetric data  into a mesh representation. The process requires reading the VDB volumetric data and using algorithms to generate a mesh representation, such as through isosurface extraction . This is feasible using libraries like OpenVDB and MeshLib (for mesh generation), but as you mentioned, C# bindings for these libraries are currently limited.

seghier, 2024-12-02:
> Hi
>
> Could you please clarify a couple of questions:
>
> 1. What do you mean by creating mesh from point cloud? Just triangulation or something else? Triangulation is supported in MeshLib but it is not [...]

Hi
Thanks for your reply.
Yes i talk about triangulated mesh from pointcloud and vdb if posiible  to pointloud.
I see these features in your videos , and i hope you add them to c#

astrowander, 2024-12-03:
Hi
Triangulation of point cloud is already available in C# MeshLib. Here is an example:
`
using static MR.DotNet;

public static class TriangulationExample
{
    public static void Run()
    {
        try
        {
            PointCloud pc = PointCloud.FromAnySupportedFormat("points.ply");
            var mesh = TriangulatePointCloud(pc, new TriangulationParameters());
            if ( mesh is not null )
                MeshSave.ToAnySupportedFormat(mesh, "triangulated_mesh.stl");
        }
        catch (Exception e)
        {
            Console.WriteLine("Error: {0}", e.Message);
        }
    }
}
`
Reading voxel data from VDB is not implemented yet. We could plan it in future.

astrowander, 2024-12-03:
Please check the last pre-release version 3.0.1.294-beta.

seghier, 2024-12-03:
> Please check the last pre-release version 3.0.1.294-beta.

Thank you @astrowander 
Triangulate pointcloud works !
I see this dll : `MRVoxels.dll` is that mean we can read vdb file now?

astrowander, 2024-12-06:
Hi
Yes, we are going to add reading of vdb in c# next week.
According to the reading list of points, please, clarify what do you mean? Now we support a very simple point cloud format .xyz, which is actually just list of coordinates. Is it what you look for?

seghier, 2024-12-06:
> Yes, we are going to add reading of vdb in c# next week.

That's great, thank you.

I load ply file in Rhinoand i use rhinocommon api to color the mesh created with MeshLib, but it's better if MeshLib can read the colors

[image]

> According to the reading list of points,

The api read file from path , in Rhino or Grasshopper or any other software if pointcloud is exist in the scene or any list of points and we want convert them to mesh we need to read their coordinates directly

astrowander, 2024-12-06:
Okay, we will add reading colors from .PLY to c# too.
Also, if you have already loaded a point cloud, you can access to its list of points with Points property

seghier, 2024-12-06:
> Okay, we will add reading colors from .PLY to c# too. Also, if you have already loaded a point cloud, you can access to its list of points with Points property

Yes i did that, but if the list points is created inside Grasshopper we need to use it directly
i use this function to convert list of points to `MR.DotNet.PointCloud` then `MR.DotNet.Mesh`

```
private MR.DotNet.Mesh MrMesh(List<Point3d> points)
{
    // Validate input
    if (points == null || points.Count < 3)
    {
        AddRuntimeMessage(GH_RuntimeMessageLevel.Warning, "Input points must be non-null and contain at least 3 points.");
        return null;
    }

    MR.DotNet.PointCloud pointCloud = new MR.DotNet.PointCloud();
    MR.DotNet.Mesh mesh = null;

    try
    {
        // Convert points with error checking
        foreach (var p in points)
        {
            // Validate each point to prevent potential NaN or infinite values
            if (double.IsNaN(p.X) || double.IsInfinity(p.X) ||
                double.IsNaN(p.Y) || double.IsInfinity(p.Y) ||
                double.IsNaN(p.Z) || double.IsInfinity(p.Z))
            {
                AddRuntimeMessage(GH_RuntimeMessageLevel.Warning, $"Invalid point encountered: {p}. Skipping.");
                continue;
            }

            pointCloud.AddPoint(new MR.DotNet.Vector3f((float)p.X, (float)p.Y, (float)p.Z));
        }

        // Check if we have enough points after filtering
        if (pointCloud.Points.Count < 3)
        {
            AddRuntimeMessage(GH_RuntimeMessageLevel.Warning, "Insufficient valid points to create mesh.");
            return null;
        }

        // Triangulate with default parameters
        mesh = TriangulatePointCloud(pointCloud, new TriangulationParameters());

        // Check if mesh is valid
        if (mesh == null)
        {
            AddRuntimeMessage(GH_RuntimeMessageLevel.Warning, "Mesh triangulation failed.");
            return null;
        }

        // Hole filling with additional checks
        if (mesh.HoleRepresentiveEdges != null && mesh.HoleRepresentiveEdges.Count > 2)
        {
            try
            {
                MR.DotNet.FillHoles(ref mesh, mesh.HoleRepresentiveEdges.ToList(), new FillHoleParams());
            }
            catch (Exception holeFillEx)
            {
                AddRuntimeMessage(GH_RuntimeMessageLevel.Warning, $"Hole filling failed: {holeFillEx.Message}");
                // Optionally, you might want to return the mesh without hole filling
            }
        }

        return mesh;
    }
    catch (Exception ex)
    {
        // More detailed error logging
        AddRuntimeMessage(GH_RuntimeMessageLevel.Error, $"Mesh generation failed: {ex.Message}");

        // Optional: Log more details if needed
        // LogError(ex);

        return null;
    }
    finally
    {
        // Dispose of point cloud if it implements IDisposable
        (pointCloud as IDisposable)?.Dispose();
    }
}
```

astrowander, 2024-12-06:
You would like to create a PointCloud directly from list of points, right?

astrowander, 2024-12-06:
Okay, we will make a such function

seghier, 2024-12-06:
Great thank you

astrowander, 2024-12-06:
Please check version 3.0.1.318-beta. We have added function PointCloud.FromPoints(). It receives any collection of Vector3f and returns PointCloud that then can be triangulated.

```
var points = new List<Vector3f>();
//fill list with points
var mesh = TriangulatePointCloud(PointCloud.FromPoints(points), new TriangulationParameters());
```

Also note, that we have moved PointCloud.FromAnySupportedFormat and PointCloud.ToAnySupportedFormat to PointsLoad and PointsSave classes respectively.

seghier, 2024-12-06:
Thank you so much
Excellent !

seghier, 2024-12-12:
Hello again
Thank you for the new update, i try this code to create a mesh from vdb file but i get null result

```
 var voxel = MR.DotNet.VoxelsLoad.FromAnySupportedFormat(path);
 foreach (var v in voxel)
 {
     mesh = MR.DotNet.GridToMesh(v.Data, new GridToMeshSettings());
 }
```
Is there any example code available?

astrowander, 2024-12-13:
Hi, 
You need to specify iso value and voxel size. Please note then iso must be between VdbVolume.Min and VdbVolume.Max

```
 var gridToMeshSettings = new GridToMeshSettings();
 gridToMeshSettings.voxelSize = Vector3f.Diagonal(0.1f);
 gridToMeshSettings.isoValue = 1;
mesh = MR.DotNet.GridToMesh(v.Data, gridToMeshSettings );
```
If you issue still persists after that please send me your VDB file, I will check it

seghier, 2024-12-13:
Thank you @astrowander 
I try to use a value between min max grid size and i use the same value of voxelsize but i get invalid mesh and all vertices equal to (0,0,0)
[image]

`MR.DotNet.Mesh mesh = null;
var voxel = MR.DotNet.VoxelsLoad.FromAnySupportedFormat(file);
var settings = new GridToMeshSettings();
settings.isoValue = 1;
var min = voxel[0].Min;
var max = voxel[0].Max;
var size = voxel[0].VoxelSize;
settings.isoValue = (float)iso;
settings.voxelSize = new MR.DotNet.Vector3f(0.05f, 0.05f, 0.05f);
//settings.voxelSize = MR.DotNet.Vector3f.Diagonal(0.1f);
mesh = MR.DotNet.GridToMesh(voxel[0].Data, new GridToMeshSettings());`

Is it possible to get grid points to output pointcloud? i think this will be faster to preview the vdb file specially with volumetric smoke
and can we also load animated vdb and choose frame ?
 i use files from official website: bunny, dragon and fire

[image]

and for smoke2 example i get this error
                var voxel = MR.DotNet.VoxelsLoad.FromAnySupportedFormat(file);
[image]

`System.AccessViolationException: 'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.'`

astrowander, 2024-12-13:
I confirm the error for 'smoke2.vdb', we are working on it now.
But for other models it seems pretty well. Below is my code. It loads voxels, converts first volume to mesh and lists first 10 and last 10 points. They are not zeros actually. Could you provide exact iso value? For average iso between min and max it outputs the following (for dragon.vdb).

```
using static MR.DotNet;

namespace Sample
{
    internal class Program
    {       
        static void Main(string[] args)
        {
            string path = "C:/Adalisk/models/vdb/dragon.vdb";
            Console.WriteLine("Loading " + path.Split('/').Last());

            var volumes = VoxelsLoad.FromAnySupportedFormat(path);
            Console.WriteLine("Loaded " + volumes.Count + " volumes");

            var gridToMeshSettings = new GridToMeshSettings();
            gridToMeshSettings.voxelSize = volumes[0].VoxelSize;
            gridToMeshSettings.isoValue = volumes[0].Min + (volumes[0].Max - volumes[0].Min) / 2;

            var mesh = GridToMesh(volumes[0].Data, gridToMeshSettings);
            Console.WriteLine("Mesh has " + mesh.Points.Count + " vertices");
            
            Console.WriteLine("First 10 vertices:");
            for (int i = 0; i < 10; i++)
                Console.WriteLine( "Point " + i + ": " + mesh.Points[i].X + " " + mesh.Points[i].Y + " " + mesh.Points[i].Z);

            Console.WriteLine( "Last 10 vertices:");
            for (int i = mesh.Points.Count - 10; i < mesh.Points.Count; i++)
                Console.WriteLine( "Point " + i + ": " + mesh.Points[i].X + " " + mesh.Points[i].Y + " " + mesh.Points[i].Z);
        }
    }
}

![Image](https://github.com/user-attachments/assets/1a657105-1945-4ba1-a4e7-7a27dc7e87b8)

```

seghier, 2024-12-13:
Thank you very much, my mistake i forget to set settings and i used new GridToMeshSettings()) :D

[image]

Can we decrease grid size to speedup the meshing? and can we get grid points?

seghier, 2024-12-13:
I notice when we set isoValue to min or max value we get invalid mesh

astrowander, 2024-12-13:
Yes, I understand what you mean. We can add a function for grid resampling in the next week. If you want to get points you can use the Points property of a Mesh object. Direct conversion is not supported.

Also yo could try a new version 367-beta. We have added loading and saving colors of points. Below is an example:
```
 public void TestSaveLoadWithColors()
 {
   var points = MakeCube();
   var colors = new List<Color>(8);
   colors.Add(new Color(1.0f, 0.0f, 0.0f));
   colors.Add(new Color(0.0f, 1.0f, 0.0f));
   colors.Add(new Color(0.0f, 0.0f, 1.0f));
   colors.Add(new Color(1.0f, 1.0f, 0.0f));
   colors.Add(new Color(1.0f, 0.0f, 1.0f));
   colors.Add(new Color(0.0f, 1.0f, 1.0f));
   colors.Add(new Color(1.0f, 1.0f, 1.0f));
   colors.Add(new Color(0.0f, 0.0f, 0.0f));
  
   var saveSettings = new SaveSettings();
   saveSettings.colors = new VertColors(colors);
   
   string path = Path.GetTempFileName() + ".ply";
   PointsSave.ToAnySupportedFormat(points, path, saveSettings);
  
   var loadSettings = new PointsLoadSettings();
   loadSettings.colors = new VertColors();
   var readPoints = PointsLoad.FromAnySupportedFormat(path, loadSettings);
   Assert.That(points.Points.Count == 8);
  
   var readColors = loadSettings.colors.ToList();
   Assert.That(colors.Count == readColors.Count);
   for (int i = 0; i < colors.Count; i++)
   {
       Assert.That(colors[i] == readColors[i]);
   }
  
   File.Delete(path);
}
```

astrowander, 2024-12-13:
> I notice when we set isoValue to min or max value we get invalid mesh and for some values close to min or max also give invalid mesh, i see this with` fire.vdb` file

I suppose that it could not find enough points to build a valid mesh if iso is close to min or max

seghier, 2024-12-13:
> Also yo could try a new version 367-beta. We have added loading and saving colors of points. Below is an example:

That's great i will try it with colored pointcloud

> > I notice when we set isoValue to min or max value we get invalid mesh and for some values close to min or max also give invalid mesh, i see this with` fire.vdb` file
> 
> I suppose that it could not find enough points to build a valid mesh if iso is close to min or max

In grasshopper the mesh is visible but invalid , i don't know if this related to rhino mesh or meshlib mesh because i can't view meshlib mesh directly.

https://github.com/user-attachments/assets/7fe3ea65-29a4-46a4-ba1b-0ea7cfb15f99

seghier, 2024-12-13:
Loading pointcloud colors works as expected thank you

[image]

seghier, 2024-12-14:
> We can add a function for grid resampling in the next week.

Hi , can you add available data of vdb if it is smoke for example like density ?

astrowander, 2024-12-16:
Hi, we can load dense volumes from VDB files. Do you want to get the direct access to each voxel?

seghier, 2024-12-16:
That's great thank you
What do you mean by `direct access` ?

astrowander, 2024-12-16:
I mean, are you willing to get a density value stored in each voxel?

seghier, 2024-12-16:
Hi
The new setting is very useful to quickly load the vdb file, can you add min max range information for all settings?
`settings.adaptivity = 0.8f;`

and how to use these ? because when i use these value i get error: System.SystemException: 'Vertices number limit exceeded.'
```
settings.maxVertices = 2000;
settings.maxFaces = 2000;
```
Can we get grid points and density in the new update?

astrowander, 2024-12-17:
In new version 385-beta we have added functions fo grid resampling. You can increase voxel size and thus reduce grid dimensions. Example:
```
var vdbVolume = CreateVolume(); // e.g. read from VDB file
var resampledGrid = Resampled(vdbVolume.Data, 2); // increase voxel by two times
var resampledVolume = FloatGridToVdbVolume(resampledGrid); // generate new volume
resampledVolume.VoxelSize = vdbVolume.VoxelSize * 2; // you need to increase voxel size in the new volume manually if you want to preserve physical size of the model
```

Also you can read and set values of each voxel or of region of voxels:
```
var vdbVolume = CreateVolume();
var p = new Vector3i();
Assert.That( GetValue( vdbVolume.Data, p ) == 3.0f );

var region = new VoxelBitSet( vdbVolume.Dims.X * vdbVolume.Dims.Y * vdbVolume.Dims.Z );
region.Set(0);
SetValue( vdbVolume.Data, region, 1.0f );
Assert.That( GetValue( vdbVolume.Data, p ) == 1.0f );

SetValue( vdbVolume.Data, p, 2.0f );
Assert.That( GetValue( vdbVolume.Data, p ) == 2.0f );
```

You can find info about limits in comments in source code. For example for adaptivity:
```
/// adaptivity - [0.0;1.0] ratio of combining small triangles into bigger ones (curvature can be lost on high values)
```

MaxVertices and maxFaces are needed to constrain count of vertices or faces in resulting mesh. It is impossible to predict how many of them will be in resulting mesh so we throw the exception if they are exceeded. If that happens try to increase voxel size.

seghier, 2024-12-17:
Thank you very much

Resample grid works great but i don't understand about region what is the result expected: part of vdb or section?

```
 private VdbVolume ResampleVDB(VdbVolume vdbVolume, float vdbScale)
 {
     var resampledGrid = Resampled(vdbVolume.Data, vdbScale);
     var resampledVolume = FloatGridToVdbVolume(resampledGrid); 
     resampledVolume.VoxelSize = vdbVolume.VoxelSize * vdbScale;
     return resampledVolume;
 }

 private FloatGrid VoxelValuesRegion(VdbVolume vdbVolume, float value)
 {
     var p = new Vector3i();
     //MR.DotNet.GetValue(vdbVolume.Data, p);
     var region = new VoxelBitSet(vdbVolume.Dims.X * vdbVolume.Dims.Y * vdbVolume.Dims.Z);
     region.Set(0);
     MR.DotNet.SetValue(vdbVolume.Data, region, 1.0f);
     return vdbVolume.Data;
 }

 private VdbVolume VoxelValuesVector(VdbVolume vdbVolume, float value)
 {
     var p = new Vector3i();
     //MR.DotNet.GetValue(vdbVolume.Data, p);
     MR.DotNet.SetValue(vdbVolume.Data, p, value);
     return vdbVolume;
 }
```

astrowander, 2024-12-17:
Region is a bitset where you set indices of voxels you want to change. In the example you activate only 0-th voxel. And its value will be set to 1.0f. You could also activate other indices and change several voxels at one time:
```
var region = new VoxelBitSet(vdbVolume.Dims.X * vdbVolume.Dims.Y * vdbVolume.Dims.Z);
     region.Set(0);
     region.Set(2);
     region.Set(6);
 MR.DotNet.SetValue(vdbVolume.Data, region, 1.0f); // three voxels will be set to 1.0f
```

seghier, 2024-12-17:
Thanks @astrowander 
And grid points still not available?

seghier, 2024-12-17:
Looks like `Resampled `create two meshes ?

```
{
........
var voxel = MR.DotNet.VoxelsLoad.FromAnySupportedFormat(file);
var settings = new GridToMeshSettings();
var min = voxel[0].Min;
var max = voxel[0].Max;
settings.adaptivity = (float)adapt;
settings.isoValue = RemapToRange(iso, max, min);
settings.voxelSize = voxel[0].VoxelSize;
settings.relaxDisorientedTriangles = true;

MR.DotNet.Mesh mesh = null;
var resampled = ResampleVDB(voxel[0], (float)scale);

mesh = MR.DotNet.GridToMesh(resampled, settings);
var rmesh = .MeshLibToRhinoMesh(mesh);
var scaleMesh = Transform.Scale(Point3d.Origin, scale);
rmesh.Transform(scaleMesh);
.....
}

private  Rhino.Geometry.Mesh MeshLibToRhinoMesh(MR.DotNet.Mesh mesh)
{
    Rhino.Geometry.Mesh rhinoMesh = new Rhino.Geometry.Mesh();
    try
    // Add vertices
    {
        foreach (MR.DotNet.Vector3f point in mesh.Points)
            rhinoMesh.Vertices.Add(point.X, -point.Z, point.Y);

        // Add faces
        foreach (ThreeVertIds threeVertIds in mesh.Triangulation)
        {
            rhinoMesh.Faces.AddFace(threeVertIds.v0.Id, threeVertIds.v1.Id, threeVertIds.v2.Id);
        }

        rhinoMesh.Compact();
        rhinoMesh.RebuildNormals();
    }
    catch
    {

    }
    return rhinoMesh;
}

private float RemapToRange(double input, double min, double max)
{
    if (input < 0 || input > 1)
        throw new ArgumentOutOfRangeException(nameof(input), "Input must be in the range [0, 1]");

    return (float)(min + input * (max - min));
}

 private FloatGrid ResampleVDB(VdbVolume vdbVolume, float vdbScale)
 {
     var resampledGrid = Resampled(vdbVolume.Data, vdbScale);
     return resampledGrid;
 }
```

andrewerf, 2025-01-13:
@seghier We reproduced the issue with resampling and are currently working on the fix.

andrewerf, 2025-01-19:
@seghier Hello, the fix for resampling is now implemented in master.

seghier, 2025-05-10:
Hello @andrewerf 
After the latest update i get this error 
`1. Error loading voxel file: Unable to load DLL 'MRMeshC': The specified module could not be found. (Exception from HRESULT: 0x8007007E)`

seghier, 2025-05-10:
Thanks @andrewerf 
When i compile the plugin some dlls are missing

[image]

Grantim, 2025-05-12:
Hello @seghier !

Could you please provide more details what plugin and how did you build it?

closed 2025-06-22 by Fedr