MeshInspector/MeshLib #3794: Creating a Mesh Issue

yuvashrikarunakaran, 2024-12-02:
> 1.Creating a mesh from a point cloud typically involves generating a connected geometric structure (usually a triangulated mesh) that represents the surface of an object defined by a set of unstructured points in 3D space. This process can range from simple triangulation to more complex methods like surface reconstruction and remeshing depending on the desired outcome.
> example:
> using System;
> using System.Collections.Generic;
> using System.Linq;
> 
> namespace PointCloudToMesh
> {
>     public class Point
>     {
>         public double X { get; set; }
>         public double Y { get; set; }
> 
>         public Point(double x, double y)
>         {
>             X = x;
>             Y = y;
>         }
> 
>         public override string ToString() => $"({X}, {Y})";
>     }
> 
>     public class Triangle
>     {
>         public Point A { get; }
>         public Point B { get; }
>         public Point C { get; }
> 
>         public Triangle(Point a, Point b, Point c)
>         {
>             A = a;
>             B = b;
>             C = c;
>         }
> 
>         public override string ToString() => $"Triangle: {A}, {B}, {C}";
>     }
> 
>     public class DelaunayTriangulation
>     {
>         private List<Triangle> triangles = new List<Triangle>();
> 
>         public List<Triangle> Triangulate(List<Point> points)
>         {
>             if (points == null || points.Count < 3)
>                 throw new ArgumentException("At least three points are required to create a triangulation.");
> 
>            
>             Point pMin = new Point(points.Min(p => p.X), points.Min(p => p.Y));
>             Point pMax = new Point(points.Max(p => p.X), points.Max(p => p.Y));
>             double dx = pMax.X - pMin.X;
>             double dy = pMax.Y - pMin.Y;
>             double deltaMax = Math.Max(dx, dy) * 2;
> 
>             Point p1 = new Point(pMin.X - deltaMax, pMin.Y - deltaMax);
>             Point p2 = new Point(pMax.X + deltaMax, pMin.Y - deltaMax);
>             Point p3 = new Point((pMin.X + pMax.X) / 2, pMax.Y + deltaMax);
> 
>             Triangle superTriangle = new Triangle(p1, p2, p3);
>             triangles.Add(superTriangle);
> 
>             
>             foreach (var point in points)
>             {
>                 List<Triangle> badTriangles = new List<Triangle>();
>                 foreach (var triangle in triangles)
>                 {
>                     if (IsPointInCircumcircle(point, triangle))
>                         badTriangles.Add(triangle);
>                 }
> 
>                 List<(Point, Point)> polygon = new List<(Point, Point)>();
>                 foreach (var badTriangle in badTriangles)
>                 {
>                     triangles.Remove(badTriangle);
>                     polygon.Add((badTriangle.A, badTriangle.B));
>                     polygon.Add((badTriangle.B, badTriangle.C));
>                     polygon.Add((badTriangle.C, badTriangle.A));
>                 }
> 
>                 polygon = RemoveDuplicateEdges(polygon);
> 
>                 foreach (var edge in polygon)
>                 {
>                     triangles.Add(new Triangle(edge.Item1, edge.Item2, point));
>                 }
>             }
> 
>           
>             triangles.RemoveAll(t => t.A == p1 || t.B == p1 || t.C == p1 ||
>                                       t.A == p2 || t.B == p2 || t.C == p2 ||
>                                       t.A == p3 || t.B == p3 || t.C == p3);
> 
>             return triangles;
>         }
> 
>         private bool IsPointInCircumcircle(Point p, Triangle t)
>         {
>             double ax = t.A.X - p.X, ay = t.A.Y - p.Y;
>             double bx = t.B.X - p.X, by = t.B.Y - p.Y;
>             double cx = t.C.X - p.X, cy = t.C.Y - p.Y;
> 
>             double det = (ax * (by * cy - by * cy) - ay * (bx * cx - cx * bx) + ax * (bx * cy - by * cx));
>             return det > 0;
>         }
> 
>         private List<(Point, Point)> RemoveDuplicateEdges(List<(Point, Point)> edges)
>         {
>             var uniqueEdges = new List<(Point, Point)>();
>             foreach (var edge in edges)
>             {
>                 var reversedEdge = (edge.Item2, edge.Item1);
>                 if (!uniqueEdges.Contains(edge) && !uniqueEdges.Contains(reversedEdge))
>                     uniqueEdges.Add(edge);
>             }
>             return uniqueEdges;
>         }
>     }
> 
>     class Program
>     {
>         static void Main(string[] args)
>         {
>             List<Point> pointCloud = new List<Point>
>             {
>                 new Point(0, 0),
>                 new Point(1, 0),
>                 new Point(0, 1),
>                 new Point(1, 1),
>                 new Point(0.5, 0.5)
>             };
> 
>             var delaunay = new DelaunayTriangulation();
>             List<Triangle> mesh = delaunay.Triangulate(pointCloud);
> 
>             Console.WriteLine("Generated Mesh:");
>             foreach (var triangle in mesh)
>             {
>                 Console.WriteLine(triangle);
>             }
>         }
>     }
> }
> 2.The scenario you're describing involves converting volumetric data  into a mesh representation. The process requires reading the VDB volumetric data and using algorithms to generate a mesh representation, such as through isosurface extraction . This is feasible using libraries like OpenVDB and MeshLib (for mesh generation), but as you mentioned, C# bindings for these libraries are currently limited.
> 
> 
>  

 _Originally posted by @yuvashrikarunakaran in [#3739](https://github.com/MeshInspector/MeshLib/issues/3739#issuecomment-2511656714)_

yuvashrikarunakaran, 2024-12-02:
If anybody has an idea about this we can collaborate and create a project !!

Grantim, 2024-12-02:
Hello!

We have point cloud triangulation algorithm:
https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRPointCloudTriangulation.h
And marching cubes algorithm for creating mesh from voxels (vdb too):
https://github.com/MeshInspector/MeshLib/blob/master/source/MRVoxels/MRMarchingCubes.h

We have plans to expose all functionality to c# too, but it will take time.

---

In this section we help people with the issues they face using meshlib, not sure that this issue is relevant, so closing it. Please reopen if you have question or suggestion about meshlib.