MeshInspector/MeshLib #3799: GUI Issues on Mac (Apple Silicon)

Doomerdinger, 2024-12-03:
I have a c++ application that renders the GUI and adds various meshes to it.
On a linux (Ubuntu22 x86_64) machine this works fine.

When I try to run the same code on a Mac (Apple Silicon) I get all of the info print outs in the console like I would expect. but I see the following error `*** Assertion failure in -[NSMenu _setMenuName:], NSMenu.m:777` and the GUI itself does not render.
The rest of the code works fine.

oitel, 2024-12-04:
Hello, do you run the GUI from a non-main thread in your app? It's a known limitation on macOS that requires the GUI rendering to be performed in the main thread only.

Doomerdinger, 2024-12-04:
That was it 🤦‍♀️

Thanks!

closed 2024-12-04 by Doomerdinger