MeshInspector/MeshLib #3810: NuGet Package Error

ccheng-design, 2024-12-06:
[image]
[image]

astrowander, 2024-12-06:
Hi
Could you provide more details about getting this error? What are the steps to reproduce?

ccheng-design, 2024-12-06:
1. Installing Nuget package from Visual Studios
2. Error code provided tells me to contact owner for package. 
3. Just followed given documentation source provided by you on simple installation and getting some sort of cycle error.

ccheng-design, 2024-12-06:
[image]

astrowander, 2024-12-06:
As I can see your project's name is MeshLib. This is the same name as our library and it may cause a name conflict. Please try to rename your project

ccheng-design, 2024-12-06:
@astrowander ended up doing that and worked fine. Thanks!

closed 2024-12-06 by ccheng-design