MeshInspector/MeshLib #3832: Debugging in another project

Doomerdinger, 2024-12-09:
I am encountering an issue where `MR::dividePolylineWithPlane` causes an infinite loop on an ARM based Mac (works fine on x86 linux and arm on linux - can't confirm x86 Mac).

I want to debug the code, but I struggle to add MeshLib as a sub-project of my project to enable the debugging of meshlib code.
I encounter many CMake issues. I tamped down the first few, but before I continue down this path I was wondering if this is a supported workflow and I am missing something easy to make it work.

oitel, 2024-12-10:
Hello, right now MeshLib doesn't support including as a subdirectory. You might copy-paste the `CMakeLists.txt` file contents to your config and edit paths to MeshLib directories. We'll create a minimal config file example for your case and notify you later.

Fedr, 2026-09-23:
MeshLib can now be used as a CMake subdirectory: support was added in #3960, and #6916 made plain `add_subdirectory(MeshLib)` work without any extra `CMAKE_MODULE_PATH` setup in the parent project.

Also, `MRPolylineTrimWithPlane.cpp`, which contains `dividePolylineWithPlane`, was heavily reworked in #4176 and #4207. If you still see the infinite loop on ARM/macOS with the current version, please reopen this issue and attach the polyline and plane that trigger it.

closed 2026-09-23 by Fedr