MeshInspector/MeshLib #3835: Reduce mesh to make milling result consistent

sysuyl, 2024-12-10:
Hello,

I have a question about milling. The problem is that when milling an object, if there are sharp concave areas on the mesh, the tool may not be able to reach these locations, resulting in the inability to cut out the same shape as the mesh. The problem that needs to be solved here is how to reduce the mesh region so that the final cut shape is consistent. In addition, the result mesh is best kept unchanged in the originally reachable areas, and the area that needs to be reduced is minimized as much as possible.

I briefly made a few pictures to illustrate the problem in the 2D case, which may be mainly based on my initial understanding of the problem.

First, the black shape is the target shape.
[image]

The green circle represents the 2D shape of the tool. The red area here indicates that the tool may not be able to cut here, resulting in the inability to cut out the target shape.
[image]

In order to make the designed shape be accurately cut out, it may be necessary to subtract some blue areas.
[image]

The final shape is as shown below. The blue edge indicates the edge area that changes after the shape is updated.
[image]

In addition, there is an effect corresponding to a 3D partial cube. Here the mesh can be regarded as the inner edge part of the 2D example above. The white area shows that the vertices of this part have basically not changed, and the blue area indicates that the vertices here have extended outwards.[image]
[image]

Any ideas or algorithms to process this kind of problem? Or do you have plans to write a general method to handle it?
Thanks!

Grantim, 2024-12-10:
Hello!

It is very interesting question indeed!
- there is a way to find the places that cannot be milled:
 1.  Do expand/shrink with radius of tool [image]
 2. It will result with such mesh: [image]
 3. Then boolean subtract original mesh from new one (slightly offset to avoid zero volumed areas): [image]
 4. Result in: [image]
 
- next we need to determine how to mill these parts out minimizing intersections with original part, and this is not trivial task

Unfortunately now we don't have single function for it. 
You can try to find tool position that fully covers each small exceeding part and then subtract it from original mesh and it will result in the mesh you want. But again this step is dot well defined.

sysuyl, 2024-12-11:
Yes, it is possible to determine whether the mesh is cut everywhere and the main problem lies in the process. Thank you, please let me know you have new ideas or methods in the future.