MeshInspector/MeshLib #3985: Discrepancy in mrmeshpy.boolean Difference Operation

benjaminchristie, 2025-01-17:
`mrmeshpy.boolean` is inconsistent depending on the order of parameters:

```python
import meshlib.mrmeshpy as meshpy
import meshlib.mrmeshnumpy as meshnp

sphere1 = meshpy.makeUVSphere(1.0, 16, 16)
sphere2 = meshpy.makeUVSphere(0.8, 16, 16)
xf = meshpy.AffineXf3f.translation(meshpy.Vector3f(0.0, 0.0, 0.2))
sphere2.transform(xf)

d_ba = meshpy.boolean(sphere1, sphere2, meshpy.BooleanOperation.DifferenceBA)
d_ab = meshpy.boolean(sphere2, sphere1, meshpy.BooleanOperation.DifferenceAB)

v_ba = meshnp.getNumpyVerts(d_ba.mesh)
v_ab = meshnp.getNumpyVerts(d_ab.mesh)

print(len(v_ba))
print(len(v_ab))
```

The code snippet prints 
```
0
34
```

Is this intended behavior? If so, what is the difference between the boolean subtraction operations aside from syntax?

Grantim, 2025-01-17:
Hello!

Looks like this configuration is degenerated:
[image]

You see that two spheres have coincident zone, and volume of the result is zero:

[image] [image]

This difference in results is expected and it happens because of the way we solve degenerations, it depends on order of the meshes.

P.S.
I got this result:
```
96
66
```

benjaminchristie, 2025-01-17:
Thanks for your help! I'll close for now and reopen if I have other questions.

closed 2025-01-17 by benjaminchristie