MeshInspector/MeshLib #4032: isosurface extract

1358361259, 2025-01-29:
can meshlib extract isosurface from volume data by marching cubes？

Grantim, 2025-01-29:
Hello!

Sure meshlib can do it:
https://github.com/MeshInspector/MeshLib/blob/master/source/MRVoxels/MRMarchingCubes.h

1358361259, 2025-01-29:
wow thanks！are there some python examples？

Grantim, 2025-01-29:
You can use [this](https://meshlib.io/documentation/Py/namespacemeshlib_1_1mrmeshnumpy.html#a55d5b941bd43dea83c19e37eee0039f8) function to create volume in meshlib, and then call [this](https://meshlib.io/documentation/Py/namespacemeshlib_1_1mrmeshpy.html#a80e3154cd5d18c1fbb1f9ec9ee9d09c4) to extract iso surface. Also please have a look at [loadVoxels](https://meshlib.io/documentation/Py/namespacemeshlib_1_1mrmeshpy.html#a99f3cec749da53f047aef069aaf65561) functions set, that might be useful to load volume data directly from files.

Unfortunately we now don't have clear example for it, but will add it in the future.

1358361259, 2025-01-29:
I will try it thanks！