MeshInspector/MeshLib #4050: Surface remeshing

leonard-gleyzer, 2025-02-02:
Hello,

pygalmesh offers [surface remeshing](https://github.com/meshpro/pygalmesh?tab=readme-ov-file#surface-remeshing), keeping triangles fairly uniform, with refinement near feature edges to stay accurate to the original geometry:

[image]

How can I achieve a similar result in meshlib/mrmeshpy?

Thank you

Fedr, 2025-02-02:
Thanks for your question.

Actually, on above pictures the size of tringles differ very much, and they hardly can be named "uniform".

This is what is achievable in MeshInspector/MeshLib right now:
[image]

From left to right:
1) original mesh,
2) remesh result with default settings in MeshInspector UI,
3) remesh result with additional 3 final relax iterations, making triangles almost uniform.

In API please look for this function:
```
bool remesh( Mesh& mesh, const RemeshSettings & settings );
```

There are many parameters in `RemeshSettings`. Probably you would like to try 
```
    /// Edges specified by this bit-set will never be flipped or collapsed, but they can be replaced during collapse of nearby edges so it is updated during the operation;
    /// also the vertices incident to these edges are excluded from relaxation
    UndirectedEdgeBitSet* notFlippable = nullptr;
```
for preserving some features in original mesh.

leonard-gleyzer, 2025-02-02:
I see, thank you. Uniform wasn't the right description. I meant more locally uniform/even within a particular feature/patch, but in particular I was looking for preserving the original shape with refinement near feature edges. Thank you!

closed 2025-02-02 by leonard-gleyzer

Fedr, 2025-02-03:
Yes, you can first find feature edges, e.g. by dihedral angle threshold:
[image]

And pass them as `notFlippable` setting in `remesh` function. Then they will be preserved as much as possible in the result:
[image]