MeshInspector/MeshLib #4091: Drill compensation shut down the project

EunryongLee, 2025-02-12:
Hello
I want to use radius compensation. However, using radius compensation causes the project to be terminated for unknown reasons. I'm inquiring because it's an error that didn't occur in the previous version of radius compensation. I current use meshlib with export DLL in c++.

example:
```cpp
MR::Mesh mesh = MR::MeshLoad::fromAnyStl(inFilePath + ".stl").value();
MR::MeshSave::toAnySupportedFormat(mesh, inFilePath + "-" + std::to_string(dRadius) + ".stl");

MR::VertCoords vertCoords = mesh.points;

MR::Vector3f bdCenter;
size_t size = 0;
MR::ParallelFor(vertCoords.beginId(), vertCoords.endId(), [mesh, &bdCenter, vertCoords, &size](MR::VertId v) {
	if (mesh.topology.isBdVertex(v))
	{
		bdCenter += vertCoords[v];
		size++;
	}
});
if (size != 0)
	bdCenter = {bdCenter.x / size, bdCenter.y / size, bdCenter.z / size };

MR::Vector3f target = mesh.getBoundingBox().center();
MR::Mesh Arrow = MR::makeArrow(bdCenter, target);
MR::MeshSave::toAnySupportedFormat(Arrow, inFilePath + "-arrow.stl");

MR::Vector3f dir = target - bdCenter;
dir = dir.normalized();

MR::CompensateRadiusParams params;
{
	params.distanceMapResolution = MR::Vector2i(100, 100);
	params.direction = dir;
	params.toolRadius = dRadius;
}
MR::compensateRadius(mesh, params);
MR::MeshSave::toAnySupportedFormat(mesh, outFilePath);
```

Grantim, 2025-02-12:
Hello!

Could you please provide more details about the error that occurs?

EunryongLee, 2025-02-12:
At first, I put in random directions without using parallele_for. I don't think that's the problem.
I called two cases in a row, and when I used debug mode in the first case, the code went all the way, but it didn't finish when I ran it in release and the program ended after 5 seconds. Even if I went to debug mode, the program ended in the second case. When I used debug mode, error code c0000374 occurred, and mesh destructor encountered a heap memory error. But I'm not sure if it's a debug mode related error or a meshlib error.
If there is a lack of explanation, I will also analyze it further and inform you of the error if an error occurs.

Grantim, 2025-02-12:
Looks like this error is due to unsafe writing in `ParallelFor` could you please verify it?

EunryongLee, 2025-02-12:
The same happens with this code. I'll let you know if I get a more accurate error. Thank you for your help.

```cpp
MR::Vector3f bdCenter;
size_t size = 0;
for (MR::VertId v = vertCoords.beginId(); v != vertCoords.endId(); v++) {
	if (mesh.topology.isBdVertex(v))
	{
		bdCenter += vertCoords[v];
		size++;
	}
};
if (size != 0)
	bdCenter = {bdCenter.x / size, bdCenter.y / size, bdCenter.z / size };
```

Grantim, 2025-02-12:
Could you please provide files and code minimal code example so we could reproduce it on our side, (if you don't want to share your data publicly you can send it to our support email: `support@meshinspector.com`)

EunryongLee, 2025-02-14:
It wasn't a problem with meshlib, but an issue with my import settings. Thank you for your help.

closed 2025-02-14 by EunryongLee