MeshInspector/MeshLib #4119: Voxel binary operations

mariuszhermansdorfer, 2025-02-15:
Hey @Fedr,

I just saw that this PR has been merged: https://github.com/MeshInspector/MeshLib/commit/d705e227d02543474d7a5b362894077dd3121b8a

Does it allow for SDF-style modelling similar to this one? Specifically the smooth blending options between primitives.
https://www.youtube.com/watch?v=2EPTambcvyM

Fedr, 2025-02-15:
Hello @mariuszhermansdorfer ,

This merged PR just adds UI command that performs some binary operation on voxel objects. Indeed, it is mostly useful if the voxels are filled with signed distances (SDF). So it takes two voxel objects and produces the third voxel object as a result. Actually it was present in MeshInspector under "Voxel Operations" name on Test tab. We just made a few improvements in it and moved in MeshLib. The command mostly adds UI to the functions from OpenVDB library, which performs actual processing.

At this moment the tool is not interactive, so it just produces the result for the current positions of input objects and closes. But technically it shall work rather fast for small voxel volumes and one can try to show its result immediately after moving one of input objects. Thanks for the idea and for the video.

mariuszhermansdorfer, 2025-02-15:
I’d love to see more SDF functionality in MeshLib. OpenVDB is a great foundation but a lot of value could be gained from making it more interactive. Especially the blending aspect, where multiple bodies blend with each other based on proximity and don’t necessarily have to touch each other.

One of the dev branches of the OpenVDB repo introduced fVDB. It has plenty of ML-related functionality, but an interesting aspect for  MeshLib could be CUDA-based mesh to grid conversions. 

From an end-user perspective, I’m very interested in the following workflow:
1. Import custom meshes 
2. Convert them to SDF/grid representations 
3. Perform various blending operations 
4. Convert the result back to mesh while preserving initial detail

Now, imagine that this whole sequence can be done interactively with hundreds of binary operations. This opens up a very interesting CSG workflow which very few geometry kernels can provide.

Fedr, 2025-02-16:
Yes, it is an interesting direction for further MeshLib development. Yet it requires significant efforts, so we need to prioritize it against other ideas first.

> One of the dev branches of the OpenVDB repo introduced fVDB. It has plenty of ML-related functionality, but an interesting aspect for MeshLib could be CUDA-based mesh to grid conversions.

Yes, we currently support both OpenVDB and CUDA in MeshLib, but they are hardly compatible in between. OpenVDB presents memory-efficient tree-based storage for voxels close to input meshes, which is hard to process in CUDA. So all our CUDA algorithms work with not-OpenVDB voxel representations.

> Convert the result back to mesh while preserving initial detail

As with all voxel-based algorithms, all details smaller than one voxel will be inevitably lost. For most detailed mode not-voxel approaches will be still necessary.

mariuszhermansdorfer, 2025-02-16:
> Yes, we currently support both OpenVDB and CUDA in MeshLib, but they are hardly compatible in between. OpenVDB presents memory-efficient tree-based storage for voxels close to input meshes, which is hard to process in CUDA. So all our CUDA algorithms work with not-OpenVDB voxel representations.

fVDB introduces a few methods to at least create the grids with CUDA. Have a look here:
https://github.com/AcademySoftwareFoundation/openvdb/blob/06f003e85acd353644cbd80497394f788ca9e234/fvdb/src/detail/build/FromMesh.cpp#L91-L104

> Yes, it is an interesting direction for further MeshLib development. Yet it requires significant efforts, so we need to prioritize it against other ideas first.

Fully understand, but one can always dream :)

mariuszhermansdorfer, 2025-02-19:
Here is an interesting paper about a general method of creating SDFs even from broken geometry:

https://nzfeng.github.io/research/SignedHeatMethod/index.html
https://github.com/nzfeng/signed-heat-3d
https://x.com/nicolefeng_/status/1892052847287533915