MeshInspector/MeshLib #4127: some questions about boolean operations in MeshLib

DWR-CMQ, 2025-02-16:
I am very happy to learn about Meshlib research,and have been looking at boolean operations.I have several questions
the function-findCollidingEdgeTrisPrecise in MRMeshCollidePrecise.cpp
[image]
What is the logic in this function based on? Is it from any paper?
for example,
1: Where does the variable 16 in the green box come from?Is it from an algorithm?
2: What do these three variables(subtasks, nextSubtasks, leafTasks) represent?

Fedr, 2025-02-17:
This function must find all intersecting pairs consisting of
1. An edge from one mesh and
2. a triangle from another mesh.

Both meshes are represented by their own bounding volume hierarchy (tree).

`subtasks`, `nextSubtasks`, `leafTasks` - are all a pair of nodes: the first node is from first mesh's tree and the second node is from second mesh's tree.

Initially we start from the roots of both trees: `{ NodeId{ 0 }, NodeId{ 0 } }`. Then we descend in the trees by replacing a node with its two child nodes. If pair's nodes do not intersect, such pair is discarded.

Initially descend is done sequentially until the level 16. This value was found just experimentally, an some other value can be better for specific scenarios. And after that the descend is performed in parallel starting from a number of pairs in `subtasks`.

`nextSubtasks` is just a temporary storage for the results of `subtasks` splitting on the current iteration.

`leafTasks` is a pair of terminal or leaf nodes in the trees, each of which contain just one triangles. These nodes cannot be split any more.

Please let us know if you have more questions.

Fedr, 2026-09-24:
Small update: since f29d4de8b (June 2025) the sequential phase runs 14 iterations instead of 16; the timings behind that choice are in the comment at `MRMeshCollidePrecise.cpp:50-53`. Each iteration splits one node of every intersecting pair (the one with the larger box), so it produces at most 2^14 subtasks for the parallel phase. The overall scheme isn't from a specific paper: it's the standard simultaneous traversal of two bounding volume hierarchies, as used in OBBTree/RAPID-style collision detection. Closing as answered; feel free to reopen if you have more questions.

closed 2026-09-24 by Fedr