MeshInspector/MeshLib #4142: Infinite loop when calling dividePolylineWithPlane

ammarhattab, 2025-02-18:
Hello, 

For some reason, my code get stuck when calling dividePolylineWithPlane in C++ for some cases on ARM machines (MacOS). 

Here's the sample code that I am using:

```
    auto plane = MR::Plane3f::fromDirAndPt(direction, point);
    MR::dividePolylineWithPlane(polyline, plane);
```

Here's the input data for your reference:

```
direction: [0.968297, -0.000418756, -0.249801]
point: [-20.6947, 5.16071, -14.7379]
polyline points: [-25.3436, 6.0357, -17.5618 , -25.3212, 6.04543, -17.6112 , -25.2984, 6.05643, -17.6607 , -25.2729, 6.06959, -17.7142 , -25.2432, 6.08515, -17.7745 , -25.2098, 6.10222, -17.8408 , -25.1747, 6.1191, -17.9089 , -25.1393, 6.13461, -17.9758 , -25.1031, 6.14891, -18.0422 , -25.0651, 6.16291, -18.1092 , -25.026, 6.17719, -18.175 , -24.987, 6.19195, -18.2372 , -24.9483, 6.20723, -18.2952 , -24.9101, 6.22295, -18.3495 , -24.8728, 6.23894, -18.3992 , -24.8362, 6.25534, -18.4448 , -24.7983, 6.27264, -18.4889 , -24.7575, 6.29079, -18.5344 , -24.7139, 6.30865, -18.5819 , -24.6687, 6.32483, -18.6306 , -24.6224, 6.33902, -18.6794 , -24.5748, 6.35204, -18.7278 , -24.5258, 6.36471, -18.7755 , -24.4756, 6.37705, -18.8223 , -24.4259, 6.38841, -18.8675 , -24.3782, 6.39826, -18.9102 , -24.3324, 6.407, -18.9503 , -24.2858, 6.41584, -18.9894 , -24.2357, 6.4257, -19.0291 , -24.1813, 6.43637, -19.0694 , -24.1228, 6.4468, -19.11 , -24.0584, 6.45595, -19.1511 , -23.9869, 6.4631, -19.1922 , -23.9106, 6.46746, -19.2305 , -23.8348, 6.46801, -19.2628 , -23.7634, 6.46395, -19.2886 , -23.6956, 6.45545, -19.3096 , -23.6288, 6.44381, -19.3274 , -23.5606, 6.43018, -19.343 , -23.4892, 6.41453, -19.3566 , -23.4155, 6.39641, -19.3689 , -23.3439, 6.37628, -19.3797 , -23.2789, 6.35525, -19.3888 , -23.2205, 6.33356, -19.3963 , -23.1644, 6.31018, -19.4026 , -23.1069, 6.28422, -19.4083 , -23.0485, 6.25658, -19.4133 , -22.9919, 6.22934, -19.4174 , -22.9376, 6.20369, -19.4204 , -22.883, 6.17876, -19.4226 , -22.8254, 6.15293, -19.4248 , -22.7646, 6.12565, -19.4271 , -22.7024, 6.09755, -19.4295 , -22.6401, 6.06954, -19.4315 , -22.579, 6.04246, -19.4329 , -22.5213, 6.01697, -19.4337 , -22.468, 5.99298, -19.4343 , -22.4182, 5.9694, -19.4355 , -22.3689, 5.9446, -19.438 , -22.3175, 5.91709, -19.4425 , -22.2625, 5.88636, -19.4487 , -22.2042, 5.85302, -19.4561 , -22.1427, 5.81771, -19.4642 , -22.076, 5.78, -19.4726 , -22.0015, 5.73943, -19.4814 , -21.9203, 5.69744, -19.4897 , -21.8379, 5.65736, -19.4967 , -21.76, 5.62194, -19.502 , -21.6887, 5.59133, -19.5058 , -21.6219, 5.56352, -19.5089 , -21.5571, 5.5364, -19.5121 , -21.493, 5.509, -19.5154 , -21.4292, 5.48126, -19.5189 , -21.365, 5.45348, -19.5222 , -21.2994, 5.42565, -19.5249 , -21.231, 5.3973, -19.5273 , -21.1594, 5.3682, -19.5297 , -21.0874, 5.33897, -19.5322 , -21.0179, 5.31006, -19.5351 , -20.9519, 5.28095, -19.5381 , -20.8901, 5.25115, -19.5412 , -20.833, 5.22097, -19.5443 , -20.7801, 5.19051, -19.5478 , -20.7282, 5.15889, -19.5522 , -20.6744, 5.12557, -19.5572 , -20.6192, 5.09181, -19.5625 , -20.5651, 5.05975, -19.5674 , -20.5127, 5.03014, -19.5721 , -20.4596, 5.00166, -19.577 , -20.4037, 4.97291, -19.582 , -20.3457, 4.94419, -19.5867 , -20.2888, 4.91664, -19.5909 , -20.2337, 4.8899, -19.5948 , -20.1769, 4.86129, -19.5991 , -20.1129, 4.82791, -19.6043 , -20.0387, 4.78958, -19.6103 , -19.9565, 4.74945, -19.6165 , -19.8731, 4.712, -19.6223 , -19.7959, 4.68034, -19.6275 , -19.729, 4.65517, -19.6318 , -19.6711, 4.63539, -19.6354 , -19.6177, 4.61925, -19.6385 , -19.5644, 4.60526, -19.6416 , -19.5074, 4.59218, -19.6446 , -19.4454, 4.57894, -19.6478 , -19.3792, 4.56464, -19.6509 , -19.31, 4.54857, -19.6537 , -19.2374, 4.5303, -19.6562 , -19.1618, 4.51028, -19.6583 , -19.086, 4.48996, -19.6595 , -19.0117, 4.47081, -19.6594 , -18.9374, 4.45324, -19.6576 , -18.8618, 4.4368, -19.6541 , -18.7881, 4.42105, -19.6497 , -18.7211, 4.4058, -19.6451 , -18.6623, 4.39081, -19.6409 , -18.6091, 4.37591, -19.6366 , -18.5583, 4.3613, -19.6319 , -18.5079, 4.34747, -19.6266 , -18.4573, 4.33481, -19.6207 , -18.4062, 4.3234, -19.6144 , -18.355, 4.31299, -19.6077 , -18.3033, 4.30298, -19.6004 , -18.25, 4.29248, -19.5927 , -18.1932, 4.28072, -19.5842 , -18.1321, 4.26791, -19.575 , -18.0688, 4.25553, -19.5653 , -18.0072, 4.24507, -19.5558 , -17.9496, 4.23651, -19.5467 , -17.8951, 4.22832, -19.5375 , -17.841, 4.21879, -19.5277 , -17.7846, 4.20689, -19.5164 , -17.7247, 4.19223, -19.5035 , -17.6619, 4.17463, -19.4886 , -17.5971, 4.15343, -19.4715 , -17.5295, 4.12749, -19.451 , -17.4584, 4.09659, -19.4263 , -17.3864, 4.06331, -19.3982 , -17.3189, 4.03249, -19.3695 , -17.2592, 4.00824, -19.3429 , -17.2065, 3.99163, -19.3195 , -17.1574, 3.98127, -19.2982 , -17.1092, 3.97505, -19.2776 , -17.0611, 3.97098, -19.2561 , -17.0134, 3.96715, -19.2328 , -16.9663, 3.96178, -19.2069 , -16.9195, 3.95355, -19.178 , -16.8722, 3.9423, -19.1453 , -16.8237, 3.92937, -19.1079 , -16.775, 3.91696, -19.066 , -16.7274, 3.9065, -19.0202 , -16.6816, 3.89777, -18.9718 , -16.638, 3.88997, -18.9224 , -16.5971, 3.8831, -18.8747 , -16.5596, 3.87766, -18.8298 , -16.5247, 3.87365, -18.7872 , -16.4913, 3.87034, -18.7458 , -16.4583, 3.86688, -18.704 , -16.4246, 3.86274, -18.6601 , -16.388, 3.85761, -18.6111 , -16.3461, 3.85123, -18.5535 , -16.2982, 3.84386, -18.4859 , -16.2465, 3.83693, -18.4116 , -16.1951, 3.83286, -18.3374 , -16.148, 3.83359, -18.2701 , -16.1068, 3.83917, -18.2125 , -16.0699, 3.84801, -18.162 , -16.0342, 3.85855, -18.113 , -15.9974, 3.87004, -18.0619 , -15.9599, 3.88191, -18.0093 , -15.9235, 3.89337, -17.9584 , -15.8899, 3.90392, -17.911 , -15.8584, 3.9139, -17.8668 , -15.8281, 3.92404, -17.8244 , -15.7979, 3.93479, -17.7827 , -15.7675, 3.94612, -17.7412 , -15.7366, 3.95787, -17.6991 , -15.7047, 3.97024, -17.6557 , -15.6709, 3.98372, -17.6096 , -15.6344, 3.99842, -17.5602 , -15.5954, 4.01359, -17.5082 , -15.5539, 4.02823, -17.4538 , -15.5093, 4.04218, -17.3965 , -15.4621, 4.05613, -17.3365 , -15.415, 4.07079, -17.2764 , -15.3707, 4.08644, -17.2186 , -15.3295, 4.1028, -17.163 , -15.2906, 4.11908, -17.108 , -15.2527, 4.1346, -17.0519 , -15.2153, 4.14952, -16.9928 , -15.178, 4.16441, -16.9297 , -15.1417, 4.17944, -16.8632 , -15.1067, 4.19424, -16.7947 , -15.0725, 4.20838, -16.7241 , -15.0387, 4.22121, -16.6524 , -15.0061, 4.23199, -16.5834 , -14.9768, 4.24047, -16.521 , -14.9514, 4.2473, -16.4655 , -14.929, 4.25359, -16.4138 , -14.9076, 4.26022, -16.362 , -14.8855, 4.26752, -16.3073 , -14.8616, 4.27516, -16.2485 , -14.8363, 4.28235, -16.1864 , -14.8106, 4.28852, -16.1225 , -14.7855, 4.29427, -16.057 , -14.7615, 4.30142, -15.9894 , -14.7385, 4.31186, -15.9191 , -14.7161, 4.32618, -15.8457 , -14.6941, 4.34321, -15.7699 , -14.673, 4.36066, -15.694 , -14.6529, 4.37646, -15.6198 , -14.6335, 4.38958, -15.545 , -14.6133, 4.39957, -15.4649 , -14.5922, 4.40579, -15.3772 , -14.5712, 4.40751, -15.2848 , -14.5519, 4.40486, -15.1944 , -14.536, 4.39923, -15.1126 , -14.5238, 4.39235, -15.0427 , -14.5147, 4.38515, -14.9828 , -14.5074, 4.3775, -14.9282 , -14.5008, 4.36864, -14.874 , -14.4946, 4.35767, -14.8152 , -14.4888, 4.34408, -14.7489 , -14.4838, 4.32853, -14.6762 , -14.4803, 4.31284, -14.603 , -14.4787, 4.29891, -14.535 , -14.4793, 4.28749, -14.4723 , -14.4827, 4.27826, -14.4092 , -14.4893, 4.27055, -14.3402 , -14.4989, 4.26389, -14.265 , -14.5106, 4.25796, -14.1884 , -14.5236, 4.25236, -14.1149 , -14.5377, 4.24658, -14.0451 , -14.5528, 4.2401, -13.9772 , -14.5691, 4.23248, -13.9093 , -14.5869, 4.22326, -13.8397 , -14.6062, 4.21221, -13.7675 , -14.6269, 4.19956, -13.6939 , -14.649, 4.18571, -13.6204 , -14.6728, 4.17093, -13.5466 , -14.6983, 4.15559, -13.4727 , -14.7236, 4.14063, -13.4018 , -14.7471, 4.12717, -13.338 , -14.7686, 4.11567, -13.2818 , -14.7895, 4.10556, -13.2303 , -14.8123, 4.09579, -13.1796 , -14.8392, 4.0855, -13.1254 , -14.8709, 4.07448, -13.0648 , -14.9057, 4.06305, -12.9986 , -14.9402, 4.05154, -12.9324 , -14.9717, 4.04008, -12.872 , -15.0002, 4.02861, -12.8194 , -15.0277, 4.01731, -12.7718 , -15.0568, 4.00655, -12.725 , -15.0904, 3.99653, -12.6743 , -15.1312, 3.98687, -12.6161 , -15.18, 3.97679, -12.5495 , -15.235, 3.96586, -12.4776 , -15.2913, 3.95447, -12.4069 , -15.344, 3.94346, -12.3435 , -15.3907, 3.93321, -12.2896 , -15.4325, 3.92299, -12.243 , -15.4729, 3.91146, -12.1985 , -15.5144, 3.89792, -12.1527 , -15.5568, 3.88313, -12.1057 , -15.5982, 3.86878, -12.0606 , -15.6377, 3.8562, -12.0192 , -15.6763, 3.84572, -11.9814 , -15.7153, 3.83693, -11.946 , -15.7557, 3.82914, -11.9118 , -15.7985, 3.8214, -11.877 , -15.8452, 3.81246, -11.8389 , -15 [... cut ...] 9, 5.83613, -16.5634 , -25.7203, 5.85143, -16.6358 , -25.6973, 5.86494, -16.7093 , -25.6737, 5.87641, -16.7809 , -25.651, 5.88636, -16.8456 , -25.6297, 5.89596, -16.902 , -25.6093, 5.90619, -16.9531 , -25.5884, 5.91746, -17.0032 , -25.5659, 5.93, -17.0559 , -25.5416, 5.94383, -17.1124 , -25.5162, 5.95833, -17.1708 , -25.4913, 5.97225, -17.2276 , -25.4674, 5.98472, -17.282 , -25.4434, 5.99586, -17.3366 , -25.4182, 6.00636, -17.3938 , -25.3923, 6.01654, -17.4525 , -25.3671, 6.02628, -17.5092 ]
```

oitel, 2025-02-18:
Hello, we cannot reproduce the issue using the given data (running on a Mac M1 machine). This might be a result of precision loss after printing the values. Could you send us the data in a binary format?

ammarhattab, 2025-02-18:
I am not sure how to send you the input data in a binary format!
Note: the error happens when running this code as part of a native build on Apple M1 Max (MacOS 15.3), or by running it in an ubuntu 22:04 based docker image.

Here's the same data with higher precision output: 
```
direction: [0.9682972431, -0.0004187558079, -0.249800697]
point: [-20.69468689, 5.160713673, -14.73788357]
polyline points: [-25.34358788, 6.035698414, -17.56179237 , -25.32123947, 6.045425415, -17.61123848 , -25.29838943, 6.056428432, -17.66069984 , -25.27289581, 6.069586754, -17.71415329 , -25.24321938, 6.085149288, -17.77447891 , -25.20977783, 6.102220058, -17.84081078 , -25.17465973, 6.119100094, -17.90891647 , -25.13929176, 6.134608746, -17.97580338 , -25.10307693, 6.148914337, -18.04220772 , -25.06512642, 6.1629076, -18.10920906 , -25.02604866, 6.177194118, -18.1750145 , -24.98698997, 6.191949844, -18.23716927 , -24.94829559, 6.207227707, -18.29524994 , -24.91007996, 6.222952366, -18.34947395 , -24.87283134, 6.238943577, -18.39922905 , -24.8361969, 6.255344391, -18.44482231 , -24.79831886, 6.272641182, -18.48889732 , -24.75748825, 6.290793896, -18.53435707 , -24.71386147, 6.308652401, -18.58191681 , -24.66865158, 6.324826717, -18.63057327 , -24.62239838, 6.339017868, -18.67935371 , -24.57484245, 6.352036953, -18.72776413 , -24.52577782, 6.36470747, -18.77548599 , -24.47564507, 6.377051353, -18.82228088 , -24.42588234, 6.388409138, -18.86750221 , -24.37819672, 6.398255348, -18.91015244 , -24.33240509, 6.406998158, -18.95025253 , -24.28580856, 6.415841103, -18.98939133 , -24.23565865, 6.425700665, -19.02908516 , -24.18131447, 6.436367989, -19.06942749 , -24.12277222, 6.446799278, -19.11004257 , -24.05844688, 6.455954075, -19.15114594 , -23.9869194, 6.463097095, -19.19220924 , -23.91055679, 6.467456818, -19.23049736 , -23.83483315, 6.46801281, -19.26284981 , -23.76337433, 6.463946342, -19.28861809 , -23.69556808, 6.455453396, -19.30956268 , -23.62878609, 6.443813324, -19.32743454 , -23.56056786, 6.430179119, -19.34296036 , -23.48919296, 6.414530754, -19.35661507 , -23.41549683, 6.396408558, -19.36887932 , -23.34389687, 6.376275063, -19.37973404 , -23.27893066, 6.35524559, -19.38883591 , -23.22053528, 6.333563805, -19.39625168 , -23.16440392, 6.310176849, -19.40255737 , -23.10691643, 6.284222603, -19.40825844 , -23.04851913, 6.256577015, -19.41333008 , -22.99188995, 6.22934103, -19.41740608 , -22.93761635, 6.203688622, -19.42037773 , -22.8830204, 6.178759098, -19.42263603 , -22.8253727, 6.152933598, -19.42477226 , -22.76463699, 6.125651836, -19.42711258 , -22.70243263, 6.097551346, -19.42949295 , -22.64008713, 6.069544315, -19.43150902 , -22.57901192, 6.042463303, -19.43287659 , -22.52128029, 6.016967773, -19.43366432 , -22.46804047, 5.992980003, -19.43430138 , -22.41816711, 5.969403744, -19.43548775 , -22.36892319, 5.944595814, -19.43802834 , -22.31751823, 5.917093754, -19.44246292 , -22.26249886, 5.886363983, -19.44870758 , -22.20419502, 5.853018761, -19.45613289 , -22.14272881, 5.817706585, -19.46416092 , -22.07600784, 5.780004025, -19.47262383 , -22.00151443, 5.739431858, -19.4813652 , -21.92032814, 5.697444439, -19.48970985 , -21.83787155, 5.657361984, -19.49672508 , -21.75999451, 5.621940613, -19.50196266 , -21.68865204, 5.591331005, -19.50576973 , -21.6218586, 5.563521862, -19.50893402 , -21.55705452, 5.536403179, -19.51207733 , -21.49299812, 5.508998394, -19.51544189 , -21.42916298, 5.481262684, -19.51890755 , -21.36499214, 5.453477383, -19.52215767 , -21.29943466, 5.425652027, -19.52494621 , -21.23095703, 5.397296429, -19.52733803 , -21.15942001, 5.368197441, -19.52966309 , -21.08740044, 5.338967323, -19.53223991 , -21.01787376, 5.31005621, -19.5351162 , -20.95194626, 5.280951023, -19.53814125 , -20.89005089, 5.2511549, -19.54116821 , -20.83300781, 5.220972061, -19.54428864 , -20.78012276, 5.190506458, -19.54784012 , -20.72821236, 5.158891201, -19.55216599 , -20.6744194, 5.125573635, -19.55721092 , -20.61919785, 5.091805458, -19.56246376 , -20.56506538, 5.059751511, -19.56742668 , -20.51268578, 5.0301404, -19.5721283 , -20.45964813, 5.00165987, -19.5769558 , -20.40365791, 4.972912312, -19.58197403 , -20.34570122, 4.944189548, -19.58672523 , -20.28878403, 4.916641235, -19.590868 , -20.23367691, 4.889898777, -19.59475327 , -20.17685509, 4.861287117, -19.59911156 , -20.11285591, 4.827913761, -19.60434914 , -20.03867149, 4.789576054, -19.61029053 , -19.9564991, 4.749450207, -19.6164608 , -19.87307358, 4.711998463, -19.62233543 , -19.79592133, 4.680337429, -19.62750816 , -19.72899055, 4.655174255, -19.63181686 , -19.67105675, 4.635387421, -19.63539314 , -19.61774635, 4.619252205, -19.63854218 , -19.56437111, 4.605255127, -19.64156723 , -19.50739098, 4.592175484, -19.64464378 , -19.4453907, 4.578937531, -19.64778519 , -19.37924004, 4.564642429, -19.65086174 , -19.31002617, 4.548567295, -19.65371895 , -19.23742867, 4.530302525, -19.65623856 , -19.16183662, 4.51028347, -19.65826607 , -19.08597565, 4.489961624, -19.65947914 , -19.0116806, 4.470812321, -19.65938187 , -18.93736649, 4.453244209, -19.65756035 , -18.8617878, 4.436804295, -19.65408516 , -18.78808212, 4.42105484, -19.6496582 , -18.72111511, 4.405799389, -19.64513588 , -18.66230202, 4.390808105, -19.64087296 , -18.60912132, 4.375913143, -19.63661194 , -18.55830956, 4.36130476, -19.63191032 , -18.50792694, 4.347467899, -19.62658119 , -18.45727348, 4.334806919, -19.62070656 , -18.40624237, 4.323400497, -19.61439896 , -18.35496712, 4.312988758, -19.60765457 , -18. [... cut ...]  -25.62971115, 5.895959377, -16.90203857 , -25.60926247, 5.90618515, -16.95311928 , -25.58839226, 5.917460918, -17.00321007 , -25.56594658, 5.929995537, -17.05589867 , -25.54156494, 5.943826675, -17.11241341 , -25.51618195, 5.958327293, -17.17075729 , -25.4913044, 5.972254753, -17.22759056 , -25.46738625, 5.984718323, -17.28203201 , -25.44336891, 5.99586153, -17.33662605 , -25.41819, 6.006362438, -17.39380836 , -25.39226723, 6.016539097, -17.4525013 , -25.36709213, 6.026284695, -17.50920486]
```

oitel, 2025-02-19:
You can export a polyline using the `LinesSave::toAnySupportedFormat` function:
```cpp
#include <MRMesh/MRLinesSave.h>

MR::LinesSave::toAnySupportedFormat( polyline, "polyline.mrlines" ); // mrlines is the MeshLib's internal format
```
But I believe the new data you provided are precise enough.

Still we couldn't reproduce the issue neither on Mac nor on Ubuntu. You can check the test results from our CI:
https://github.com/MeshInspector/MeshLib/actions/runs/13409187248/job/37455243497?pr=4146#step:11:2603

There's a chance that the issue has already been resolved actually. Could you try to run your code with the latest release of MeshLib?
https://github.com/MeshInspector/MeshLib/releases/tag/v3.0.3.233

ammarhattab, 2025-02-19:
Unfortunately, using the latest version still didn't work. In general is there an alternative this method: dividePolylineWithPlane? Can we try to do it manually?

Fedr, 2025-02-19:
`dividePolylineWithPlane` function is designed to cut input polyline on two parts: one on the positive side of the plane, and the other one on the negative side of plane. Of course, you can do the same operation by your own, but we will be grateful, if you cand help us resolving the issues with the current MeshLib implementation.

As was already stated above, we made a test from your data: https://github.com/MeshInspector/MeshLib/compare/master...fix/divide_polyline_with_plane#diff-9a0015f2fe8bbeb5daddc9146bf21a0bcd28c61a55ff642b80d5cfafb96202c6R20
but it runs successfully even on ARM machines. Do we miss something here?

Could you please attach the process callstack on the moment, when you experience infinite loop?

ammarhattab, 2025-02-20:
Thanks for the help, I will try to run your test file on my machine and see if I get the error. How do I collect the process callstack ( I am running this inside docker container)?

ammarhattab, 2025-02-20:
Here's the result of executing that:

```
Thread 1 "CtoM" received signal SIGINT, Interrupt.
0x0000fffff7c7efd4 in MR::fillPolylineLeft(MR::Polyline<MR::Vector3<float> > const&, MR::TaggedBitSet<MR::EdgeTag> const&, std::vector<std::pair<MR::Id<MR::VertTag>, MR::Id<MR::VertTag> >, std::allocator<std::pair<MR::Id<MR::VertTag>, MR::Id<MR::VertTag> > > >*) () from /usr/local/lib/MeshLib/libMRMesh.so
(gdb) bt
#0  0x0000fffff7c7efd4 in MR::fillPolylineLeft(MR::Polyline<MR::Vector3<float> > const&, MR::TaggedBitSet<MR::EdgeTag> const&, std::vector<std::pair<MR::Id<MR::VertTag>, MR::Id<MR::VertTag> >, std::allocator<std::pair<MR::Id<MR::VertTag>, MR::Id<MR::VertTag> > > >*) ()
   from /usr/local/lib/MeshLib/libMRMesh.so
#1  0x0000fffff7c7f36c in MR::dividePolylineWithPlane(MR::Polyline<MR::Vector3<float> >&, MR::Plane3<float> const&, MR::DividePolylineParameters const&)
    () from /usr/local/lib/MeshLib/libMRMesh.so
#2  0x0000fffff7f782ec in meshlib::topography::cToM (input_crown=..., insertion_center_point=..., insertion_direction=..., 
    margin_line_cloud=..., use_cut=true, num_hops=15, render_visuals_suffix=std::optional<std::string> [no contained value], 
    laplacian_edge_weights_type=MR::EdgeWeights::CotanWithAreaEqWeight, smooth=false, snap_to_margin_whitelist_vertices=0x0)
    at /app/src/margin_line.cpp:412
#3  0x0000aaaaaab5b8c4 in CtoM (crown_mesh=..., metadata1=..., 
    maybe_metadata2=std::optional<meshlib::MarginLineMetadataJson> [no contained value], margin_parameters=...)
    at /app/src/c_to_m.cpp:157
#4  0x0000aaaaaaaca5f8 in cliMain (argc=3, argv=0xfffffffff608) at /app/src/scripts/c_to_m.cpp:195
#5  0x0000aaaaaaacb344 in main (argc=3, argv=0xfffffffff608) at /app/src/scripts/c_to_m.cpp:227
```

Fedr, 2025-02-21:
Thanks. We already have pushed one fix in master branch of MeshLib that might fix your case. And we are rewriting `dividePolylineWithPlane` to be completely sure that no infinite loop can happen again.

ammarhattab, 2025-02-21:
Thank you so much, I will try it once it's ready.

ABSitf, 2025-02-26:
The problem you are most likely experiencing has been found and fixed.
Try to check with the actual master.

ammarhattab, 2025-02-27:
Thanks for the update. I was using the prebuilt version from here:
`https://github.com/MeshInspector/Meshlib/releases/download/v3.0.3.233/meshlib_v3.0.3.233_ubuntu22-arm64-dev.deb`
and here:
`https://github.com/MeshInspector/Meshlib/releases/download/v3.0.3.233/meshlib_3.0.3.233_ubuntu22-dev.deb`

Is there a new prebuilt version?

Grantim, 2025-02-28:
Hello!
Here there are draft versions:
[meshlib_v3.0.3.305_ubuntu22-dev](https://meshinspector-my.sharepoint.com/:u:/p/grant_karapetyan/EdxhNUCKRoxEi9JNC52675UBkVVzLFEMv7vRwwBVALmz7g?e=tAicI2)
[meshlib_v3.0.3.305_ubuntu22-arm64-dev](https://meshinspector-my.sharepoint.com/:u:/p/grant_karapetyan/ESGqDpZhW0pMvx9uC8rYNZMB_tl4vuyAn4hNfwtUsWmywg?e=psLmJs)

Usually new versions are available after release.

closed 2025-06-18 by ABSitf