MeshInspector/MeshLib #4223: what dose ‘packmesh’ mean？

1358361259, 2025-03-01:
Hello everyone,

I came across the pack operation while working with meshes, but I'm not entirely sure what it does.
Could someone explain its purpose? How does it influence the final mesh generation?

I would really appreciate any examples or references.

Thanks in advance for your help! 🙏

[image]

Fedr, 2025-03-01:
Hello,

Mesh contains array of edges, triangles and vertices. Some array elements may be unused (contain deleted items) after some operations like decimation/simplification and others. Mesh packing eliminates all unused elements and reduces the size of mesh in memory, but the IDs of existing elements are changed during packing, so it is not always desirable to do. That is why we have a option in remesh whether to perform packing or not.

1358361259, 2025-03-01:
Thank you for your reply. I look forward to more detailed documentation for meshlib, as it is the fastest and most effective mesh processing library I have encountered.