MeshInspector/MeshLib #4237: Undefined Symbol Errors When Compiling MeshLib with Emscripten

Liu-langqi, 2025-03-09:
Description:

The commands used are:  
```bash
$ emcmake cmake -G Ninja -B ./build  
$ ninja -C ./build  
```

I encountered wasm-ld undefined symbol errors while compiling MeshLib with Emscripten. The error messages are as follows:

```bash
ninja: Entering directory `./build/'
[2/2] Linking CXX executable test1.js
FAILED: test1.js 
: && /home/sugar/emsdk/upstream/emscripten/em++  -lembind -s WASM=1 -s ALLOW_MEMORY_GROWTH=1 -flto -s DISABLE_EXCEPTION_CATCHING=1 -s MODULARIZE=1 -s EXPORT_NAME=test1Module -s EXPORT_ES6=1 -s DEMANGLE_SUPPORT=1 "--closure 1" -Oz -g2 CMakeFiles/test1.dir/createBase.cpp.o -o test1.js -L/home/sugar/MeshLib/build/Release/bin   -L/home/sugar/MeshLib/lib -lMRMesh  -lMRVoxels && :
em++: warning: DEMANGLE_SUPPORT is deprecated (mangled names no longer appear in stack traces). Please open a bug if you have a continuing need for this setting [-Wdeprecated]
cache:INFO: generating system asset: symbol_lists/25294393c8cc7e3775e70149e0840775c3806439.json... (this will be cached in "/home/sugar/emsdk/upstream/emscripten/cache/symbol_lists/25294393c8cc7e3775e70149e0840775c3806439.json" for subsequent builds)
cache:INFO:  - ok
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::Message::Message()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::operator=(testing::Message const&) const
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::~AssertHelper()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::Message::Message()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::operator=(testing::Message const&) const
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::~AssertHelper()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::Message::Message()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::operator=(testing::Message const&) const
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::~AssertHelper()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::Message::Message()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::operator=(testing::Message const&) const
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::~AssertHelper()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::Message::Message()
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)
wasm-ld: error: /home/sugar/MeshLib/build/Release/bin/libMRMesh.a(MRBitSet.cpp.o): undefined symbol: testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)
wasm-ld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
em++: error: '/home/sugar/emsdk/upstream/bin/wasm-ld -o test1.wasm -lembind-rtti CMakeFiles/test1.dir/createBase.cpp.o -L/home/sugar/MeshLib/build/Release/bin -L/home/sugar/MeshLib/lib -lMRMesh -lMRVoxels -L/home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/lto -L/home/sugar/emsdk/upstream/emscripten/src/lib -lGL-getprocaddr -lal -lhtml5 -lc_optz -lstubs -lnoexit -lc -ldlmalloc -lcompiler_rt -lc++-noexcept -lc++abi-noexcept -lsockets -mllvm -combiner-global-alias-analysis=false -mllvm -enable-emscripten-sjlj -mllvm -disable-lsr /tmp/tmp8ntz016clibemscripten_js_symbols.so -u__cxa_atexit --export=__cxa_demangle --export=free --export=_emscripten_stack_alloc --export=__getTypeName --export=__wasm_call_ctors --export=_emscripten_stack_restore --export=emscripten_stack_get_current --export-if-defined=__start_em_asm --export-if-defined=__stop_em_asm --export-if-defined=__start_em_lib_deps --export-if-defined=__stop_em_lib_deps --export-if-defined=__start_em_js --export-if-defined=__stop_em_js --export-if-defined=main --export-if-defined=__main_argc_argv --export-table -z stack-size=65536 --max-memory=2147483648 --initial-heap=16777216 --no-entry --table-base=1 --global-base=1024' failed (returned 1)
ninja: build stopped: subcommand failed.
```

Environment Details:
Ubuntu22.04
Cmake version: 3.31.6
Emscripten version: 4.0.4
Clang version: 21.0.0git
Build system: CMake + Ninja
Target platform: wasm32-unknown-emscripten
MeshLib source path: /home/sugar/MeshLib

I have successfully executed `./scripts/build_thirdparty.sh` and `./scripts/build_source.sh`. After execution, the output message was:  

**"Autoinstall script successfully finished. You could run ./build/Release/bin/MRTest next."**

CMakeLists.txt:
```
add_executable(test1 test1.cpp)
target_include_directories(test1 PUBLIC ${MESHLIB_INCLUDE_DIR} ${MESHLIB_THIRDPARTY_INCLUDE_DIR})
target_link_libraries(test1 PUBLIC MRMesh MRVoxels)
target_link_directories(test1 PUBLIC ${MESHLIB_LIB_DIR} ${MESHLIB_THIRDPARTY_LIB_DIR})
```

oitel, 2025-03-10:
Hello, unfortunately MeshLib usage as an external library with Emscripten hasn't been tested yet, we will extend its support in the future. The supported way to use MeshLib with Emscripten is to add it as a sub-project:
```cmake
add_subdirectory("/home/sugar/MeshLib")
```
You could also try to add the MeshLib's thirdparty directory to the CMake's search paths:
```bash
export MESHLIB_THIRDARTY_ROOT_DIR=/home/sugar/MeshLib # the same dir in which you've executed ./scripts/build_thirdparty.sh
emcmake cmake -G Ninja -B ./build -D CMAKE_FIND_ROOT_PATH=${MESHLIB_THIRDARTY_ROOT_DIR}
# or
cmake -G Ninja -B ./build -D CMAKE_FIND_ROOT_PATH=${MESHLIB_THIRDARTY_ROOT_DIR} -D CMAKE_TOOLCHAIN_FILE=${EMSDK}/upstream/emscripten/cmake/Modules/Platform/Emscripten.cmake
```

Liu-langqi, 2025-03-11:
### **Bug Report: Error when packaging MeshLib code into WASM**  

#### **Problem Description**  
When attempting to compile **MeshLib** code into WebAssembly (WASM) and run it in the browser, the following error appears in the console:  

```plaintext
chunk-DEEFU7IG.js?v=2c9722c7:7009 Uncaught 
TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at Mesh.updateMorphTargets (chunk-DEEFU7IG.js?v=2c9722c7:7009:25)
    at new Mesh (chunk-DEEFU7IG.js?v=2c9722c7:6992:10)
    at i._onChange (main.js:237:24)
    at i._callOnChange (three_examples_jsm_l…js?v=2c9722c7:13:82)
    at i.setValue (three_examples_jsm_l…js?v=2c9722c7:64:50)
    at HTMLInputElement.<anonymous> (three_examples_jsm_l…js?v=2c9722c7:82:12)
```

Additionally, the console shows a runtime exception in `test1.js`, causing an **Aborted()** error:  

```plaintext
creatTeethBase.js:116 Module loaded: 
{HEAP8: Int8Array(16777216), HEAP16: Int16Array(8388608), HEAPU8: Uint8Array(16777216), HEAPU16: Uint16Array(8388608), HEAP32: Int32Array(4194304), …}
test1.js:8 Aborted()
creatTeethBase.js:142 Error in createCUBE: 
RuntimeError: Aborted(). Build with -sASSERTIONS for more info.
    at abort (test1.js:8:6371)
    at __abort_js (test1.js:8:19191)
    at test1.wasm:0x26a65
    at test1.wasm:0x5f54
    at test1.wasm:0x73d5
    at test1.wasm:0x1941
    at test1.wasm:0x1089c
    at ___cxa_increment_exception_refcount (test1.js:8:70075)
    at ___cxa_begin_catch (test1.js:8:15943)
    at test1.wasm:0xa60d
```

#### **Source Code**  
The function causing the issue is as follows:  

```cpp
MR::Mesh createCUBE() {
    // Create mesh
    MR::Mesh mesh = MR::makeCube();
    return mesh; 
}
```

I am not sure what is causing this error. Any insights or suggestions on how to resolve this issue would be greatly appreciated.

Grantim, 2025-03-11:
Hello!

Do you test release or debug compilation of MeshLib? It looks like there are no debug symbols in the logs you've provided

Liu-langqi, 2025-03-11:
---

## Compilation Issue with Emscripten and MeshLib

### **Issue Description**
**FIRST**
I encountered errors when compiling my project using Emscripten. Below is my `CMakeLists.txt`:

```cmake
cmake_minimum_required(VERSION 3.15 FATAL_ERROR)

set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

project(test1 LANGUAGES CXX VERSION 1.0)

set(MESHLIB_INCLUDE_DIR /home/sugar/MeshLib/source)
set(MESHLIB_LIB_DIR /home/sugar/MeshLib/build/Release/bin)
set(MESHLIB_THIRDPARTY_INCLUDE_DIR /home/sugar/MeshLib/include)
set(MESHLIB_THIRDPARTY_LIB_DIR /home/sugar/MeshLib/lib)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Os -Wno-pthreads-mem-growth")

add_executable(test1 createBase.cpp)

target_include_directories(test1 PUBLIC
    ${MESHLIB_INCLUDE_DIR}
    ${MESHLIB_THIRDPARTY_INCLUDE_DIR}
)
target_link_directories(test1 PUBLIC
    ${MESHLIB_LIB_DIR}
    ${MESHLIB_THIRDPARTY_LIB_DIR}
)

target_link_libraries(test1 PRIVATE
    MRMesh
    fmt
    tbb
    openvdb
    spdlog
)
```

When I run:

```sh
emmake make
```

I get the following error:

```sh
[ 50%] Linking CXX executable test1.js
wasm-ld: error: CMakeFiles/test1.dir/createBase.cpp.o: undefined symbol: MR::generalOffsetMesh(MR::MeshRegion<MR::FaceTag> const&, float, MR::GeneralOffsetParameters const&)
em++: error: '/home/sugar/emsdk/upstream/bin/wasm-ld -o test1.wasm -lembind-rtti CMakeFiles/test1.dir/createBase.cpp.o -L/home/sugar/MeshLib/build/Release/bin -L/home/sugar/MeshLib/lib -lMRMesh -lfmt -ltbb -lopenvdb -lspdlog -L/home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten -L/home/sugar/emsdk/upstream/emscripten/src/lib /home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/libboost_headers.a -lGL-getprocaddr -lal -lhtml5 -lstubs -lnoexit -lc -ldlmalloc -lcompiler_rt -lc++-noexcept -lc++abi-noexcept -lsockets -mllvm -combiner-global-alias-analysis=false -mllvm -enable-emscripten-sjlj -mllvm -disable-lsr /tmp/tmpxjyvm4eulibemscripten_js_symbols.so --strip-debug --export=_emscripten_stack_alloc --export=__getTypeName --export=__wasm_call_ctors --export=emscripten_stack_get_current --export=_emscripten_stack_restore --export-if-defined=__start_em_asm --export-if-defined=__stop_em_asm --export-if-defined=__start_em_lib_deps --export-if-defined=__stop_em_lib_deps --export-if-defined=__start_em_js --export-if-defined=__stop_em_js --export-if-defined=main --export-if-defined=__main_argc_argv --export-table -z stack-size=65536 --max-memory=4294967296 --initial-memory=16777216 --no-entry --table-base=1 --global-base=1024' failed (returned 1)
make[2]: *** [CMakeFiles/test1.dir/build.make:103: test1.js] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/test1.dir/all] Error 2
...
```

It seems that `generalOffsetMesh` is not found during linking, even though `MRMesh` is linked.

### **Second Issue (Adding MRVoxels)**
If I add `MRVoxels` to `target_link_libraries`, I get additional errors:

```sh
[ 50%] Linking CXX executable test1.js
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: compressBound
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: compress2
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: zError
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: uncompress
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: zError
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: blosc_compress_ctx
wasm-ld: error: /home/sugar/MeshLib/lib/libopenvdb.a(Compression.cc.o): undefined symbol: blosc_decompress_ctx
em++: error: '/home/sugar/emsdk/upstream/bin/wasm-ld -o test1.wasm -lembind-rtti CMakeFiles/test1.dir/createBase.cpp.o -L/home/sugar/MeshLib/build/Release/bin -L/home/sugar/MeshLib/lib -lMRMesh -lMRVoxels -lfmt -ltbb -lopenvdb -lspdlog -L/home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten -L/home/sugar/emsdk/upstream/emscripten/src/lib /home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/libboost_headers.a -lGL-getprocaddr -lal -lhtml5 -lstubs -lnoexit -lc -ldlmalloc -lcompiler_rt -lc++-noexcept -lc++abi-noexcept -lsockets -mllvm -combiner-global-alias-analysis=false -mllvm -enable-emscripten-sjlj -mllvm -disable-lsr /tmp/tmpoc3ha14qlibemscripten_js_symbols.so --strip-debug --export=_emscripten_stack_alloc --export=__getTypeName --export=__wasm_call_ctors --export=emscripten_stack_get_current --export=_emscripten_stack_restore --export-if-defined=__start_em_asm --export-if-defined=__stop_em_asm --export-if-defined=__start_em_lib_deps --export-if-defined=__stop_em_lib_deps --export-if-defined=__start_em_js --export-if-defined=__stop_em_js --export-if-defined=main --export-if-defined=__main_argc_argv --export-table -z stack-size=65536 --max-memory=4294967296 --initial-memory=16777216 --no-entry --table-base=1 --global-base=1024' failed (returned 1)
make[2]: *** [CMakeFiles/test1.dir/build.make:103: test1.js] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/test1.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
emmake: error: 'make' failed (returned 2)
...
```

---

### **Questions**
1. Why is `generalOffsetMesh` missing during linking even though `MRMesh` is linked?
2. How should I properly link OpenVDB's dependencies (`zlib`, `blosc`) in an Emscripten environment?
3. Could there be an issue with how `MRMesh` or `MRVoxels` was built? How can I verify it?
4. Is there a recommended way to check if a library was built correctly for WebAssembly?

Any insights would be greatly appreciated!

---

Grantim, 2025-03-11:
> Why is generalOffsetMesh missing during linking even though MRMesh is linked?

`generalOffsetMesh` is part of MRVoxels, so you need to link `MRVoxels`

> How should I properly link OpenVDB's dependencies (zlib, blosc) in an Emscripten environment?

Please note that in our emscripten build all libraries are ended up being statically linked into `.wasm` file, 
So you need to add `blosc` and `zip` to `target_link_libraries` as well (just somewhere because actual linkage is happening only once in the end)
Also, note that some libs should be among Emscripten linker flags too
https://github.com/MeshInspector/MeshLib/blob/20d58540a3c770fd86793a04e06f20c165a6c25f/cmake/Modules/CompilerOptions.cmake#L148-L189

> Could there be an issue with how MRMesh or MRVoxels was built? How can I verify it?

Most likely it is built correctly, but we now does not have support for simply linking MeshLib wasm into external project (we are working on it). So easiest way is to use our root cmake file an just add your project to it.

> Is there a recommended way to check if a library was built correctly for WebAssembly?

You can check it like we do in our test system.
https://github.com/MeshInspector/MeshLib/blob/20d58540a3c770fd86793a04e06f20c165a6c25f/.github/workflows/build-test-emscripten.yml#L92-L105
(starting local server and running tests in it)

Liu-langqi, 2025-03-12:
Hello, today I followed your suggestion again—"**to use our root CMake file and just add your project to it**" I encountered some issues, but it seems that I can generate the `.wasm` file. Please do not close this issue yet. I will organize a tutorial on how to package the project into `.wasm` to address common problems that beginners often face. Thank you again for your guidance!

Grantim, 2025-03-12:
Sure, Thanks!

Liu-langqi, 2025-03-12:
**Description:**  
The project structure and `CMakeLists.txt` file configuration for `my_project` in MeshLib.  

#### Project Directory Structure  
```bash
MeshLib/
├── MeshLib/
│   ├── source/
│   │   ├── my_project/
│   │   │   ├── test2.cpp
│   │   │   ├── CMakeLists.txt
```

#### `CMakeLists.txt` Content  
```cmake
# Project name and language
project(my_project LANGUAGES CXX)

# Include Emscripten toolchain
include(${EMSCRIPTEN_ROOT_PATH}/cmake/Modules/Platform/Emscripten.cmake)

# Set C++ standard
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# Add compilation options
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -frtti -fexceptions -pthread")

# Add executable file
add_executable(test2
  test2.cpp
)

# Link libraries
target_link_libraries(test2 PRIVATE
  MRMesh           # MeshLib core library
  MRVoxels         # MeshLib voxel library
  embind           # Emscripten binding library
  fmt              # Formatting library
  spdlog           # Logging library
)

# Set link options
set_target_properties(test2 PROPERTIES
  LINK_FLAGS "-s MODULARIZE=1 -s EXPORT_ES6=1 -s WASM=1 -s SHARED_MEMORY"
)

# Define compile macros
target_compile_definitions(test2 PRIVATE
  SPDLOG_FMT_EXTERNAL    # Use external fmt library
  MR_EMSCRIPTEN_ASYNCIFY # Enable Emscripten asyncify (if needed)
)
```

### Build failure when compiling `test2.js` with Emscripten

#### **Issue Description**
When running the following command in the `~/MeshLib` directory:

```sh
mkdir build_wasm && cd build_wasm
emcmake cmake .. -DCMAKE_BUILD_TYPE=Release -DMR_EMSCRIPTEN=ON -DMESHLIB_BUILD_MRVIEWER=OFF -DMESHLIB_BUILD_MESHVIEWER=OFF -DBUILD_TESTING=OFF
emmake make -j8 VERBOSE=1
```

I encountered a build error at the linking stage:

#### **Error Output**
```
[100%] Linking CXX executable ../../bin/test2.js
cd /home/sugar/MeshLib/build_wasm/source/my_project && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/test2.dir/link.txt --verbose=1
/home/sugar/emsdk/upstream/emscripten/em++ -Wall -Wextra -Wno-missing-field-initializers -Wno-unknown-pragmas -Wno-sign-compare -Wno-newline-eof -Werror -fvisibility=hidden -pedantic-errors -DIMGUI_DISABLE_OBSOLETE_FUNCTIONS -DIMGUI_ENABLE_FREETYPE -pthread -Wno-pthreads-mem-growth -frtti -fexceptions -pthread -O3 -DNDEBUG -Wl,-z,defs -s EXPORTED_RUNTIME_METHODS=[ccall] -s ALLOW_MEMORY_GROWTH=1 -s LLD_REPORT_UNDEFINED=1 -s USE_WEBGL2=1 -s USE_GLFW=3 -s USE_ZLIB=1 -s FULL_ES3=1 -s USE_LIBPNG=1 -s ENVIRONMENT=web,worker -pthread -s PTHREAD_POOL_SIZE_STRICT=0 -s PTHREAD_POOL_SIZE=navigator.hardwareConcurrency -s MAXIMUM_MEMORY=4GB -s ASYNCIFY -Wno-limited-postlink-optimizations -s MODULARIZE=1 -s EXPORT_ES6=1 -s WASM=1 -s SHARED_MEMORY @CMakeFiles/test2.dir/objects1.rsp -o ../../bin/test2.js @CMakeFiles/test2.dir/linkLibs.rsp
wasm-ld: warning: unknown -z value: defs
wasm-ld: error: --shared-memory is disallowed by Grid.cc.o because it was not compiled with 'atomics' or 'bulk-memory' features.
em++: error: '/home/sugar/emsdk/upstream/bin/wasm-ld -o ../../bin/test2.wasm -z defs CMakeFiles/test2.dir/test2.cpp.o -L/home/sugar/MeshLib/lib ../../bin/libMRMesh.a ../../bin/libMRVoxels.a -lembind-rtti -lfmt -lspdlog ../../bin/libMRMesh.a -lblosc -lmbedtls -lmbedx509 -lmbedcrypto -lzip -lopenvdb -ljsoncpp -ltbb /home/sugar/MeshLib/lib/libgdcmMEXD.a /home/sugar/MeshLib/lib/libsocketxx.a /home/sugar/MeshLib/lib/libgdcmMSFF.a /home/sugar/MeshLib/lib/libgdcmDICT.a /home/sugar/MeshLib/lib/libgdcmIOD.a /home/sugar/MeshLib/lib/libgdcmexpat.a /home/sugar/MeshLib/lib/libgdcmDSED.a /home/sugar/MeshLib/lib/libgdcmCommon.a /home/sugar/MeshLib/lib/libgdcmzlib.a /home/sugar/MeshLib/lib/libgdcmjpeg8.a /home/sugar/MeshLib/lib/libgdcmjpeg12.a /home/sugar/MeshLib/lib/libgdcmjpeg16.a /home/sugar/MeshLib/lib/libgdcmopenjp2.a /home/sugar/MeshLib/lib/libgdcmcharls.a /home/sugar/MeshLib/lib/libgdcmuuid.a -lfmt -lspdlog -L/home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten -L/home/sugar/emsdk/upstream/emscripten/src/lib /home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/libpng-mt.a /home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/libz.a /home/sugar/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/crtbegin.o -lGL-mt-webgl2-full_es3-getprocaddr -lal -lhtml5 -lstubs -lnoexit -lc-mt -ldlmalloc-mt -lcompiler_rt-mt -lc++-mt -lc++abi-mt -lsockets-mt -mllvm -combiner-global-alias-analysis=false -mllvm -enable-emscripten-cxx-exceptions -mllvm -enable-emscripten-sjlj -mllvm -disable-lsr /tmp/tmppm3xv4uwlibemscripten_js_symbols.so --import-memory --shared-memory --export=_emscripten_stack_alloc --export=__getTypeName --export=_emscripten_thread_free_data --export=_emscripten_thread_crashed --export=_embind_initialize_bindings --export=__cxa_can_catch --export=__cxa_increment_exception_refcount --export=__cxa_decrement_exception_refcount --export=setThrew --export=__cxa_free_exception --export=__wasm_call_ctors --export=_emscripten_tls_init --export=_emscripten_thread_init --export=emscripten_stack_get_current --export=_emscripten_stack_restore --export=emscripten_stack_set_limits --export=_emscripten_thread_exit --export=malloc --export=free --export-if-defined=__start_em_asm --export-if-defined=__stop_em_asm --export-if-defined=__start_em_lib_deps --export-if-defined=__stop_em_lib_deps --export-if-defined=__start_em_js --export-if-defined=__stop_em_js --export-if-defined=main --export-if-defined=__main_argc_argv --export-table -z stack-size=65536 - [... cut ...]
emmake: error: 'make -j8 VERBOSE=1' failed (returned 2)
```

#### **Possible Cause**
The key error message appears to be:
```
wasm-ld: error: --shared-memory is disallowed by Grid.cc.o because it was not compiled with 'atomics' or 'bulk-memory' features.
```
This suggests that `Grid.cc.o` was compiled without `atomics` or `bulk-memory` features, but the linking process requires shared memory, which depends on those features.

**The main usage in `test2.cpp` is:  **

```cpp
#include <MRVoxels/MROffset.h>
MR::generalOffsetMesh();
```

Liu-langqi, 2025-03-12:
```bash
em++   -Wall -Wextra -Wno-missing-field-initializers -Wno-unknown-pragmas -Wno-sign-compare -Wno-newline-eof -Werror -fvisibility=hidden -pedantic-errors -DIMGUI_DISABLE_OBSOLETE_FUNCTIONS -DIMGUI_ENABLE_FREETYPE -pthread -Wno-pthreads-mem-growth -frtti -fexceptions -pthread -O3 -DNDEBUG  -Wl,-z,defs -s EXPORTED_RUNTIME_METHODS=[ccall] -s ALLOW_MEMORY_GROWTH=1 -s LLD_REPORT_UNDEFINED=1 -s USE_WEBGL2=1 -s USE_GLFW=3 -s USE_ZLIB=1 -s FULL_ES3=1 -s USE_LIBPNG=1 -s ENVIRONMENT=web,worker -pthread -s PTHREAD_POOL_SIZE_STRICT=0 -s PTHREAD_POOL_SIZE=navigator.hardwareConcurrency -s MAXIMUM_MEMORY=4GB -s ASYNCIFY -Wno-limited-postlink-optimizations -s MODULARIZE=1 -s EXPORT_ES6=1 -s WASM=1 -s SHARED_MEMORY @CMakeFiles/test2.dir/objects1.rsp -o ../../bin/test2.js @CMakeFiles/test2.dir/linkLibs.rsp 
```

From the additional information above, we can see that -pthread is still being used.

Liu-langqi, 2025-03-12:
1. I will take note of that. The above example can basically generate a .wasm file. I am using single-threaded mode. However, it is important to note that the third-party build uses multi-threaded mode, so my own project also needs to use multi-threaded mode.
2. Thank you again for your guidance. This answer can also serve as an example for beginners regarding .wasm files. Congratulations on your success at the 2025 Cologne International Exhibition in Germany!

Grantim, 2025-03-12:
Thanks! Actually we are working now on providing easier wasm integration options along with guides.

closed 2025-08-28 by Liu-langqi