MeshInspector/MeshLib #4266: Inquiry About the Python API for "Distance Compare" in MeshInspector

1358361259, 2025-03-15:
Excuse me, may I ask which Python API corresponds to the `Distance Compare` function in MeshInspector? This function can calculate the minimum, maximum, and average distances between two surface meshes, which is exactly what I need.
[image]

1358361259, 2025-03-15:
I've tried writing the following code, but as you can see, I can only compute the distance from each point on one surface (mesh1) to the target surface (mesh2) in a loop. This approach is very slow. Could you please provide any suggestions for improving the performance?

Here is the code I wrote:
```
import time  
import numpy as np  
from meshlib import mrmeshpy  
from meshlib import mrmeshnumpy  

# Load an STL file and create a Mesh object  
mesh = mrmeshpy.loadMesh('surface.stl')  
mesh.pack()  

meshpart = mrmeshpy.MeshPart(mesh)  
settings = mrmeshpy.MeshComponents.LargeByAreaComponentsSettings()  
settings.maxLargeComponents = 2  
twosurface = mrmeshpy.MeshComponents.getNLargeByAreaComponents(meshpart, settings)  

mesh1 = mesh.cloneRegion(twosurface[0])  
mesh2 = mesh.cloneRegion(twosurface[1])  

# Initialize a numpy array to store distances, with the same length as the number of vertices in mesh1  
num_points = len(mrmeshnumpy.getNumpyVerts(mesh1))  
distances = np.zeros(num_points)  

# Start timing  
start_time = time.time()  

# Loop through all vertices of mesh1 and compute the distance to mesh2  
for i, point in enumerate(mrmeshnumpy.getNumpyVerts(mesh1)):  
    point = mrmeshpy.Vector3f(point[0], point[1], point[2])  
    distances[i] = abs(mesh2.signedDistance(point))  

# End timing  
end_time = time.time()  

# Print statistics  
print(f"Min distance: {distances.min()}")  
print(f"Max distance: {distances.max()}")  
print(f"Mean distance: {distances.mean()}")  

# Print execution time  
print(f"Execution time: {end_time - start_time:.4f} seconds")  

```

Fedr, 2025-03-15:
Please try to call this function:
```c++
/// Computes signed distances from valid vertices of test mesh to the closest point on the reference mesh:
/// positive value - outside reference mesh, negative - inside reference mesh;
/// this method can return wrong sign if the closest point is located on self-intersecting part of the mesh
[[nodiscard]] MRMESH_API VertScalars findSignedDistances(
    const Mesh& refMesh, ///< the distances to this mesh will be returned
    const Mesh& mesh,    ///< the distances from valid vertices of this mesh will be returned
    const MeshProjectionParameters & params = {},
    IPointsToMeshProjector * projector = {} ); ///< if projector is not given then CPU's computations will be used
```

It shall be available in Python API as well.

1358361259, 2025-03-16:
thank you so much！It is what I need. In fact，I tried searching for singeddistance, but the findsigneddistances option didn’t appear in the suggestion list. You could consider adjusting the search strategy.

[image]